osrf/homebrew-simulation #384: Conflict with CMake's FindDart.cmake module when compiling a downstream project that depends on gazebo

traversaro, 2018-03-13:
Some homebrew users of the https://github.com/robotology/robotology-superbuild on macOS (@DanielePucci) are reporting the impossibility of compiling a project that calls `find_package(gazebo REQUIRED)` when using the latest gazebo9 installed by homebrew. 

I investigated a bit, and the problem is that:
* gazebo9 is now depending on Dart, that is installed by default
* The line [`find_package(DART ${GAZEBO_DART_MIN_REQUIRED_VERSION} REQUIRED)`](https://bitbucket.org/osrf/gazebo/src/55627fc420a41afc49356d0f7269484759af3394/cmake/gazebo-config.cmake.in?at=default&fileviewer=file-view-default#gazebo-config.cmake.in-124) contained in the `gazebo-config.cmake` file does not work on macOs due to the conflict with the `FindDart.cmake` module distributed with CMake (see https://github.com/dartsim/dart/pull/889 for a related Dart PR). 

The possibility of fixing this are (as far as I can think):
* either to patch the `gazebo-config.cmake` contained in the bottle to contain the line `find_package(DART ${GAZEBO_DART_MIN_REQUIRED_VERSION} REQUIRED CONFIG)`
* fix the problem in Gazebo upstream, and do a new release from which to generate a bottle.

traversaro, 2018-03-13:
Upstream fix: https://bitbucket.org/osrf/gazebo/pull-requests/2919/fix-https-githubcom-osrf-homebrew/diff
Related Gazebo PR: https://bitbucket.org/osrf/gazebo/pull-requests/2919/fix-https-githubcom-osrf-homebrew/diff .

scpeters, 2018-03-15:
I'm adding a patch to the gazebo9 formula in #386

PR #386 (2018-03-15): gazebo9: patch for find dart in cmake config
Fixes #384.
closed 2018-03-15 by scpeters