osrf/homebrew-simulation #441: brew install gazebo8/9 fails building on High Sierra

TSC21, 2018-06-22:
Currently on a Travis CI building job, with the following script lines:
```
    ...
    brew tap osrf/simulation;
    brew install opencv ${GAZEBO_VERSION};
    ...
```
where `GAZEBO_VERSION` corresponds to the respective Gazebo version to install.

The following build error is reported: https://travis-ci.org/PX4/sitl_gazebo/jobs/395170004#L1351.

If it helps, I may add the `--verbose` tag if required. This started to happen 2/3 days ago, for each build.

Any tips debugging it are appreciated.

scpeters, 2018-06-22:
it will be fixed by https://github.com/osrf/homebrew-simulation/pull/440

I'm just waiting for the travis CI to finish testing, then I will merge it. Sorry for the outage.

* https://bitbucket.org/osrf/gazebo/issues/2483/gazebo-compilation-broken-with-protobuf-36

scpeters, 2018-06-22:
bottles should now be fixed

closed 2018-06-22 by scpeters