osrf/homebrew-simulation #447: gazebo9 fails because of missing ros/deps formula

TSC21, 2018-07-13:
Dears,

While taping, and brew installing gazebo9, I get the following:
```
==> Installing gazebo9 from osrf/simulation
==> Tapping ros/deps
Cloning into '/usr/local/Homebrew/Library/Taps/ros/homebrew-deps'...
remote: Counting objects: 11, done.
remote: Compressing objects: 100% (11/11), done.
remote: Total 11 (delta 0), reused 4 (delta 0), pack-reused 0
Unpacking objects: 100% (11/11), done.
Tapped 6 formulae (42 files, 50KB).
Error: No available formula with the name "ros/deps/urdfdom" (dependency of osrf/simulation/gazebo9)
==> Searching for a previously deleted formula (in the last month)...
Warning: ros/deps is shallow clone. To get complete history run:
  git -C "$(brew --repo ros/deps)" fetch --unshallow
```
Is this supposed to happens with the `ros/deps` formulas? Any solution for this?

Thanks!

TSC21, 2018-07-13:
Probably related: https://github.com/osrf/homebrew-simulation/issues/380

jslee02, 2018-07-13:
I have the same problem.

> Probably related: #380

That's possible. But it's wired because all the `urdf` dependencies of formula in this repo are optional so it shouldn't be a problem. 🤔

TSC21, 2018-07-13:
> That's possible. But it's wired because all the urdf dependencies of formula in this repo are optional so it shouldn't be a problem. 

True. Let's wait for @scpeters feedback.

scpeters, 2018-07-13:
I'm not sure why it's happening just to you, but @jslee02 is right in #380, we should switch to the core urdfdom. See #448