osrf/homebrew-simulation #460: Server down?

jslee02, 2018-08-03:
This could be a temporary issue, but I see following errors when installing `gazebo9` using Homebrew:
```
$ brew upgrade
==> Upgrading 1 outdated package, with result:
osrf/simulation/gazebo9 9.3.0 -> 9.3.0_1
==> Upgrading osrf/simulation/gazebo9 
==> Downloading http://gazebosim.org/distributions/gazebo/releases/gazebo9-9.3.0

curl: (22) The requested URL returned error: 504 GATEWAY_TIMEOUT
Error: Failed to download resource "gazebo9"
Download failed: http://gazebosim.org/distributions/gazebo/releases/gazebo9-9.3.0_1.high_sierra.bottle.tar.gz
Warning: Bottle installation failed: building from source.
==> Downloading http://gazebosim.org/distributions/gazebo/releases/gazebo-9.3.0.

curl: (22) The requested URL returned error: 504 GATEWAY_TIMEOUT
Error: Failed to download resource "gazebo9"
Download failed: http://gazebosim.org/distributions/gazebo/releases/gazebo-9.3.0.tar.bz2
```

scpeters, 2018-08-03:
gazebosim.org was down but it's back up now

closed 2018-08-03 by scpeters

jslee02, 2018-08-04:
It's fine now. Thanks!