osrf/homebrew-simulation #509: macOS Mojave Gazebo9 build failure

smnogar, 2018-10-06:
I'm attempting to get ROS Melodic up and running on Mojave and I'm running into Gazebo9 build problems. Current version of Gazebo9 is 9.4.1. The gist is [here](https://gist.github.com/4e6b5f95406a9ebb33776af493d1b42c). Gazebo8 seems to build fine. 

The last few lines of output are:

```cd /tmp/gazebo9-20181006-74114-1flwtuc/gazebo-9.4.1/build/gazebo && /usr/local/Cellar/cmake/3.12.3/bin/cmake -E cmake_link_script CMakeFiles/gzserver.dir/link.txt --verbose=1
/usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++      -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Wswitch-default -Wmissing-include-dirs -pedantic -std=c++11   -O3 -DNDEBUG -mssse3 -msse3 -msse -msse2  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names   CMakeFiles/gzserver.dir/server_main.cc.o  -o gzserver-9.4.1  -L/tmp/gazebo9-20181006-74114-1flwtuc/gazebo-9.4.1/build/test  -L/usr/local/Cellar/tinyxml2/6.2.0/lib  -L/usr/local/Cellar/ffmpeg/4.0.2/lib  -L/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_6/lib  -L/usr/local/Cellar/sdformat6/6.1.0/lib  -L/usr/local/Cellar/bullet/2.87_1/lib -Wl,-rpath,/tmp/gazebo9-20181006-74114-1flwtuc/gazebo-9.4.1/build/test -Wl,-rpath,/usr/local/Cellar/tinyxml2/6.2.0/lib -Wl,-rpath,/usr/local/Cellar/ffmpeg/4.0.2/lib -Wl,-rpath,/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_6/lib -Wl,-rpath,/usr/local/Cellar/sdformat6/6.1.0/lib -Wl,-rpath,/usr/local/Cellar/bullet/2.87_1/lib libgazebo.9.4.1.dylib sensors/libgazebo_sensors.9.4.1.dylib rendering/libgazebo_rendering.9.4.1.dylib /usr/local/lib/libfreeimage.dylib /usr/local/lib/libtbb.dylib -lOgreRTShaderSystem -lOgreMain -lpthread -lOgreMain -lpthread -lOgreTerrain -lOgrePaging -lOgreMain -lpthread -lOgreOverlay -lOgreMain -lpthread -lpthread physics/libgazebo_physics.9.4.1.dylib util/libgazebo_util.9.4.1.dylib -ltinyxml2 ../deps/opende/libgazebo_ode.9.4.1.dylib ../deps/opende/GIMPACT/libgazebo_gimpact.9.4.1.dylib ../deps/opende/ou/libgazebo_opende_ou.9.4.1.dylib -lccd ../deps/opende/OPCODE/libgazebo_opcode.9.4.1.dylib -lBulletSoftBody -lBulletDynamics -lBulletCollision -lLinearMath /usr/local/lib/libdart-utils-urdf.6.6.1.dylib /usr/local/lib/libdart-utils.6.6.1.dylib /usr/local/lib/libtinyxml2.dylib /usr/local/opt/urdfdom/lib/liburdfdom_sensor.dylib /usr/local/opt/urdfdom/lib/liburdfdom_model_state.dylib /usr/local/opt/urdfdom/lib/liburdfdom_model.dylib /usr/local/opt/urdfdom/lib/liburdfdom_world.dylib -lconsole_bridge /usr/local/lib/libdart.6.6.1.dylib /usr/local/opt/libccd/lib/libccd.dylib /usr/local/opt/fcl/lib/libfcl.dylib /usr/local/lib/libassimp.dylib /usr/local/lib/libdart-external-odelcpsolver.6.6.1.dylib /usr/local/lib/liboctomap.dylib /usr/local/lib/liboctomath.dylib /usr/local/Cellar/simbody/3.5.4_2/lib/libSimTKsimbody.dylib /usr/local/Cellar/simbody/3.5.4_2/lib/libSimTKmath.dylib /usr/local/Cellar/simbody/3.5.4_2/lib/libSimTKcommon.dylib -framework Accelerate -ldl -lm -L/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_6/lib -lOgreRTShaderSystem -lOgreMain -lpthread -lOgreTerrain -lOgrePaging -lOgreOverlay -lOgreMain -lpthread -lOgreTerrain -lOgrePaging -lOgreOverlay transport/libgazebo_transport.9.4.1.dylib -lOgreTerrain -lOgrePaging -lOgreOverlay /usr/local/lib/libignition-transport4.4.0.0.dylib /usr/local/lib/libuuid.dylib /usr/local/lib/libuuid.dylib -framework OpenGL msgs/libgazebo_msgs.9.4.1.dylib common/libgazebo_common.9.4.1.dylib /usr/lib/libdl.dylib /usr/local/lib/libfreeimage.dylib /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_signals-mt.dylib /usr/local/lib/libboost_system-mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_program_options-mt.dylib /usr/local/lib/libboost_regex-mt.dylib /usr/local/lib/libboost_iostreams-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_chrono-mt.dylib /usr/local/lib/libboost_atomic-mt.dylib -ltinyxml -lavcodec -lavformat -lavutil -lavdevice /usr/lib/libcurl.dylib -lswscale /usr/local/lib/libtar.dylib /usr/local/lib/libtbb.dylib /usr/local/Cellar/sdformat6/6.1.0/lib/libsdformat.dylib /usr/local/lib/libignition-fuel_tools1.1.2.0.dylib -lgts -lgthread-2.0 -lgmodule-2.0 -lglib-2.0 -lintl -framework OpenAL /usr/local/lib/libignition-msgs1.1.0.0.dylib /usr/local/lib/libignition-math4.4.0.0.dylib /usr/local/lib/libprotobuf.dylib /usr/local/lib/libprotobuf.dylib -lpthread 
cd /tmp/gazebo9-20181006-74114-1flwtuc/gazebo-9.4.1/build/gazebo && /usr/local/Cellar/cmake/3.12.3/bin/cmake -E cmake_symlink_executable gzserver-9.4.1 gzserver
[ 78%] Built target gzserver
[ 78%] Linking CXX static library libgazebo_test_fixture.a
cd /tmp/gazebo9-20181006-74114-1flwtuc/gazebo-9.4.1/build/gazebo/test && /usr/local/Cellar/cmake/3.12.3/bin/cmake -P CMakeFiles/gazebo_test_fixture.dir/cmake_clean_target.cmake
cd /tmp/gazebo9-20181006-74114-1flwtuc/gazebo-9.4.1/build/gazebo/test && /usr/local/Cellar/cmake/3.12.3/bin/cmake -E cmake_link_script CMakeFiles/gazebo_test_fixture.dir/link.txt --verbose=1
/usr/bin/ar qc libgazebo_test_fixture.a  CMakeFiles/gazebo_test_fixture.dir/ServerFixture.cc.o CMakeFiles/gazebo_test_fixture.dir/ServerFixtureRecord.cc.o
/usr/bin/ranlib libgazebo_test_fixture.a
[ 78%] Built target gazebo_test_fixture
make: *** [all] Error 2```

scpeters, 2018-10-08:
~~~
In file included from /tmp/gazebo9-20181006-74114-1flwtuc/gazebo-9.4.1/gazebo/gui/qgv/private/QGVGraphPrivate.cpp:19:
/tmp/gazebo9-20181006-74114-1flwtuc/gazebo-9.4.1/gazebo/gui/qgv/private/QGVGraphPrivate.h:22:10: fatal error: 'cgraph.h' file not found
#include <cgraph.h>
         ^~~~~~~~~~
1 error generated.
make[2]: *** [gazebo/gui/CMakeFiles/gazebo_gui.dir/qgv/private/QGVGraphPrivate.cpp.o] Error 1
~~~

Thanks for the report. I started building and uploading mojave bottles on Friday and hadn't quite gotten to gazebo yet. I still need to build ignition-common, ignition-fuel-tools, and sdformat. In the meantime, it looks like the error is related to the schematic view in the model editor and graphviz support.

scpeters, 2018-10-08:
I just uploaded a mojave bottle without QGV / graphviz support, but you should be able to use it for your ROS Melodic work

smnogar, 2018-10-10:
Great, it works now. Thanks!

closed 2018-10-10 by smnogar