osrf/homebrew-simulation #512: brew install ogre fails

CrazyForName, 2018-10-15:
log link:https://gist.github.com/CrazyForName/31ad980279b428dc83d6388d15f86702

scpeters, 2018-10-15:
Before trying to debug the failure, I'll just note that the `ogre` formula is for an old version: 1.7.4. There are working bottles for `ogre1.9`, so my first suggestion is to use 1.9 if that is an option for you.

Regarding the 1.7 build failure, it looks like it's using the 10.13 SDK even though you are on 10.12. I've seen this problem before and tried to work around it with simbody:

* https://github.com/osrf/homebrew-simulation/blob/master/simbody.rb#L30-L32

If you're still going to try building ogre 1.7, can you try the following patch?

~~~
diff --git a/ogre.rb b/ogre.rb
index 78e008e..9096071 100644
--- a/ogre.rb
+++ b/ogre.rb
@@ -59,6 +59,9 @@ class Ogre < Formula
   end
 
   def install
+    # Don't use 10.13 SDK frameworks on 10.12
+    ENV.delete("MACOSX_DEPLOYMENT_TARGET")
+    ENV.delete("SDKROOT")
     ENV.m64
 
     cmake_args = [
~~~

CrazyForName, 2018-10-16:
Thank you for your answer.
Because libhand relevant dependencies is ogre1.7 and I just have tried ogre1.9. But when I install libhand, report errors. So I want to try it again by ogre1.7.
But I don't konw how to install your patch, can you tell me
I am very sorry to ask a low-level question, but I really don't know how to install this patch.

scpeters, 2018-10-16:
I committed the patch to a branch and will make a pull request if it fixes your problem. You can test it with the following:

~~~
brew install https://raw.githubusercontent.com/scpeters/homebrew-simulation/a5eced291e1eb621a1a207bf3b3519246c11cf1c/ogre.rb
~~~

CrazyForName, 2018-10-18:
It still report errors. I have installed the software that dependence is ogre1.7.4 in ubuntu16.04。
Thank you for your answer.

closed 2018-10-18 by CrazyForName