osrf/homebrew-simulation #514: Unable to install gazebo

AnaghaSasikumar, 2018-10-17:
This is an error I got when I tried to run gazebo.sh file:

==> Tapping osrf/simulation
Cloning into '/usr/local/Library/Taps/osrf/homebrew-simulation'...
remote: Enumerating objects: 109, done.
remote: Counting objects: 100% (109/109), done.
remote: Compressing objects: 100% (67/67), done.
remote: Total 109 (delta 35), reused 49 (delta 0), pack-reused 0
Receiving objects: 100% (109/109), 46.36 KiB | 0 bytes/s, done.
Resolving deltas: 100% (35/35), done.
Error: Invalid formula: /usr/local/Library/Taps/osrf/homebrew-simulation/gazebo@1.rb
No available formula with the name "gazebo@1" 
In formula file: /usr/local/Library/Taps/osrf/homebrew-simulation/gazebo@1.rb
Expected to find class Gazebo@1, but only found: GazeboAT1.
Error: Invalid formula: /usr/local/Library/Taps/osrf/homebrew-simulation/ignition-math@1.rb
No available formula with the name "ignition-math@1" 
In formula file: /usr/local/Library/Taps/osrf/homebrew-simulation/ignition-math@1.rb
Expected to find class IgnitionMath@1, but only found: IgnitionMathAT1.
Error: Invalid formula: /usr/local/Library/Taps/osrf/homebrew-simulation/ignition-msgs3.rb
unknown version :high_sierra
Error: Invalid formula: /usr/local/Library/Taps/osrf/homebrew-simulation/ignition-physics1.rb
unknown version :mojave
Error: Invalid formula: /usr/local/Library/Taps/osrf/homebrew-simulation/ignition-plugin1.rb
unknown version :high_sierra
Error: Invalid formula: /usr/local/Library/Taps/osrf/homebrew-simulation/ignition-tools.rb
/usr/local/Library/Taps/osrf/homebrew-simulation/ignition-tools.rb:27: syntax error, unexpected tLABEL
      format: 1.0.0
             ^
/usr/local/Library/Taps/osrf/homebrew-simulation/ignition-tools.rb:27: no .<digit> floating literal anymore; put 0 before dot
      format: 1.0.0
                  ^
/usr/local/Library/Taps/osrf/homebrew-simulation/ignition-tools.rb:28: syntax error, unexpected ':', expecting keyword_end
      library_name: test
                   ^
/usr/local/Library/Taps/osrf/homebrew-simulation/ignition-tools.rb:29: syntax error, unexpected ':', expecting keyword_end
      library_path: path
                   ^
/usr/local/Library/Taps/osrf/homebrew-simulation/ignition-tools.rb:30: syntax error, unexpected ':', expecting keyword_end
      library_version: 2.0.0
                      ^
/usr/local/Library/Taps/osrf/homebrew-simulation/ignition-tools.rb:30: no .<digit> floating literal anymore; put 0 before dot
      library_version: 2.0.0
                           ^
/usr/local/Library/Taps/osrf/homebrew-simulation/ignition-tools.rb:31: syntax error, unexpected ':', expecting keyword_end
Error: Invalid formula: /usr/local/Library/Taps/osrf/homebrew-simulation/ignition-transport6.rb
unknown version :high_sierra
Error: Invalid formula: /usr/local/Library/Taps/osrf/homebrew-simulation/sdformat8.rb
unknown version :mojave
Error: Cannot tap osrf/simulation: invalid syntax in tap!
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]

Could you tell me what is going wrong?

scpeters, 2018-10-17:
That is an old version. Can you try `brew install gazebo9`?

AnaghaSasikumar, 2018-10-18:
Ok, will try this...thanks!

AnaghaSasikumar, 2018-10-23:
We actually needed to upgrade our ruby to version 2.3.7 - one of the files actually needed that. It's done now, though, thanks a lot!

closed 2018-10-23 by AnaghaSasikumar