osrf/homebrew-simulation #517: installation error

cooldesigns, 2018-10-24:
Dart is installed via brew. Than:
➜  ~  brew install gazebo9 --build-from-source
==> Installing gazebo9 from osrf/simulation
==> Downloading http://gazebosim.org/distributions/gazebo/releases/gazebo-9.4.1.tar.
Already downloaded: /Users/eva/Library/Caches/Homebrew/downloads/824f01976c997ed9a53a785b169bfc453f0f0b5c73a6341dfc9be061d0fee7e2--gazebo-9.4.1.tar.bz2
==> Downloading https://gist.githubusercontent.com/scpeters/9199370/raw/afe595587e38
Already downloaded: /Users/eva/Library/Caches/Homebrew/downloads/12171d5787bf5709495604c1591fd598a9cffc3a055eb509bea588d33019cabc--brew_python_fix.patch
==> Patching
==> Applying brew_python_fix.patch
patching file deps/opende/CMakeLists.txt
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DC
==> make install
Last 15 lines from /Users/eva/Library/Logs/Homebrew/gazebo9/02.make:
[ 84%] Automatic MOC for target PressurePlugin
cd /tmp/gazebo9-20181024-50520-1yffoo4/gazebo-9.4.1/build/plugins && /usr/local/Cellar/cmake/3.12.3/bin/cmake -E cmake_autogen /tmp/gazebo9-20181024-50520-1yffoo4/gazebo-9.4.1/build/plugins/CMakeFiles/PressurePlugin_autogen.dir/AutogenInfo.cmake Release
[ 84%] Built target BuoyancyPlugin_autogen
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f plugins/CMakeFiles/ContactPlugin_autogen.dir/build.make plugins/CMakeFiles/ContactPlugin_autogen.dir/depend
cd /tmp/gazebo9-20181024-50520-1yffoo4/gazebo-9.4.1/build && /usr/local/Cellar/cmake/3.12.3/bin/cmake -E cmake_depends "Unix Makefiles" /tmp/gazebo9-20181024-50520-1yffoo4/gazebo-9.4.1 /tmp/gazebo9-20181024-50520-1yffoo4/gazebo-9.4.1/plugins /tmp/gazebo9-20181024-50520-1yffoo4/gazebo-9.4.1/build /tmp/gazebo9-20181024-50520-1yffoo4/gazebo-9.4.1/build/plugins /tmp/gazebo9-20181024-50520-1yffoo4/gazebo-9.4.1/build/plugins/CMakeFiles/ContactPlugin_autogen.dir/DependInfo.cmake --color=
Scanning dependencies of target ContactPlugin_autogen
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f plugins/CMakeFiles/ContactPlugin_autogen.dir/build.make plugins/CMakeFiles/ContactPlugin_autogen.dir/build
[ 84%] Automatic MOC for target ContactPlugin
cd /tmp/gazebo9-20181024-50520-1yffoo4/gazebo-9.4.1/build/plugins && /usr/local/Cellar/cmake/3.12.3/bin/cmake -E cmake_autogen /tmp/gazebo9-20181024-50520-1yffoo4/gazebo-9.4.1/build/plugins/CMakeFiles/ContactPlugin_autogen.dir/AutogenInfo.cmake Release
make[1]: *** [gazebo/gui/CMakeFiles/gazebo_gui.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 84%] Built target GimbalSmall2dPlugin_autogen
[ 84%] Built target PressurePlugin_autogen
[ 84%] Built target ContactPlugin_autogen
make: *** [all] Error 2

scpeters, 2018-10-24:
This unfortunately doesn't show the build failure. Can you install with `--verbose` and post the result to a gist? I think `brew gist-logs gazebo9` will work.

cooldesigns, 2018-10-31:
The output is here: https://gist.github.com/cooldesigns/eb0c81ffff94eab437510d383815b63e

cooldesigns, 2018-10-31:
There is the result with --verbose option: https://gist.github.com/a3d8e6090d60e4a31802242e978b251b

cooldesigns, 2018-10-31:
Note: I am installing with dark and dark is installed already via brew.

scpeters, 2018-10-31:
I think this error is similar to #509, an incompatibility with qgv / graphviz. We need to either fix or disable this

cooldesigns, 2018-11-04:
I need graphviz, how i can fix it? Any suggestion can help me.

scpeters, 2018-11-05:
What if you try adding `graphviz` as a dependency:

~~~
diff --git a/gazebo9.rb b/gazebo9.rb
index ac6230c..e257ae1 100644
--- a/gazebo9.rb
+++ b/gazebo9.rb
@@ -20,6 +20,7 @@ class Gazebo9 < Formula
   depends_on "boost"
   depends_on "doxygen"
   depends_on "freeimage"
+  depends_on "graphviz"
   depends_on "ignition-fuel-tools1"
   depends_on "ignition-math4"
   depends_on "ignition-msgs1"
~~~

cooldesigns, 2018-11-05:
I tried reinstalling, the output is here: https://gist.github.com/76da2b53f77fcfe8cd8348a670b301df

PR #521 (2018-11-11): gazebo[7-8]: depend on graphviz, bump revision.
Fixes #517.
commit 6b048a9c (2018-11-11): gazebo[7-8]: depend on graphviz, bump revision. (#521)

* gazebo[7-8]: depend on graphviz, bump revision.

Fixes #517.

* patch for boost 1.68

* gazebo78: update boost 1.68 patch bottle.

* fix audit
closed 2018-11-11 by scpeters