osrf/homebrew-simulation #567: Unable to run Gazebo9 - dartsim library not loaded

dlwalter, 2019-01-17:
When running gazebo9 (9.6.0_1) I get the following error:

```
➜  ~ gazebo
dyld: Library not loaded: /usr/local/opt/dartsim/lib/libdart-utils-urdf.6.6.dylib
  Referenced from: /usr/local/bin/gazebo
  Reason: image not found
[1]    25133 abort      gazebo
```

traversaro, 2019-01-17:
I experienced the same issue, I guess the Gazebo bottle is linking the outdated Dart 6.6, while now Dart 6.7 is available on homebrew. I guess it will be fixed once https://github.com/osrf/homebrew-simulation/pull/566 is merged. 
Building Gazebo from source may solve this, but to be honest I did no try.

scpeters, 2019-01-17:
gazebo9 will be fixed by #563, but #565 is a blocking issue that either needs a fix in homebrew/brew ( https://github.com/Homebrew/brew/pull/5539 ) or upstream in zeromq.

scpeters, 2019-01-18:
new bottles are building for #568 which should resolve this

closed 2019-01-28 by scpeters