osrf/homebrew-simulation #660: ogre2.1 with GL3Plus fails to build on mojave 10.14

scpeters, 2019-03-29:
When trying to build ogre 2.1 with the tarball from #654 but with GL3Plus enabled, it fails to build, even though OpenGL is only deprecated, not removed from Mojave. Here are the build logs:

https://gist.github.com/scpeters/f0896a6ce698fb9258a06ed69a91565a#file-02-make-L661

The error is identical to the one reported in https://github.com/OGRECave/ogre/issues/817, involving the OpenGL framework header `gl3ext.h`. Based on https://github.com/OGRECave/ogre/issues/817#issuecomment-406658115 I suspect that there is a conflict between macros in that system header and the [GL/glext.h](https://bitbucket.org/sinbad/ogre/src/bf5de3029fd5a88d103ff99d7b2f94a5f8396ac9/RenderSystems/GL3Plus/include/GL/glext.h?at=v2-1&fileviewer=file-view-default) included in the GL3Plus source code. Maybe I'll try modifying the Ogre copy of `glext.h` to see if that has any effect.

~~~
Scanning dependencies of target RenderSystem_GL3Plus
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f RenderSystems/GL3Plus/CMakeFiles/RenderSystem_GL3Plus.dir/build.make RenderSystems/GL3Plus/CMakeFiles/RenderSystem_GL3Plus.dir/build
[ 61%] Building CXX object RenderSystems/GL3Plus/CMakeFiles/RenderSystem_GL3Plus.dir/src/GLSL/OgreGLSLExtSupport.cpp.o
cd /tmp/ogre2.1-20190328-74703-ves1zv/sinbad-ogre-bf5de3029fd5/build/RenderSystems/GL3Plus && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++  -DBOOST_THREAD_VERSION=4 -DOGRE_GLPLUGIN_EXPORTS -DRenderSystem_GL3Plus_EXPORTS -I/tmp/ogre2.1-20190328-74703-ves1zv/sinbad-ogre-bf5de3029fd5/RenderSystems/GL3Plus/include -I/tmp/ogre2.1-20190328-74703-ves1zv/sinbad-ogre-bf5de3029fd5/OgreMain/include -I/tmp/ogre2.1-20190328-74703-ves1zv/sinbad-ogre-bf5de3029fd5/build/include -I/tmp/ogre2.1-20190328-74703-ves1zv/sinbad-ogre-bf5de3029fd5/OgreMain/include/OSX -I/usr/local/include/freetype2 -I/tmp/ogre2.1-20190328-74703-ves1zv/sinbad-ogre-bf5de3029fd5 -I/tmp/ogre2.1-20190328-74703-ves1zv/sinbad-ogre-bf5de3029fd5/OgreMain/include/Threading -I/tmp/ogre2.1-20190328-74703-ves1zv/sinbad-ogre-bf5de3029fd5/RenderSystems/GL3Plus/src/windowing/OSX -I/tmp/ogre2.1-20190328-74703-ves1zv/sinbad-ogre-bf5de3029fd5/RenderSystems/GL3Plus/include/windowing/OSX -I/tmp/ogre2.1-20190328-74703-ves1zv/sinbad-ogre-bf5de3029fd5/RenderSystems/GL3Plus/include/GLSL  -Wall -Winit-self -Wno-overloaded-virtual -Wcast-qual -Wwrite-strings -Wextra -Wno-unused-parameter -Wshadow -Wno-missing-field-initializers -Wno-long-long  -msse -msse2 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk -fPIC -fvisibility=hidden -fvisibility-inlines-hidden   -DOGRE_GCC_VISIBILITY -std=c++11 -o CMakeFiles/RenderSystem_GL3Plus.dir/src/GLSL/OgreGLSLExtSupport.cpp.o -c /tmp/ogre2.1-20190328-74703-ves1zv/sinbad-ogre-bf5de3029fd5/RenderSystems/GL3Plus/src/GLSL/OgreGLSLExtSupport.cpp

In file included from /tmp/ogre2.1-20190328-74703-ves1zv/sinbad-ogre-bf5de3029fd5/RenderSystems/GL3Plus/src/GLSL/OgreGLSLExtSupport.cpp:29:
In file included from /tmp/ogre2.1-20190328-74703-ves1zv/sinbad-ogre-bf5de3029fd5/RenderSystems/GL3Plus/include/GLSL/OgreGLSLExtSupport.h:34:
In file included from /tmp/ogre2.1-20190328-74703-ves1zv/sinbad-ogre-bf5de3029fd5/RenderSystems/GL3Plus/include/OgreGL3PlusPrerequisites.h:80:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h:187:77: error: expected function body after function declarator
GLAPI void APIENTRY glVertexAttribDivisorARB (GLuint index, GLuint divisor) OPENGL_DEPRECATED(10.6, 10.14);
                                                                            ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h:249:104: error: expected function body after function declarator
GLAPI void APIENTRY glLabelObjectEXT (GLenum type, GLuint object, GLsizei length, const GLchar *label) OPENGL_DEPRECATED(10.6, 10.14);
                                                                                                       ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h:250:119: error: expected function body after function declarator
GLAPI void APIENTRY glGetObjectLabelEXT (GLenum type, GLuint object, GLsizei bufSize, GLsizei *length, GLchar *label) OPENGL_DEPRECATED(10.6, 10.14);
                                                                                                                      ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h:259:81: error: expected function body after function declarator
GLAPI void APIENTRY glInsertEventMarkerEXT (GLsizei length, const char *marker) OPENGL_DEPRECATED(10.6, 10.14);
                                                                                ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h:260:79: error: expected function body after function declarator
GLAPI void APIENTRY glPushGroupMarkerEXT (GLsizei length, const char *marker) OPENGL_DEPRECATED(10.6, 10.14);
                                                                              ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h:261:48: error: expected function body after function declarator
GLAPI void APIENTRY glPopGroupMarkerEXT (void) OPENGL_DEPRECATED(10.6, 10.14);
                                               ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h:271:69: error: expected function body after function declarator
GLAPI void APIENTRY glDepthBoundsEXT (GLclampd zmin, GLclampd zmax) OPENGL_DEPRECATED(10.6, 10.14);
                                                                    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h:307:47: error: expected function body after function declarator
GLAPI void APIENTRY glFlushRenderAPPLE (void) OPENGL_DEPRECATED(10.6, 10.14);
                                              ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h:308:48: error: expected function body after function declarator
GLAPI void APIENTRY glFinishRenderAPPLE (void) OPENGL_DEPRECATED(10.6, 10.14);
                                               ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h:309:40: error: expected function body after function declarator
GLAPI void APIENTRY glSwapAPPLE (void) OPENGL_DEPRECATED(10.6, 10.14);
                                       ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h:319:94: error: expected function body after function declarator
GLAPI GLenum APIENTRY glObjectPurgeableAPPLE (GLenum objectType, GLuint name, GLenum option) OPENGL_DEPRECATED(10.6, 10.14);
                                                                                             ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h:320:96: error: expected function body after function declarator
GLAPI GLenum APIENTRY glObjectUnpurgeableAPPLE (GLenum objectType, GLuint name, GLenum option) OPENGL_DEPRECATED(10.6, 10.14);
                                                                                               ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h:321:111: error: expected function body after function declarator
[... cut ...]
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl3ext.h:353:47: error: expected function body after function declarator
GLAPI void APIENTRY glTextureBarrierNV (void) OPENGL_DEPRECATED(10.6, 10.14);
                                              ^
16 errors generated.
make[2]: *** [RenderSystems/GL3Plus/CMakeFiles/RenderSystem_GL3Plus.dir/src/GLSL/OgreGLSLExtSupport.cpp.o] Error 1
make[1]: *** [RenderSystems/GL3Plus/CMakeFiles/RenderSystem_GL3Plus.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

~~~

cc @iche033 @wjwwood @j-rivero

iche033, 2019-03-29:
Maybe the `OPENGL_DEPRECATED` macro is not found? Another thing to try is to add the `<OpenGL/OpenGLAvailability.h>` header before `<OpenGL/gl3ext.h>`

scpeters, 2019-03-29:
That's it!!!! It worked. I'll submit a pull request upstream and add the patch to this formula and build a bottle

scpeters, 2019-03-29:
submitted https://bitbucket.org/sinbad/ogre/pull-requests/857/include-openglavailabilityh-before-gl3exth/diff

updated #654 with patch and started a bottle build

scpeters, 2019-03-30:
fixed by #654

closed 2019-03-30 by scpeters