osrf/homebrew-simulation #747: Unable to build Gazebo on macOS Mojave

Carpetfizz, 2019-05-28:
OS: macOS Mojave 10.14.15

[logs](https://gist.github.com/Carpetfizz/8b7a54123d5565a461918c4ad369f51e)

scpeters, 2019-05-28:
As I said in #656, gazebo8 is now end-of-life, so we aren't proactively fixing issues like this when it fails to build. I would be happy to review a pull request that fixes the build if you can identify a fix. Looking at the build log, I'm wondering if you have `qt@4` installed? I think you need to remove or `brew unlink qt@4` in order to build successfully against `qt5`.

Carpetfizz, 2019-05-28:
Thanks, will give the unlink a shot and update here if it worked. Just curious, what does end-of-life mean? Is the entire Gazebo project dead/unsupported, or is there a new version of Gazebo? I just followed the installation instructions on the website.

scpeters, 2019-05-28:
It just means gazebo8 is getting old; I recommend `brew install gazebo9` or `brew install gazebo10`. I'll update the tutorials to reflect this; thanks for the heads-up!

Carpetfizz, 2019-05-28:
Got it thanks! So I was actually able to get it to work after your recommendation! Is there an easy way for me to update gazebo8 to gazebo10?

scpeters, 2019-05-29:
they are separate formula that can't be linked at the same time, but you can `brew unlink gazebo8; brew install gazebo10`, which should install gazebo10 from a bottle instead of compiling it from source.

Carpetfizz, 2019-05-29:
Thanks again for the suggestion, upon doing that it installs fine, but I get the following error when I try to run `gazebo`

```
dyld: Library not loaded: /usr/local/opt/dartsim/lib/libdart-collision-bullet.6.8.dylib
  Referenced from: /usr/local/bin/gazebo
  Reason: image not found
Abort trap: 6
```

scpeters, 2019-05-29:
Ok, that is an unrelated issue that the bottles need to be rebuilt. I'm working on that in #749

scpeters, 2019-05-29:
The gazebo9 bottle should be fixed right now, but gazebo10 requires an extra patch:

* https://bitbucket.org/osrf/gazebo/pull-requests/3106/fix-gazebo10-build-with-dart-69/diff

Carpetfizz, 2019-05-29:
Thanks! I just ended up installing gazebo9 and it seems to work fine. Will keep an eye on the linked PR so I can install gazebo10 when it's fixed.

Really appreciate the help as a first time user!

closed 2019-05-29 by Carpetfizz

scpeters, 2019-05-29:
:tada:

scpeters, 2019-05-29:
gazebo10 bottle should be fixed now that #750 is merged

Carpetfizz, 2019-05-29:
Perfect, that worked great. Thank you so much! Looking forward to learning how to use it.