osrf/homebrew-simulation #756: Gazebo 7.15.0 fails to launch due to broken sdformat4 link to boost

fabianschilling, 2019-06-14:
After installing the `gazebo7` bottle on macOS 10.14.5, `sdformat4` seems to link to a `boost` symbol which does not exists:

```
~ $ gazebo

dyld: lazy symbol binding failed: Symbol not found: __ZN5boost16re_detail_10690027cpp_regex_traits_char_layerIcE4initEv
  Referenced from: /usr/local/opt/sdformat4/lib/libsdformat.4.dylib
  Expected in: /usr/local/opt/boost/lib/libboost_regex-mt.dylib

dyld: Symbol not found: __ZN5boost16re_detail_10690027cpp_regex_traits_char_layerIcE4initEv
  Referenced from: /usr/local/opt/sdformat4/lib/libsdformat.4.dylib
  Expected in: /usr/local/opt/boost/lib/libboost_regex-mt.dylib
```

It seems like a [related issue](https://github.com/osrf/homebrew-simulation/issues/272) was fixed by re-building the bottles with an updated boost version.

Thanks a lot in advance!

scpeters, 2019-06-14:
Thanks for the report. boost was recently updated, which typically requires a bottle rebuild, but `brew test gazebo*` didn't fail, which I thought was weird. For gazebo 9/10, you can even run gzserver, but a colleague just reported that gzclient was failing. The gazebo7 case is slightly different because sdformat4 links against more boost libraries and it fails even with gzserver. I've updated the `test do` blocks in #759 for gazebo7 and sdformat4 and am rebuilding bottles. gazebo9 and gazebo10 have new tests as well in #757 and #758.

scpeters, 2019-06-14:
Closed by #759.

closed 2019-06-14 by scpeters