osrf/homebrew-simulation #850: Gazebo is broken because of new libprotobuf release

pschatzmann, 2019-10-07:
I have installed gazebo with 
brew install gazebo9  or
brew install gazebo10

Starting gazebo gives the following errors:
dyld: Library not loaded: /usr/local/opt/protobuf/lib/libprotobuf.20.dylib
  Referenced from: /usr/local/bin/gazebo
  Reason: image not found
Abort trap: 6

Please note that libprotobuf has been upgraded to the release 3.10.0! but we have /usr/local/lib/libprotobuf.21.dylib

Unfortunatly homebrew also does not provide the old protobuf version any more.

violapaul, 2019-10-08:
Same here.  

Note, even with a --build-from-sources it does not work.  See attached.  It fails when it gets to code that was autogenerated with protobuf (most likely a previous version).
[brewoutput.txt](https://github.com/osrf/homebrew-simulation/files/3704466/brewoutput.txt)

[~]$ brew config
HOMEBREW_VERSION: 2.1.13
ORIGIN: https://github.com/Homebrew/brew
HEAD: aa043c1d678afd26f50156a83417c296cb9d5a4b
Last commit: 2 hours ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: a035f9f7cbc4de3353b703a250280d5e2e78b064
Core tap last commit: 13 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_DEV_CMD_RUN: 1
HOMEBREW_GITHUB_API_TOKEN: set
CPU: octa-core 64-bit haswell
Homebrew Ruby: 2.3.7 => /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.7/bin/ruby
Clang: 10.0 build 1000
Git: 2.17.2 => /Library/Developer/CommandLineTools/usr/bin/git
Curl: 7.54.0 => /usr/bin/curl
Java: 11.0.4, 9.0.4
macOS: 10.13.6-x86_64
CLT: 10.1.0.0.1.1539992718
Xcode: N/A
XQuartz: 2.7.11 => /opt/X11

scpeters, 2019-10-09:
I already rebuilt ign-msgs2 and ign-msgs4 and associated dependents in #845 and #848 

I made #847 and #849 to update gazebo7 and gazebo9/10, but the builds failed due to an issue with boost:

* [image] https://build.osrfoundation.org/job/generic-release-homebrew_bottle_builder/481/label=osx_highsierra/console
* [image] https://build.osrfoundation.org/job/generic-release-homebrew_bottle_builder/483/label=osx_highsierra/console

I haven't had a chance to figure out what's wrong yet

scpeters, 2019-10-09:
it looks like the bottles built correctly, but there was an issue with the pkg-config files which caused `brew test gazebo*` to fail

scpeters, 2019-10-09:
Yeah, the generated pkg-config file has cmake targets: `-lBoost::thread -lBoost::system -lBoost::filesystem -lBoost::program_options -lBoost::regex -lBoost::iostreams -lBoost::date_time`

I'll disable the pkg-config part of the test and rebuilt the bottles until we can figure out how to fix the pkg-config file

scpeters, 2019-10-09:
the bottles should be fixed by now, with the exception of the pkg-config file

closed 2019-10-14 by scpeters