osrf/homebrew-simulation #1003: Version update for bullet in gazebo

ad1t7a, 2020-04-29:
dyld: Library not loaded: /usr/local/opt/bullet/lib/libBulletSoftBody.2.88.dylib
  Referenced from: /usr/local/bin/gazebo
  Reason: image not found

The system installed version for Bullet is 2.89

Jaeyoung-Lim, 2020-04-29:
PX4 gazebo toolchain also failing with the same error: https://travis-ci.org/github/PX4/sitl_gazebo/jobs/680886246

Jaeyoung-Lim, 2020-04-30:
@scpeters @chapulina Any ideas?

LorenzMeier, 2020-05-04:
@scpeters This is indeed broken for good.

scpeters, 2020-05-04:
I need to rebuild the gazebo bottles to account for the bullet update, but there is also a problem with ogre (#1002), which demonstrated to me that there have been quite a few changes in homebrew infrastructure recently that has meant it isn't a simple matter of queuing a new build like it usually is.

I will try to fix it this week, but it's a bit tricky.

sarahallen111, 2020-05-08:
I am having this issue as well... Tried my best to install bullet 2.88, but it looks like there's only versions 2.89 and 2.87 (through osrf/simulation/bullet@2.87) available on brew right now.

kainism, 2020-05-09:
Seem like the same error I met when running " make px4_sitl gazebo " on macOS Catalina 10.15.4.  Still getting to figure it out, anyone kindly has some solutions ?

kevin@kevinJdeMacBook-Pro Firmware % make px4_sitl gazebo
[0/4] Performing build step for 'sitl_gazebo'
[1/1] Generating /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf
FAILED: /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf 
cd /Users/kevin/Firmware/Tools/sitl_gazebo && rm -f /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf && /usr/local/bin/python3 /Users/kevin/Firmware/Tools/sitl_gazebo/scripts/xacro.py -o /Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf /Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.xacro enable_mavlink_interface:=true enable_ground_truth:=false enable_wind:=false enable_logging:=false rotors_description_dir:=/Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description send_vision_estimation:=true send_odometry:=false && gz sdf -p /Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf >> /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf && rm -f /Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf

dyld: Library not loaded: /usr/local/opt/bullet/lib/libBulletSoftBody.2.88.dylib
  Referenced from: /usr/local/bin/gz
  Reason: image not found
/bin/sh: line 1: 75689 Abort trap: 6           gz sdf -p /Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf >> /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf
ninja: build stopped: subcommand failed.
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /Users/kevin/Firmware/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.17.2/bin/cmake --build .
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1

ryanjAA, 2020-05-09:
Same error as @kainism line for line on a fresh 10.15.4 Mac install as well.

scpeters, 2020-05-11:
I managed to rebuild many bottles broken by a console_bridge update (see #1005) using a back-dated branch of homebrew-test-bot. I also removed the bottle block from the gazebo11 formula in https://github.com/osrf/homebrew-simulation/commit/c8b4288a45dac50f2822aad2ac46576992f2a667, so it now always attempts to build from source.

Unfortunately, there are still some problems:

* [image] https://build.osrfoundation.org/job/gazebo-install-one_liner-homebrew-amd64/1696/

~~~
+ brew test gazebo11
Testing osrf/simulation/gazebo11
==> /usr/local/Cellar/gazebo11/11.0.0_2/bin/gz sdf
Last 15 lines from /Users/jenkins/Library/Logs/Homebrew/gazebo11/test.01.gz:
2020-05-11 14:28:51 +0200

/usr/local/Cellar/gazebo11/11.0.0_2/bin/gz
sdf

dyld: Symbol not found: __ZN6gazebo3gui10DataLogger16staticMetaObjectE
  Referenced from: /usr/local/lib/libgazebo_gui.11.dylib
  Expected in: flat namespace
 in /usr/local/lib/libgazebo_gui.11.dylib
Error: osrf/simulation/gazebo11: failed
~~~

krmcelhinney, 2020-05-12:
I have built from source and get the same error - I have Catalina 10.15.5. ( 0reference to libgazebo_gui.11.dylib)

scpeters, 2020-05-12:
I just looked at the `qt` installation and it appears to only have static libraries now? I need to look into recent changes in the `qt` formula to see if they don't want to provide shared libraries anymore.

scpeters, 2020-05-13:
actually, it still has dynamic libraries in the frameworks. I'm still investigating the `staticMetaObject` error

scpeters, 2020-05-13:
I just built gazebo11 from source manually using cmake, and it doesn't seem to have the qt symbol resolution problems. Perhaps something changed in the homebrew build environment that makes `brew install gazebo11` behave differently?

LorenzMeier, 2020-05-26:
Friendly ping - I tried to get Gazebo 10 or 11 up as alternative, but no luck.

scpeters, 2020-05-26:
I haven't gotten past the error  in https://github.com/osrf/homebrew-simulation/issues/1003#issuecomment-626865911

I can build gazebo manually, but if I `brew install gazeboX --build-from-source` it is missing that `__ZN6gazebo3gui10DataLogger16staticMetaObjectE` symbol

j-rivero, 2020-06-08:
for the symbol problem, let's follow https://github.com/osrf/homebrew-simulation/issues/1011

closed 2020-06-08 by j-rivero