osrf/homebrew-simulation #1005: 👩‍🌾 Broken console_bridge

chapulina, 2020-05-08:
There are hundreds of new test failures across the Ignition stack with runtime errors like:

```
44: dyld: Library not loaded: /usr/local/opt/console_bridge/lib/libconsole_bridge.0.4.dylib
44:   Referenced from: /usr/local/opt/dartsim@6.10.0/lib/libdart-utils-urdf.6.10.dylib
```

```
59: dyld: Library not loaded: /usr/local/opt/console_bridge/lib/libconsole_bridge.0.4.dylib
59:   Referenced from: /usr/local/opt/sdformat9/lib/libsdformat9.9.dylib
```

This was also mentioned here: https://github.com/ignitionrobotics/ign-physics/pull/55#issuecomment-625973407

@scpeters , let me know if there's anything I can help with.

scpeters, 2020-05-08:
it looks like we need to rebuild some bottles

scpeters, 2020-05-08:
I started looking at rebuilding bottles last night; I'll start again now

* my first step was to create a branch of `homebrew-test-bot` from before many of the recent breaking changes, currently pointing at https://github.com/Homebrew/homebrew-test-bot/commit/b7ceb2665708850000a89dbe8abd794417d76dc9: https://github.com/scpeters/homebrew-test-bot/tree/still_working
* using an old branch of test-bot requires a slight change in release-tools: https://github.com/ignition-tooling/release-tools/commit/ae7b8873e67a973d39d25e1eee96a0bfd78fbec7

using those two branches wasn't enough to rebuild ogre1.9 in #1002, which is what I was investigating last night, though they may be sufficient to rebuild dartsim@6.10; I'm going to try testing that now on my local mojave machine

scpeters, 2020-05-09:
To test on my laptop running mojave; I installed `ignition-blueprint` and `ignition-citadel` and then ran `brew linkage --test | grep -v "No broken"`, which lists the packages that have broken linkage. It indicated that `dartsim@6.10.0`, `sdformat8`, `sdformat9`, `ignition-physics[12]` had broken linkage. So I made a branch and ran `brew bump-revision $formula_name --message="for console_bridge"` for each of those formulae. I'll make a pull request for that branch and try building bottles again.

scpeters, 2020-05-09:
just to be safe, I removed the bottle block from gazebo11 in https://github.com/osrf/homebrew-simulation/commit/c8b4288a45dac50f2822aad2ac46576992f2a667, since it uses `sdformat9`, and having a dependent with a bottle block means it will be tested, so this should cause it to ignore `gazebo11` for now

scpeters, 2020-05-09:
just made https://github.com/osrf/homebrew-simulation/pull/1006 and will start a bottle build job

scpeters, 2020-05-09:
the bottles miraculously rebuilt and are merged, so this might be fixed

closed 2020-05-10 by scpeters

scpeters, 2020-05-10:
I submitted https://github.com/ignition-tooling/release-tools/pull/199 for that release-tools branch