osrf/homebrew-simulation #1011: Gazebo11 runtime: dyld: Symbol not found

ronwalf, 2020-05-15:
A fresh install of Homebrew and Gazebo11 is giving a dyld error on startup on Catalina 10.15.4. Any hints on tracking it down?

```
~ % brew install gazebo11
==> Installing gazebo11 from osrf/simulation
Warning: A newer Command Line Tools release is available.
Update them from Software Update in System Preferences or run:
  softwareupdate --all --install --force

If that doesn't show you an update run:
  sudo rm -rf /Library/Developer/CommandLineTools
  sudo xcode-select --install

Alternatively, manually download them from:
  https://developer.apple.com/download/more/.

==> Downloading https://osrf-distributions.s3.amazonaws.com/gazebo/releases/gazebo-11.0.0.tar.bz2
Already downloaded: /Users/ralford/Library/Caches/Homebrew/downloads/fa71b7739ef87205dc24a9f69686425ee3fd208d38a14c83f2245cc2b979250d--gazebo-11.0.0.tar.bz2
==> Downloading https://gist.githubusercontent.com/scpeters/9199370/raw/afe595587e38737c537124a3652db99de026c272/brew_python_fix.patch
Already downloaded: /Users/ralford/Library/Caches/Homebrew/downloads/12171d5787bf5709495604c1591fd598a9cffc3a055eb509bea588d33019cabc--brew_python_fix.patch
==> Patching
==> Applying brew_python_fix.patch
patching file deps/opende/CMakeLists.txt
==> cmake .. -DQWT_WIN_INCLUDE_DIR=/usr/local/lib/qwt.framework/Headers -DQWT_WIN_LIBRARY_DIR=/usr/local/lib/qwt.framework
==> make install
Warning: osrf/simulation/gazebo11 dependency gcc was built with a different C++ standard
library (libstdc++ from clang). This may cause problems at runtime.
🍺  /usr/local/Cellar/gazebo11/11.0.0_2: 1,272 files, 122.5MB, built in 10 minutes 50 seconds
~ % gazebo
dyld: Symbol not found: __ZN6gazebo3gui10DataLogger16staticMetaObjectE
  Referenced from: /usr/local/lib/libgazebo_gui.11.dylib
  Expected in: flat namespace
 in /usr/local/lib/libgazebo_gui.11.dylib
zsh: abort      gazebo
```

Some additional system info from brew doctor and config:
```
~ % brew doctor
Please note that these warnings are just used to help the Homebrew maintainers
with debugging if you file an issue. If everything you use Homebrew for is
working fine: please don't worry or file an issue; just ignore this. Thanks!

Warning: A newer Command Line Tools release is available.
Update them from Software Update in System Preferences or run:
  softwareupdate --all --install --force

If that doesn't show you an update run:
  sudo rm -rf /Library/Developer/CommandLineTools
  sudo xcode-select --install

Alternatively, manually download them from:
  https://developer.apple.com/download/more/.

~ % brew config
HOMEBREW_VERSION: 2.2.16
ORIGIN: https://github.com/Homebrew/brew
HEAD: c0f5a52d85d3606e7600162a9d38936ab4b46f4f
Last commit: 5 days ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: cd6493288b22113fdbbcd5b1766a9b76668d7281
Core tap last commit: 4 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_DISPLAY: :0.0
HOMEBREW_EDITOR: vim
HOMEBREW_MAKE_JOBS: 12
CPU: dodeca-core 64-bit kabylake
Homebrew Ruby: 2.6.3 => /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby
Clang: 11.0 build 1103
Git: 2.24.2 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 7.64.1 => /usr/bin/curl
macOS: 10.15.4-x86_64
CLT: 11.0.33.17
Xcode: 11.4.1
XQuartz: 2.7.11 => /opt/X11
```

scpeters, 2020-05-15:
yeah, I'm seeing the same thing ( https://github.com/osrf/homebrew-simulation/issues/1003#issuecomment-626865911 ). `gzserver` seems to work, but both `gz` and `gzclient` link to `qt` and have that symbol resolution error. the weird thing is that if I build gazebo from source using cmake without brew, it works fine. so I'm a bit perplexed by that

Ellipsoul, 2020-05-16:
Hi, are there instructions available on how to build gezebo from source for OS Catalina? I am also having the same issue with running Gazebo. Thanks

scpeters, 2020-05-18:
you can install the gazebo11 dependencies with `brew install gazebo11 --only-dependencies`

and then follow the build from source instructions here:

* https://github.com/osrf/gazebo_tutorials/blob/master/install_from_source/tutorial_default.md#build-and-install-gazebo

j-rivero, 2020-06-08:
Let me summarize the current situation (correct me if I'm wrong): gazebo10, gazebo11 bottles and from source builds using brew are ending up in loading problems for the `__ZN6gazebo3gui10DataLogger16staticMetaObjectE` in the binaries that use/link qt capabilities.

The current workaround involve to install dependencies using Brew and build Gazebo from source outside brew mechanisms, using cmake as described @scpeters in https://github.com/osrf/homebrew-simulation/issues/1011#issuecomment-630336068

zmk5, 2020-06-25:
I have this same issue too but I'm not sure how to build it from source without brew.

fnazia, 2020-07-07:
> you can install the gazebo11 dependencies with `brew install gazebo11 --only-dependencies`
> 
> and then follow the build from source instructions here:
> 
> * https://github.com/osrf/gazebo_tutorials/blob/master/install_from_source/tutorial_default.md#build-and-install-gazebo

Hello, I followed the procedure; however I am getting following problem when trying to run "gazebo" -

> dyld: Library not loaded: @rpath/libQt5Widgets.5.dylib
>   Referenced from: /usr/local/bin/gazebo
>   Reason: image not found
> zsh: abort      gazebo

Is there any way to get around this issue?

supergra, 2020-07-07:
Concur with @fnazia -- I completely removed the previously built gazebo executables, called `rew install gazebo11 --only-dependencies`, built from scratch according the instructions given, and I still get the same error.

fnazia, 2020-07-08:
I found a solution to the library-not-found problem. Please refer to this link -
[https://github.com/yegord/snowman/issues/199#issuecomment-635766442](url)

bromeara, 2020-07-11:
@fnazia The link seems to be broken

fnazia, 2020-07-11:
> @fnazia The link seems to be broken

Could you try again? I think it works now.

bromeara, 2020-07-11:
That fixed the not found problem from me. Although building from source still gives the error.
```
dyld: Symbol not found: __ZN6gazebo3gui10DataLogger16staticMetaObjectE
  Referenced from: /Users/braedon/opt/lib/libgazebo_gui.11.dylib
  Expected in: flat namespace
 in /Users/braedon/opt/lib/libgazebo_gui.11.dylib
zsh: abort      gazebo
```

j-rivero, 2020-08-14:
I gave this a try using the latest gazebo11 11.1.0 version released this week. Seems to start fine for me using:
```
brew install --build-from-source gazebo11
```

sw_vers:
```
ProductName:    Mac OS X
ProductVersion: 10.15.4
BuildVersion:   19E266
```

Brew config:
```
HOMEBREW_VERSION: 2.4.11-86-g535e7dc
ORIGIN: https://github.com/Homebrew/brew
HEAD: 535e7dc836641ab15a97df3a48a49f81d45ceff5
Last commit: 3 hours ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 1b1a88973d664a22e3425a73f89d8333047899a8
Core tap last commit: 2 hours ago
Core tap branch: master
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CASK_OPTS: []
HOMEBREW_MAKE_JOBS: 8
CPU: octa-core 64-bit ivybridge
Homebrew Ruby: 2.6.3 => /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3_2/bin/ruby
Clang: 11.0 build 1103
Git: 2.28.0 => /usr/local/bin/git
Curl: 7.64.1 => /usr/bin/curl
Java: 1.8.0_131, 1.8.0_45
macOS: 10.15.4-x86_64
CLT: 11.5.0.0.1.1588476445
Xcode: N/A
XQuartz: 2.7.11 => /opt/X11
```

<details>
 <summary>List of packages installed</summary>
aom 2.0.0
assimp 5.0.1
atk 2.36.0
boost 1.73.0
bullet 2.89
cairo 1.16.0_3
cmake 3.18.1
console_bridge 1.0.1
cppzmq 0.0.0-brew
dartsim 6.9.2_8
dav1d 0.7.1
doxygen 1.8.19
eigen 3.3.7
fcl 0.6.1
ffmpeg 4.3.1
flac 1.3.3
flann 1.9.1_9
fontconfig 2.13.1
freeimage 3.18.0
freetype 2.10.2
frei0r 1.7.0
fribidi 1.0.10
gazebo11 11.1.0
gcc 10.2.0
gd 2.3.0
gdbm 1.18.1_1
gdk-pixbuf 2.40.0_1
gettext 0.21
giflib 5.2.1
git 2.28.0
glib 2.64.4_2
gmp 6.2.0
gnutls 3.6.14
gobject-introspection 1.64.1_2
graphite2 1.3.14
graphviz 2.44.1
gtk+ 2.24.32_3
gtkglext 1.2.0_3
gts 0.7.6_2
harfbuzz 2.7.1
hdf5 1.12.0_1
hicolor-icon-theme 0.17
icu4c 67.1
ignition-cmake2 2.3.0
ignition-common3 3.6.0
ignition-fuel-tools4 4.1.0_3
ignition-math6 6.4.0
ignition-msgs5 5.3.0_3
ignition-tools 1.0.0
ignition-transport8 8.0.0_1
ipopt 3.12.13_9
isl 0.22.1
jasper 2.0.19
jpeg 9d
jpeg-turbo 2.0.5
jsoncpp 1.9.3
lame 3.100
leptonica 1.80.0
libass 0.14.0_1
libbluray 1.2.0
libccd 2.1
libevent 2.1.12
libffi 3.3
libidn2 2.3.0
libmpc 1.1.0
libogg 1.3.4
libpng 1.6.37
libsamplerate 0.1.9_1
libsndfile 1.0.29pre2_1
libsoxr 0.1.3
libtar 1.2.20
libtasn1 4.16.0
libtiff 4.1.0
libtool 2.4.6_2
libunistring 0.9.10
libvidstab 1.1.0
libvorbis 1.3.7
libvpx 1.9.0
libyaml 0.2.5
libzip 1.7.3
libzzip 0.13.71
little-cms2 2.11
lzo 2.10
mercurial 5.5
mpfr 4.1.0
netpbm 10.86.15
nettle 3.6
nlopt 2.6.2
octomap 1.9.5
ode 0.16.2
ogre1.9 1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8
open-scene-graph 3.6.5
openblas 0.3.10_1
opencore-amr 0.1.5
openjpeg 2.3.1
openssl@1.1 1.1.1g
opus 1.3.1
ossp-uuid 1.6.2_2
p11-kit 0.23.20_1
pango 1.46.0
pcre 8.44
pcre2 10.35
pixman 0.40.0
pkg-config 0.29.2_3
protobuf 3.12.4
protobuf-c 1.3.3_1
python@3.8 3.8.5
qt 5.15.0
qwt 6.1.5
rav1e 0.3.3
readline 8.0.4
rtmpdump 2.4+20151223_1
rubberband 1.8.2_1
sdformat9 9.2.0_3
sdl 1.2.15_1
sdl2 2.0.12_1
simbody 3.7
snappy 1.1.8
speex 1.2.0
sqlite 3.32.3
srt 1.4.1
szip 2.1.1_1
tbb 2020_U2
tesseract 4.1.1
theora 1.1.1
tinyxml 2.6.2
tinyxml2 8.0.0
unbound 1.11.0
urdfdom 1.0.4_2
urdfdom_headers 1.0.5
webp 1.1.0
x264 r3011
x265 3.4
xvid 1.3.7
xz 5.2.5
zeromq 4.3.2
</details>

Could someone confirm is the problem is still present in the current conditions?

happydasch, 2020-08-15:
tried it yesterday and it compiled successfully for me with:

`brew install --build-from-source gazebo11`

starts as expected and no more missing symbol error

zmk5, 2020-08-19:
Building is successful, but I keep getting this error when I run Gazebo with `gazebo`:

```bash
~$ gazebo
dyld: Library not loaded: @rpath/libgazebo.11.dylib
  Referenced from: /usr/local/bin/gazebo
  Reason: image not found
Abort trap: 6
```

This is my `brew config`
```
HOMEBREW_VERSION: 2.4.12-202-gdd8e6f4
ORIGIN: https://github.com/Homebrew/brew
HEAD: dd8e6f42a2f47ca99bbafb226a7566ff59cf7930
Last commit: 2 hours ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: edc2cb20d446a647331b479f68b08202e7398692
Core tap last commit: 67 minutes ago
Core tap branch: master
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.TkUkpsHbza/org.macosforge.xquartz:0
HOMEBREW_MAKE_JOBS: 12
CPU: dodeca-core 64-bit kabylake
Homebrew Ruby: 2.6.3 => /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby
Clang: 11.0 build 1103
Git: 2.28.0 => /usr/local/bin/git
Curl: 7.64.1 => /usr/bin/curl
Java: 11.0.2
macOS: 10.15.6-x86_64
CLT: 11.0.33.17
Xcode: 11.6
XQuartz: 2.7.11 => /opt/X11
```

j-rivero, 2020-09-15:
> `brew install --build-from-source gazebo11`
> 
> starts as expected and no more missing symbol error

We can not reproduce the error of staticMetaObject anymore. It might be solved by an update of qt5 version in brew, if someone has time would be nice to confirm with the changelog what was the real cause. Closing the issue by now, please comment in the staticMetaObject problem appears again.

> ```shell
> ~$ gazebo
> dyld: Library not loaded: @rpath/libgazebo.11.dylib
>   Referenced from: /usr/local/bin/gazebo
>   Reason: image not found
> Abort trap: 6
> ```

This seems like a different problem, please open a new issue if you have not solve it yet.

closed 2020-09-15 by j-rivero

scpeters, 2020-09-28:
The `dyld: Symbol not found: __ZN6gazebo3gui10DataLogger16staticMetaObjectE` error seemed to disappear after #1094, which is why this issue is closed, but I'm investigating it a little further just to know what caused it. I'll comment here with my findings.

scpeters, 2020-09-28:
I've made a branch to reproduce this error that reverts `gazebo11` to 11.0.0 and adds some patches to fix boost compilation errors (https://github.com/scpeters/homebrew-simulation/commit/70bd9a6d3c88e3faf8630b4052e5c71bda74e900)

scpeters, 2020-09-29:
I just noticed gazebo issue https://github.com/osrf/gazebo/issues/2702, which seems very closely related. It was fixed by https://github.com/osrf/gazebo/commit/2e4f1b185cb46b2c06fe63881d60f3fbeb868923, so I'm going to test this patch next.

scpeters, 2020-09-29:
adding the `CMP0100` patch in https://github.com/scpeters/homebrew-simulation/commit/faab9314c9a20763b79d07b1558e5b9f2af589fe resolves the problem

r4881t, 2021-11-06:
> Building is successful, but I keep getting this error when I run Gazebo with `gazebo`:
> 
> ```shell
> ~$ gazebo
> dyld: Library not loaded: @rpath/libgazebo.11.dylib
>   Referenced from: /usr/local/bin/gazebo
>   Reason: image not found
> Abort trap: 6
> ```
> 
> ```

Did you found out the cause for this? i am facing same issue

r4881t, 2021-11-07:
> @zmk5
> 
> > Building is successful, but I keep getting this error when I run Gazebo with `gazebo`:
> > ```shell
> > ~$ gazebo
> > dyld: Library not loaded: @rpath/libgazebo.11.dylib
> >   Referenced from: /usr/local/bin/gazebo
> >   Reason: image not found
> > Abort trap: 6
> > ```
> 
> Did you found out the cause for this? i am facing same issue

For anyone who stumbles upon this, follow the following to install gazebo on mac (M1 OS X Monterey 12.1)

1. Open your terminal in Rosseta mode.
2. Install brew in Rosseta. This is the key bit here. You may already have brew installed in non-rosseta mode, so check your brew by `which brew`. If it is located at /opt/homebrew/bin/brew then it's non-rosseta, and if it's located at /usr/local/bin/brew then it is rosseta one.
3. Using the rosseta brew, install gazebo.

It will work.

whwang88, 2021-11-15:
> For anyone who stumbles upon this, follow the following to install gazebo on mac (M1 OS X Monterey 12.1)
>
> 1. Open your terminal in Rosseta mode.
> 2. Install brew in Rosseta. This is the key bit here. [...]

For this solution, did you have to uninstall Homebrew and then reinstall it using Rosetta?

r4881t, 2021-11-15:
@whwang88 - I did not uninstall the non-rosseta homwbrew. Since they are at different paths, it shouldn't matter.

zarathustr, 2022-10-20:
There is an update regarding this error:

1. Simply adding cmake rules for gazebo 9.0 is helpful for a successful build but not helpful for running. On macOS 10.15, 11 and 12, there may be errors on missing symbols of the dyld.

2. I propose the following modifications to make the gazebo run smoothly:

1) In  ```gazebo/gui/CMakeLists.txt```,  change
```
set (internal_qt_headers
  VideoRecorder.hh)
```
to
```
set (internal_qt_headers
  VideoRecorder.hh DataLogger.hh)
```
which will let the program find the ```staticMetaObject``` in the ```libgazebo.dylib```.

2) In ```gazebo/common/CMakeLists.txt```, add the following lines:
```
if(APPLE)
  target_link_libraries(gazebo_common
    /System/Library/Frameworks/CoreMedia.framework/Versions/A/CoreMedia
    /System/Library/Frameworks/VideoToolbox.framework/Versions/A/VideoToolbox)
endif()
```
This will prevent the program not finding out the symbols in ```CoreMedia``` and ```VidelToolbox``` frameworks.