osrf/homebrew-simulation #1020: #error regenerate this file with a newer version of protoc

lingjiekong, 2020-05-21:
Build Gazebo11 from source by `brew install gazebo11 --build-from-source` due to Gazebo11 runtime: dyld: Symbol not found #1011 

However, it looks like protobuf is not compatible. I issued `protoc --version` and get libprotoc 3.12.0. Therefore, want to know how to fix the below issue.

patching file deps/opende/CMakeLists.txt
==> cmake .. -DQWT_WIN_INCLUDE_DIR=/usr/local/lib/qwt.framework/Headers -DQWT_WIN_LIBRARY_DIR=/usr/local/lib/qwt.framework
==> make install
Last 15 lines from /Users/lingjik/Library/Logs/Homebrew/gazebo11/02.make:
#error regenerate this file with a newer version of protoc.
 ^
/usr/local/include/ignition/msgs5/ignition/msgs/material.pb.h:300:38: error: no type named 'InternalMetadataWithArena' in namespace 'google::protobuf::internal'
  ::PROTOBUF_NAMESPACE_ID::internal::InternalMetadataWithArena _internal_metadata_;
  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/usr/local/include/ignition/msgs5/ignition/msgs/material.pb.h:644:38: error: no type named 'InternalMetadataWithArena' in namespace 'google::protobuf::internal'
  ::PROTOBUF_NAMESPACE_ID::internal::InternalMetadataWithArena _internal_metadata_;
  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/usr/local/include/ignition/msgs5/ignition/msgs/material.pb.h:960:38: error: no type named 'InternalMetadataWithArena' in namespace 'google::protobuf::internal'
  ::PROTOBUF_NAMESPACE_ID::internal::InternalMetadataWithArena _internal_metadata_;
  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
19 errors generated.
make[2]: *** [gazebo/msgs/CMakeFiles/gazebo_msgs.dir/msgs.cc.o] Error 1
make[1]: *** [gazebo/msgs/CMakeFiles/gazebo_msgs.dir/all] Error 2
make: *** [all] Error 2

then include the link in the issue. Thanks!

fjp, 2020-05-23:
I have the same problem with protobuf and gazebo9. I've installed gazebo9 from source (`brew install gazebo9 --build-from-source`) hoping to solve the bullet issue #1003. 
<details>
 <summary>bullet issue summary</summary>

```console
dyld: Library not loaded: /usr/local/opt/bullet/lib/libBulletSoftBody.2.88.dylib
Referenced from: /usr/local/bin/gazebo
Reason: image not found
```
</details>

Here is the output trying to install gazebo9:

```console
brew install gazebo9
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> Updated Formulae
ant@1.9                                                            jetty                                                              jetty-runner

==> Installing gazebo9 from osrf/simulation
==> Downloading https://gist.githubusercontent.com/scpeters/9199370/raw/afe595587e38737c537124a3652db99de026c272/brew_python_fix.patch
Already downloaded: /Users/fjp/Library/Caches/Homebrew/downloads/12171d5787bf5709495604c1591fd598a9cffc3a055eb509bea588d33019cabc--brew_python_fix.patch
==> Downloading https://osrf-distributions.s3.amazonaws.com/gazebo/releases/gazebo-9.13.0.tar.bz2
Already downloaded: /Users/fjp/Library/Caches/Homebrew/downloads/b89493ea5c4d99a4e108eaa82a894b87fc49a6a6251ae4d20bcb8f2577a79212--gazebo-9.13.0.tar.bz2
==> Patching
==> Applying brew_python_fix.patch
patching file deps/opende/CMakeLists.txt
==> cmake .. -DQWT_WIN_INCLUDE_DIR=/usr/local/lib/qwt.framework/Headers -DQWT_WIN_LIBRARY_DIR=/usr/local/lib/qwt.framework
==> make install
Last 15 lines from /Users/fjp/Library/Logs/Homebrew/gazebo9/02.make:
#error incompatible with your Protocol Buffer headers. Please
 ^
/usr/local/include/ignition/msgs1/ignition/msgs/material.pb.h:19:2: error: regenerate this file with a newer version of protoc.
#error regenerate this file with a newer version of protoc.
 ^
/usr/local/include/ignition/msgs1/ignition/msgs/material.pb.h:276:38: error: no type named 'InternalMetadataWithArena' in namespace 'google::protobuf::internal'
  ::PROTOBUF_NAMESPACE_ID::internal::InternalMetadataWithArena _internal_metadata_;
  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/usr/local/include/ignition/msgs1/ignition/msgs/material.pb.h:584:38: error: no type named 'InternalMetadataWithArena' in namespace 'google::protobuf::internal'
  ::PROTOBUF_NAMESPACE_ID::internal::InternalMetadataWithArena _internal_metadata_;
  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
18 errors generated.
make[2]: *** [gazebo/msgs/CMakeFiles/gazebo_msgs.dir/msgs.cc.o] Error 1
make[1]: *** [gazebo/msgs/CMakeFiles/gazebo_msgs.dir/all] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/osrf/homebrew-simulation/issues

These open issues may also help:
Rebuild bottles for: msgs1, transport4, gazebo9 and gazebo10 https://github.com/osrf/homebrew-simulation/pull/1028
```

scpeters, 2020-06-05:
all versions of ignition-msgs were rebuilt, and this should now be fixed

closed 2020-06-05 by scpeters