osrf/homebrew-simulation #1022: Failure in downstream projects that use ignition-msgs due to protobuf 3.12 release

traversaro, 2020-05-21:
As bottles of ignition-msgs has been generated with protobuf 3.11, there is now a failure when compiling any downstream project due to the fact that protobuf has been updated to 3.12 (https://github.com/Homebrew/homebrew-core/pull/54950), see for example: 
~~~
2020-05-21T08:25:11.2850930Z In file included from /usr/local/Cellar/gazebo10/10.2.0_2/include/gazebo-10/gazebo/physics/Entity.hh:25:
2020-05-21T08:25:11.2853360Z In file included from /usr/local/include/ignition/transport4/ignition/transport/Node.hh:32:
2020-05-21T08:25:11.2855640Z In file included from /usr/local/include/ignition/msgs1/ignition/msgs.hh:26:
2020-05-21T08:25:11.2858200Z In file included from /usr/local/include/ignition/msgs1/ignition/msgs/MessageTypes.hh:31:
2020-05-21T08:25:11.2862700Z /usr/local/include/ignition/msgs1/ignition/msgs/altimeter.pb.h:17:2: error: This file was generated by an older version of protoc which is
2020-05-21T08:25:11.2866630Z #error This file was generated by an older version of protoc which is
2020-05-21T08:25:11.2871670Z  ^
2020-05-21T08:25:11.2874400Z /usr/local/include/ignition/msgs1/ignition/msgs/altimeter.pb.h:18:2: error: incompatible with your Protocol Buffer headers. Please
2020-05-21T08:25:11.2878390Z #error incompatible with your Protocol Buffer headers. Please
2020-05-21T08:25:11.2881060Z  ^
2020-05-21T08:25:11.2882360Z /usr/local/include/ignition/msgs1/ignition/msgs/altimeter.pb.h:19:2: error: regenerate this file with a newer version of protoc.
2020-05-21T08:25:11.2883620Z #error regenerate this file with a newer version of protoc.
2020-05-21T08:25:11.2885040Z  ^
2020-05-21T08:25:11.2887810Z In file included from /Users/runner/runners/2.262.1/work/robotology-superbuild/robotology-superbuild/robotology/GazeboYARPPlugins/libraries/singleton
~~~

See https://github.com/robotology/robotology-superbuild/pull/410/checks?check_run_id=695577449 .

traversaro, 2020-05-21:
cc @nicogene

traversaro, 2020-05-21:
Fixed by https://github.com/osrf/homebrew-simulation/pull/1019 .

traversaro, 2020-05-21:
My bad, the problem is still present for ignition-msgs1 .

scpeters, 2020-06-05:
this was fixed by #1026

closed 2020-06-05 by scpeters