osrf/homebrew-simulation #1030: test-bot fork seems to be broken with latest brew

j-rivero, 2020-05-25:
While trying to generate bottles for pr #1028, test-bot from our fork returns an error message.

```
==> brew cleanup --prune=3
Error: Calling Formula#installed? is deprecated! Use Formula#latest_version_installed? (or Formula#any_version_installed? ) instead.
Please report this issue to the homebrew/test-bot tap (not Homebrew/brew or Homebrew/core), or even better, submit a PR to fix it:
  /usr/local/Homebrew/Library/Taps/homebrew/homebrew-test-bot/lib/test.rb:372
```

Seems to me like we need to update our fork of test-bot or patch it to work with the latest brew changes. @scpeters what would you recommend?

scpeters, 2020-05-26:
ok, I'll try cherry-picking https://github.com/Homebrew/homebrew-test-bot/commit/9597d5c3389b54ae8c3f6a462c818bd2c883bfb0

scpeters, 2020-05-26:
https://github.com/scpeters/homebrew-test-bot/commit/abec74eb6dc7615faceddfde59473369f548f298

j-rivero, 2020-05-27:
Thanks Steve, things seems to be working now.

closed 2020-05-27 by j-rivero