osrf/homebrew-simulation #1042: ignition-msgs* bottles broken due to AuxillaryParseTableField rename in protobuf 3.12.3

scpeters, 2020-06-05:
I noticed a failure of the `ignition-msgs1` bottle test:

* [image] https://build.osrfoundation.org/job/ignition_msgs1-install_bottle-homebrew-amd64/4/

~~~
+ brew test ignition-msgs1
Testing osrf/simulation/ignition-msgs1
==> pkg-config ignition-msgs1
==> /usr/bin/clang test.cpp -pthread -std=c++14 -I/usr/local/Cellar/ignition-msgs1/1.0.0_7/include/ignition/msgs1 -I/usr/local/Cellar/protobuf/3.12.3/include -I/usr/local/Cellar/ignition-math4/4.0.0/include/ignition/math4 -L/usr/local/Cellar/ignition-msgs1/1.0.0_7/lib -lignition-msgs1 -lc++ -o test
Last 15 lines from /Users/jenkins/Library/Logs/Homebrew/ignition-msgs1/test.02.clang:
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/usr/local/Cellar/protobuf/3.12.3/include/google/protobuf/generated_message_table_driven.h:141:7: note: 'AuxiliaryParseTableField' declared here
union AuxiliaryParseTableField {
      ^
In file included from test.cpp:1:
In file included from /usr/local/Cellar/ignition-msgs1/1.0.0_7/include/ignition/msgs1/ignition/msgs.hh:26:
In file included from /usr/local/Cellar/ignition-msgs1/1.0.0_7/include/ignition/msgs1/ignition/msgs/MessageTypes.hh:41:
/usr/local/Cellar/ignition-msgs1/1.0.0_7/include/ignition/msgs1/ignition/msgs/camerasensor.pb.h:59:51: error: no type named 'AuxillaryParseTableField' in namespace 'google::protobuf::internal'; did you mean 'AuxiliaryParseTableField'?
  static const ::PROTOBUF_NAMESPACE_ID::internal::AuxillaryParseTableField aux[]
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/usr/local/Cellar/protobuf/3.12.3/include/google/protobuf/generated_message_table_driven.h:141:7: note: 'AuxiliaryParseTableField' declared here
union AuxiliaryParseTableField {
      ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
~~~

Apparently a symbol was renamed in protobuf's most recent patch release https://github.com/protocolbuffers/protobuf/pull/7496 (merged into homebrew in https://github.com/Homebrew/homebrew-core/pull/55750), so we'll need to rebuild the `ignition-msgs*` bottles, but I think downstream bottles won't need to be rebuilt.

LorenzMeier, 2020-06-07:
Yep, just ran into this as well.

LorenzMeier, 2020-06-07:
I have removed Gazebo9 and protobuf but I'm still getting the compile error.

LorenzMeier, 2020-06-07:
@scpeters This is maybe because the Gazebo headers also need to be rebuilt: /usr/local/Cellar/gazebo9/9.13.0_1/include/gazebo-9/gazebo/msgs/quaternion.pb.h

LorenzMeier, 2020-06-07:
Verified with a rebuild from source, which fixes this.

hanyang2013212, 2020-06-11:
> brew install --build-from-source gazebo9
> ```
Hi, I've encountered the same error on protobuf 3.12. I've tired to build gazebo9 from the source, but it turned up with Cmake errors:
cmake .. -DQWT_WIN_INCLUDE_DIR=/usr/local/lib/qwt.framework/Headers -DQWT_WI
Last 15 lines from /Users/hanyang/Library/Logs/Homebrew/gazebo9/01.cmake:
-- 	ronn not found, manpages won't be generated
-- 	OpenSource Virtual Reality (OSVR) support will be disabled.
-- END BUILD WARNINGS

-- BUILD ERRORS: These must be resolved before compiling.
-- 	Missing: SDF version >=6. Required for reading and writing SDF files.
-- END BUILD ERRORS

CMake Error at CMakeLists.txt:314 (message):
  Errors encountered in build.  Please see the BUILD ERRORS above.

Do you have any idea about this?

scpeters, 2020-06-11:
~~~
-- Missing: SDF version >=6. Required for reading and writing SDF files.
~~~

it's having trouble finding your sdformat6 files. can you confirm that sdformat6 is installed and linked properly?

hanyang2013212, 2020-06-11:
> ```
> -- Missing: SDF version >=6. Required for reading and writing SDF files.
> ```
> 
> it's having trouble finding your sdformat6 files. can you confirm that sdformat6 is installed and linked properly?

Thanks for your quick response, it works! I have successfully build the gazebo9 from the source. Actually I'm trying to build PX4 gazebo model, but even when I build the gazebo from the source, I still got loads of errors regarding the AuxiliaryParseTableField

`In file included from MotorSpeed.pb.cc:4:
./MotorSpeed.pb.h:17:2: error: This file was generated by an older version of protoc which is
#error This file was generated by an older version of protoc which is
 ^
./MotorSpeed.pb.h:18:2: error: incompatible with your Protocol Buffer headers.  Please
#error incompatible with your Protocol Buffer headers.  Please
 ^
./MotorSpeed.pb.h:19:2: error: regenerate this file with a newer version of protoc.
#error regenerate this file with a newer version of protoc.
 ^
./MotorSpeed.pb.h:40:46: error: no type named 'AuxillaryParseTableField' in namespace 'google::protobuf::internal'; did you mean 'AuxiliaryParseTableField'?
  static const ::google::protobuf::internal::AuxillaryParseTableField aux[];
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
                                             AuxiliaryParseTableField
/usr/local/include/google/protobuf/generated_message_table_driven.h:141:7: note: 'AuxiliaryParseTableField' declared here
union AuxiliaryParseTableField {
      ^
In file included from MotorSpeed.pb.cc:4:
./MotorSpeed.pb.h:132:56: error: only virtual member functions can be marked 'final'
      ::google::protobuf::io::CodedInputStream* input) final;
                                                       ^~~~~
./MotorSpeed.pb.h:134:64: error: only virtual member functions can be marked 'final'
      ::google::protobuf::io::CodedOutputStream* output) const final;
                                                               ^~~~~
./MotorSpeed.pb.h:136:68: error: only virtual member functions can be marked 'final'
      bool deterministic, ::google::protobuf::uint8* target) const final;
                                                                   ^~~~~
./MotorSpeed.pb.h:174:33: error: no type named 'InternalMetadataWithArena' in namespace 'google::protobuf::internal'
  ::google::protobuf::internal::InternalMetadataWithArena _internal_metadata_;
  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
MotorSpeed.pb.cc:11:10: fatal error: 'google/protobuf/wire_format_lite_inl.h' file not found
#include <google/protobuf/wire_format_lite_inl.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
9 errors generated.
`

`In file included from CommandMotorSpeed.pb.cc:4:
./CommandMotorSpeed.pb.h:17:2: error: This file was generated by an older version of protoc which is
#error This file was generated by an older version of protoc which is
 ^
./CommandMotorSpeed.pb.h:18:2: error: incompatible with your Protocol Buffer headers.  Please
#error incompatible with your Protocol Buffer headers.  Please
 ^
./CommandMotorSpeed.pb.h:19:2: error: regenerate this file with a newer version of protoc.
#error regenerate this file with a newer version of protoc.
 ^
./CommandMotorSpeed.pb.h:40:46: error: no type named 'AuxillaryParseTableField' in namespace 'google::protobuf::internal'; did you mean 'AuxiliaryParseTableField'?
  static const ::google::protobuf::internal::AuxillaryParseTableField aux[];
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
                                             AuxiliaryParseTableField
/usr/local/include/google/protobuf/generated_message_table_driven.h:141:7: note: 'AuxiliaryParseTableField' declared here
union AuxiliaryParseTableField {
      ^
In file included from CommandMotorSpeed.pb.cc:4:
./CommandMotorSpeed.pb.h:132:56: error: only virtual member functions can be marked 'final'
      ::google::protobuf::io::CodedInputStream* input) final;
                                                       ^~~~~
./CommandMotorSpeed.pb.h:134:64: error: only virtual member functions can be marked 'final'
      ::google::protobuf::io::CodedOutputStream* output) const final;
                                                               ^~~~~
./CommandMotorSpeed.pb.h:136:68: error: only virtual member functions can be marked 'final'
      bool deterministic, ::google::protobuf::uint8* target) const final;
                                                                   ^~~~~
./CommandMotorSpeed.pb.h:174:33: error: no type named 'InternalMetadataWithArena' in namespace 'google::protobuf::internal'
  ::google::protobuf::internal::InternalMetadataWithArena _internal_metadata_;
  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
CommandMotorSpeed.pb.cc:11:10: fatal error: 'google/protobuf/wire_format_lite_inl.h' file not found
#include <google/protobuf/wire_format_lite_inl.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
9 errors generated.`

These are just from two plugins and there are many more similar errors. I'm quite confused how could I rebuild these plugins file.

TSC21, 2020-06-27:
Just to note that this remains a problem with Gazebo 9.

scpeters, 2020-06-30:
I've done some further testing and have seen some evidence that `brew install --build-from-source gazebo9` will work

I'm looking into it further

hanyang2013212, 2020-07-01:
@scpeters Thanks! Will try it and see whether it works!

hanyang2013212, 2020-07-02:
@scpeters Hi, I have tried to reinstall gazebo from the source. But making PX4 SITL model still gives error about protobuf.

closed 2020-07-20 by scpeters

TSC21, 2020-07-20:
So what' the solution for this? Does one need to install gazebo from source? What about CI systems where there's a time limit for jobs? This is simply not sustainable.

julianoes, 2020-07-22:
@scpeters why was this closed? What do you recommend to do?