osrf/homebrew-simulation #1050: macOS 10.15 Gazebo build from source failing during "make"

jasonhan-vassar, 2020-06-09:
Hi, I am trying to build Gazebo from source as [this comment](https://github.com/osrf/homebrew-simulation/issues/1011#issuecomment-630336068) suggested it is one way to get Gazebo working on macOS Catalina. However, running `make` throws some errors. Here are the commands I ran:
~~~
brew install gazebo11 --only-dependencies
cmake ../ -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt/5.14.2
make -j4
~~~
Here are the errors after running `make -j4`: [logs](https://gist.github.com/jasonhan-vassar/3b9c1c50c42b32e66b3e845622037065)

I saw a few errors during `cmake` but I assume they were all about optional packages like USB and Player. Can anyone resolve or reproduce this? Thank you.

scpeters, 2020-06-09:
I recently rebuilt the `ignition-msgs5` bottle due to #1042, but I see `AuxillaryParseTableField` in your error logs, so I suspect your installation of `ignition-msgs5` is out of date (ie. if `brew outdated` shows `ignition-msgs5`). You can upgrade it with `brew upgrade ignition-msgs5`. Let me know if that helps.

jasonhan-vassar, 2020-06-09:
Gazebo builds and runs successfully! Thank you @scpeters for the quick reply.

closed 2020-06-09 by j-rivero