osrf/homebrew-simulation #1053: Cmake error when trying to build gazebo9 from the source.

hanyang2013212, 2020-06-11:
~~~
brew install gazebo9 --build-from-source
~~~

==> cmake .. -DQWT_WIN_INCLUDE_DIR=/usr/local/lib/qwt.framework/Headers -DQWT_WI
Last 15 lines from /Users/hanyang/Library/Logs/Homebrew/gazebo9/01.cmake:
-- 	ronn not found, manpages won't be generated
-- 	OpenSource Virtual Reality (OSVR) support will be disabled.
-- END BUILD WARNINGS

-- BUILD ERRORS: These must be resolved before compiling.
-- 	Missing: SDF version >=6. Required for reading and writing SDF files.
-- END BUILD ERRORS

CMake Error at CMakeLists.txt:314 (message):
  Errors encountered in build.  Please see the BUILD ERRORS above.

-- Configuring incomplete, errors occurred!
See also "/tmp/gazebo9-20200611-7545-143c550/gazebo-9.13.0/build/CMakeFiles/CMakeOutput.log".
See also "/tmp/gazebo9-20200611-7545-143c550/gazebo-9.13.0/build/CMakeFiles/CMakeError.log".

Cmake error when trying to build gazebo9 from the source.

scpeters, 2020-06-30:
can you try `brew install --build-from-source gazebo9`?

I hadn't realized that this might work, but I think it will

julianoes, 2020-07-22:
I can confirm that this doesn't work anymore:

```
brew reinstall --build-from-source gazebo9
==> Downloading https://homebrew.bintray.com/bottles/python%403.8-3.8.5.mojave.bottle.tar.gz
==> Downloading from https://d29vzk4ow07wi7.cloudfront.net/c663ffeac543b83aeae1ca13700fcb039edd57b3f0579b78fed1c2c4836ae54f?response-content-dispos
######################################################################## 100.0%
==> Downloading https://gist.githubusercontent.com/scpeters/9199370/raw/afe595587e38737c537124a3652db99de026c272/brew_python_fix.patch
Already downloaded: /Users/julianoes/Library/Caches/Homebrew/downloads/12171d5787bf5709495604c1591fd598a9cffc3a055eb509bea588d33019cabc--brew_python_fix.patch
==> Downloading https://osrf-distributions.s3.amazonaws.com/gazebo/releases/gazebo-9.13.0.tar.bz2
Already downloaded: /Users/julianoes/Library/Caches/Homebrew/downloads/b89493ea5c4d99a4e108eaa82a894b87fc49a6a6251ae4d20bcb8f2577a79212--gazebo-9.13.0.tar.bz2
==> Reinstalling osrf/simulation/gazebo9
==> Installing dependencies for osrf/simulation/gazebo9: python@3.8
==> Installing osrf/simulation/gazebo9 dependency: python@3.8
==> Pouring python@3.8-3.8.5.mojave.bottle.tar.gz
==> /usr/local/Cellar/python@3.8/3.8.5/bin/python3 -s setup.py --no-user-cfg install --force --verbose --install-scripts=/usr/local/Cellar/python@3
==> /usr/local/Cellar/python@3.8/3.8.5/bin/python3 -s setup.py --no-user-cfg install --force --verbose --install-scripts=/usr/local/Cellar/python@3
==> /usr/local/Cellar/python@3.8/3.8.5/bin/python3 -s setup.py --no-user-cfg install --force --verbose --install-scripts=/usr/local/Cellar/python@3
==> Caveats
Python has been installed as
  /usr/local/bin/python3

Unversioned symlinks `python`, `python-config`, `pip` etc. pointing to
`python3`, `python3-config`, `pip3` etc., respectively, have been installed into
  /usr/local/opt/python@3.8/libexec/bin

You can install Python packages with
  pip3 install <package>
They will install into the site-package directory
  /usr/local/lib/python3.8/site-packages

See: https://docs.brew.sh/Homebrew-and-Python
==> Summary
🍺  /usr/local/Cellar/python@3.8/3.8.5: 4,331 files, 66.8MB
==> Installing osrf/simulation/gazebo9
==> Patching
==> Applying brew_python_fix.patch
patching file deps/opende/CMakeLists.txt
==> cmake .. -DQWT_WIN_INCLUDE_DIR=/usr/local/lib/qwt.framework/Headers -DQWT_WIN_LIBRARY_DIR=/usr/local/lib/qwt.framework
Last 15 lines from /Users/julianoes/Library/Logs/Homebrew/gazebo9/01.cmake:
-- 	ronn not found, manpages won't be generated
-- 	OpenSource Virtual Reality (OSVR) support will be disabled.
-- END BUILD WARNINGS

-- BUILD ERRORS: These must be resolved before compiling.
-- 	Missing: SDF version >=6. Required for reading and writing SDF files.
-- END BUILD ERRORS

CMake Error at CMakeLists.txt:314 (message):
  Errors encountered in build.  Please see the BUILD ERRORS above.

-- Configuring incomplete, errors occurred!
See also "/tmp/gazebo9-20200722-50735-1pj8mbg/gazebo-9.13.0/build/CMakeFiles/CMakeOutput.log".
See also "/tmp/gazebo9-20200722-50735-1pj8mbg/gazebo-9.13.0/build/CMakeFiles/CMakeError.log".

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/osrf/homebrew-simulation/issues

These open issues may also help:
gazebo9 9.13.2 https://github.com/osrf/homebrew-simulation/pull/1066
Cmake error when trying to build gazebo9 from the source.  https://github.com/osrf/homebrew-simulation/issues/1053
```

And sadly these logfiles are gone when you want to look at them.

scpeters, 2020-09-22:
I just rebuilt the bottles for gazebo9, so this should now be fixed

closed 2020-09-22 by scpeters