osrf/homebrew-simulation #1067: Bottles using protobuf need to be rebuilt

scpeters, 2020-07-21:
A quick glance at the [bottle status](https://github.com/osrf/homebrew-simulation#bottle-status) shows that the `ignition-msgs*`, `-transport*`, `-sensors*` and `-gazebo*` bottles have failing tests. A recent rebuild of the protobuf formula (https://github.com/Homebrew/homebrew-core/pull/58229) means we need to rebuild the `ignition-msgs*` bottles.

chapulina, 2020-07-23:
I think this probably explains all the red OSX builds here, right?

https://github.com/osrf/buildfarmer/blob/928f4b0ca2a7a342a022127b9111381c661a93ae/Ignition.md

scpeters, 2020-07-24:
> I think this probably explains all the red OSX builds here, right?
> 
> https://github.com/osrf/buildfarmer/blob/928f4b0ca2a7a342a022127b9111381c661a93ae/Ignition.md

yep

scpeters, 2020-07-24:
I've removed the ignition-msgs bottles for now (https://github.com/osrf/homebrew-simulation/commit/334c3befaa6a820ed74022c022dba738a93ec9e2), so they will build from source until we can fix out bottle infrastructure

scpeters, 2020-09-22:
bottles for the following packages have been rebuilt:

* gazebo[9, 10, 11] and dependencies
* ignition blueprint, citadel, dome and dependencies

these are our best supported formulae, so I will close this issue and address the remaining formulae in separate issues

closed 2020-09-22 by scpeters