osrf/homebrew-simulation #1076: Installing gazebo11 on iMac fails

RayLucchesi, 2020-07-23:
Build/install error(s):

Last 15 lines from /Users/rllucchesi/Library/Logs/Homebrew/gazebo11/02.make:
    ::PROTOBUF_NAMESPACE_ID::ConstStringParam name, Material_PBR_WorkflowType* value) {
    ~~~~~~~~~~~~~~~~~~~~~~~~~^
/usr/local/include/ignition/msgs5/ignition/msgs/material.pb.h:141:30: error: no type named 'ConstStringParam' in namespace 'google::protobuf'
    ::PROTOBUF_NAMESPACE_ID::ConstStringParam name, Material_ShaderType* value) {
    ~~~~~~~~~~~~~~~~~~~~~~~~~^
/usr/local/include/ignition/msgs5/ignition/msgs/material.pb.h:454:66: error: no type named 'ConstStringParam' in namespace 'google::protobuf'
  static inline bool WorkflowType_Parse(::PROTOBUF_NAMESPACE_ID::ConstStringParam name,
                                        ~~~~~~~~~~~~~~~~~~~~~~~~~^
/usr/local/include/ignition/msgs5/ignition/msgs/material.pb.h:823:64: error: no type named 'ConstStringParam' in namespace 'google::protobuf'
  static inline bool ShaderType_Parse(::PROTOBUF_NAMESPACE_ID::ConstStringParam name,
                                      ~~~~~~~~~~~~~~~~~~~~~~~~~^
8 errors generated.
make[2]: *** [gazebo/msgs/CMakeFiles/gazebo_msgs.dir/msgs.cc.o] Error 1
make[1]: *** [gazebo/msgs/CMakeFiles/gazebo_msgs.dir/all] Error 2
make: *** [all] Error 2

I used the one line gazebo install
curl -ssL http://get.gazebosim.org | sh
and it failed. 

Mac OS x 10.15.5
iMac with 8GB memory

EngMarcoCaiazzo, 2020-07-24:
==> Summary
🍺  /usr/local/Cellar/jpeg-turbo/2.0.5: 40 files, 3.3MB
==> Installing osrf/simulation/gazebo11 dependency: sdl
==> Pouring sdl-1.2.15_1.catalina.bottle.1.tar.gz
🍺  /usr/local/Cellar/sdl/1.2.15_1: 225 files, 1.4MB
==> Installing osrf/simulation/gazebo11 dependency: open-scene-graph
==> Pouring open-scene-graph-3.6.5.catalina.bottle.tar.gz
🍺  /usr/local/Cellar/open-scene-graph/3.6.5: 8,900 files, 219.6MB
==> Installing osrf/simulation/gazebo11 dependency: dartsim
==> Pouring dartsim-6.9.2_8.catalina.bottle.tar.gz
Warning: dartsim dependency gcc was built with a different C++ standard
library (libstdc++ from clang). This may cause problems at runtime.
🍺  /usr/local/Cellar/dartsim/6.9.2_8: 1,080 files, 44.3MB
==> Installing osrf/simulation/gazebo11 dependency: simbody
==> Pouring simbody-3.7.mojave.bottle.tar.gz
🍺  /usr/local/Cellar/simbody/3.7: 3,855 files, 135.1MB
==> Installing osrf/simulation/gazebo11
==> Patching
==> Applying brew_python_fix.patch
patching file deps/opende/CMakeLists.txt
==> cmake .. -DQWT_WIN_INCLUDE_DIR=/usr/local/lib/qwt.framework/Headers -DQWT_WI
==> make install
Last 15 lines from /Users/m.caiazzo/Library/Logs/Homebrew/gazebo11/02.make:
    ::PROTOBUF_NAMESPACE_ID::ConstStringParam name, Material_PBR_WorkflowType* value) {
    ~~~~~~~~~~~~~~~~~~~~~~~~~^
/usr/local/include/ignition/msgs5/ignition/msgs/material.pb.h:141:30: error: no type named 'ConstStringParam' in namespace 'google::protobuf'
    ::PROTOBUF_NAMESPACE_ID::ConstStringParam name, Material_ShaderType* value) {
    ~~~~~~~~~~~~~~~~~~~~~~~~~^
/usr/local/include/ignition/msgs5/ignition/msgs/material.pb.h:454:66: error: no type named 'ConstStringParam' in namespace 'google::protobuf'
  static inline bool WorkflowType_Parse(::PROTOBUF_NAMESPACE_ID::ConstStringParam name,
                                        ~~~~~~~~~~~~~~~~~~~~~~~~~^
/usr/local/include/ignition/msgs5/ignition/msgs/material.pb.h:823:64: error: no type named 'ConstStringParam' in namespace 'google::protobuf'
  static inline bool ShaderType_Parse(::PROTOBUF_NAMESPACE_ID::ConstStringParam name,
                                      ~~~~~~~~~~~~~~~~~~~~~~~~~^
8 errors generated.
make[2]: *** [gazebo/msgs/CMakeFiles/gazebo_msgs.dir/msgs.cc.o] Error 1
make[1]: *** [gazebo/msgs/CMakeFiles/gazebo_msgs.dir/all] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/osrf/homebrew-simulation/issues

These open issues may also help:
Installing gazebo11 on iMac fails https://github.com/osrf/homebrew-simulation/issues/1076

mamunul, 2020-07-24:
Same Error occurred for me. My mac has 3.12.3 version of Google protobuf library, which doesn't have the api 'ConstStringParam'. That api may be introduced after 3.12.3.

LandSquidConfirmed, 2020-07-28:
Same error here.

TLi347, 2020-07-30:
same error!!!!

RayLucchesi, 2020-07-30:
I was able to install Gazebo-10 though.

TLi347, 2020-07-31:
Hi, I successfully run brew install gazebo10 as well. It ends with:
Emacs Lisp files have been installed to:
  /usr/local/share/emacs/site-lisp/cmake

But I don't know how to start this software.

TLi347, 2020-07-31:
When I run "gazebo", it shows:
dyld: Symbol not found: __ZN6gazebo3gui10DataLogger16staticMetaObjectE
  Referenced from: /usr/local/Cellar/gazebo10/10.2.0_3/lib/libgazebo_gui.10.dylib
  Expected in: flat namespace
 in /usr/local/Cellar/gazebo10/10.2.0_3/lib/libgazebo_gui.10.dylib
Abort trap: 6

LandSquidConfirmed, 2020-08-02:
Gazebo 10 installs but fails to run with this error—Gazebo11 installation now hangs in the middle at 100% CPU (presumably the change was caused by the homebrew update?)
Not sure what to do about the C++ standard. 

Warning: osrf/simulation/gazebo10 dependency gcc was built with a different C++ standard
library (libstdc++ from clang). This may cause problems at runtime.
🍺  /usr/local/Cellar/gazebo10/10.2.0_3: 1,268 files, 127.9MB
M ~ % gazebo
dyld: Symbol not found: __ZN6gazebo3gui10DataLogger16staticMetaObjectE
  Referenced from: /usr/local/Cellar/gazebo10/10.2.0_3/lib/libgazebo_gui.10.dylib
  Expected in: flat namespace
 in /usr/local/Cellar/gazebo10/10.2.0_3/lib/libgazebo_gui.10.dylib
zsh: abort      gazebo

Bachery, 2020-08-14:
Same error when using Homebrew installing gazebo11

jimdgit, 2020-08-19:
Last 15 lines from /Users/jcdonelson/Library/Logs/Homebrew/ignition-msgs5/02.make:
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'MSCGEN_PATH' at line 2139 of file '/tmp/ignition-msgs5-20200819-52895-1o99kea/ignition-msgs-5.3.0/api.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
13 errors generated.
make[2]: *** [src/CMakeFiles/ign_msgs_gen.dir/Generator.cc.o] Error 1
make[1]: *** [src/CMakeFiles/ign_msgs_gen.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  1%] Built target doc
[  1%] Linking CXX static library ../lib/libgtest.a
cd /tmp/ignition-msgs5-20200819-52895-1o99kea/ignition-msgs-5.3.0/test && /usr/local/Cellar/cmake/3.18.1/bin/cmake -P CMakeFiles/gtest.dir/cmake_clean_target.cmake
cd /tmp/ignition-msgs5-20200819-52895-1o99kea/ignition-msgs-5.3.0/test && /usr/local/Cellar/cmake/3.18.1/bin/cmake -E cmake_link_script CMakeFiles/gtest.dir/link.txt --verbose=1
/usr/bin/ar qc ../lib/libgtest.a CMakeFiles/gtest.dir/gtest/src/gtest-all.cc.o
/usr/bin/ranlib ../lib/libgtest.a
[  1%] Built target gtest
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/osrf/homebrew-simulation/issues

mbp:src jcdonelson$ gazebo
-bash: gazebo: command not found

scpeters, 2020-09-22:
the bottle has been rebuilt; it should be working now

closed 2020-09-22 by scpeters