osrf/homebrew-simulation #1081: Mac Gazebo11 Catalina build fails at 79%

Interaze, 2020-07-30:
I've been attempting several downloads of gazebo11 and 9 with seldom luck. The makefile dump of gazebo11 install for the last 15 lines are included here. Thankyou for your help. I've attempted these commands:

brew tap osrf/simulation
brew install gazebo11

curl -ssL http://get.gazebosim.org | sh

brew install gazebo11 --HEAD

brew install gazebo11 --with-bullet --with-simbody

I also have XQuartz installed already, so I don't believe that to be the issue.

[Logs.zip](https://github.com/osrf/homebrew-simulation/files/4998395/Logs.zip)

fuying-wang, 2020-08-10:
I got the same issue.

ahcorde, 2020-08-10:
This PR should fix tghe issue https://github.com/osrf/gazebo/pull/2809

nunoguedelha, 2020-08-11:
Thank you @ahcorde , I had the same issue. I tried your branch on top of the `gazebo11` branch and followed the steps:

1. `brew install gazebo11 --only-dependencies`
2. Gazebo build from source instruction here: build from source instructions here: https://github.com/osrf/gazebo_tutorials/blob/master/install_from_source/tutorial_default.md#build-and-install-gazebo
3. Add `<CMAKE_INSTALL_PREFIX>/bin` to `PATH` env variable.
4. Add `<CMAKE_INSTALL_PREFIX>/lib` to `DYLD_LIBRARY_PATH` env variable.

Which results in a successful build and Gazebo11 running without issues.

Interaze, 2020-08-13:
Thank-you for the help. Due to my circumstances I'm currently using Gazebo 9 and am incapable of testing it, but I'll trust that the new PR fixes it. I just wanted to let you guys know of the issue for whoever else wanted to use it, thanks again for the help.

closed 2020-08-13 by Interaze