osrf/homebrew-simulation #1110: Gazebo 7 CI failure due to Qt4 cmake error

azeey, 2020-09-02:
Here's the failing build: https://build.osrfoundation.org/job/gazebo-ci-gazebo7-homebrew-amd64/279/. And the cmake configuration log:

```
++ cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/gazebo7/HEAD /Users/jenkins/workspace/gazebo-ci-gazebo7-homebrew-amd64/gazebo
-- The C compiler identification is AppleClang 10.0.0.10001044
-- The CXX compiler identification is AppleClang 10.0.0.10001044
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/opt/ccache/libexec/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/opt/ccache/libexec/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Gazebo version 7.16.1
-- Found Doxygen: /usr/local/bin/doxygen (found version "1.8.20") found components: doxygen dot 
-- High memory tests: enabled
-- 

====== Finding 3rd Party Packages ======
-- Operating system is Apple MacOS X
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'freeimage>=3.9.0'
--   No package 'freeimage' found
--   freeimage.pc not found, trying freeimage_include_dir and freeimage_library_dir flags.
--   Found /usr/local/include/FreeImage.h
--   Looking for FreeImage.h - found
--   Looking for libfreeimage - found
-- Found Protobuf: /usr/local/lib/libprotobuf.dylib (found version "3.13.0") 
-- Found OpenGL: /System/Library/Frameworks/OpenGL.framework   
-- Found OpenAL: /System/Library/Frameworks/OpenAL.framework  
-- Could NOT find HDF5 (missing: HDF5_LIBRARIES HDF5_INCLUDE_DIRS C CXX) (found version "")
-- 	HDF5 not found
-- Checking for module 'libcurl'
--   Found libcurl, version 7.54.0
-- Checking for module 'libprofiler'
--   No package 'libprofiler' found
-- Looking for libprofiler - not found
-- Checking for module 'libtcmalloc'
--   No package 'libtcmalloc' found
-- Looking for libtcmalloc - not found
-- Looking for DARTCore - not found
-- 	DART not found, for dart physics engine option, please install libdart-core4-dev.
-- Using system tinyxml.
-- Checking for module 'tinyxml'
--   Found tinyxml, version 2.6.2
-- Using system tinyxml2.
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 8.0.0
-- Looking for libtar.h - found
-- Looking for libtar.so - found
-- Checking for module 'tbb'
--   No package 'tbb' found
-- TBB not found, attempting to detect manually
-- Checking for module 'OGRE-RTShaderSystem>=1.7.4'
--   Found OGRE-RTShaderSystem, version 1.9.0
-- Checking for module 'OGRE>=1.7.4'
--   Found OGRE, version 1.9.0
-- Checking for module 'OGRE-Terrain'
--   Found OGRE-Terrain, version 1.9.0
-- Checking for module 'OGRE-Overlay'
--   Found OGRE-Overlay, version 1.9.0
-- Checking for module 'ccd>=1.4'
--   No package 'ccd' found
-- Using internal copy of libccd
-- Checking for module 'libswscale'
--   Found libswscale, version 5.7.100
-- Checking for module 'libavformat'
--   Found libavformat, version 58.45.100
-- Checking for module 'libavcodec'
--   Found libavcodec, version 58.91.100
-- Checking for module 'libavutil'
--   Found libavutil, version 56.51.100
-- Checking for modules 'playercore>=3.0;playerc++;playerwkb'
--   No package 'playercore' found
--   No package 'playerc++' found
--   No package 'playerwkb' found
-- 	Player not found, gazebo plugin for player will not be built.
-- Checking for module 'gts'
--   Found gts, version 0.7.6
-- Looking for GTS - found
-- Checking for module 'bullet>=2.82'
--   Found bullet, version 2.89
-- Checking for module 'libusb-1.0'
--   No package 'libusb-1.0' found
-- 	libusb-1.0 not found. USB peripherals support will be disabled.
-- Checking for module 'OculusVR'
--   No package 'OculusVR' found
-- 	Oculus Rift support will be disabled.
-- Found Boost: /usr/local/lib/cmake/Boost-1.73.0/BoostConfig.cmake (found version "1.73.0")  
-- Looking for SDFormat - found
CMake Error at /usr/local/Cellar/cmake/3.18.2/share/cmake/Modules/FindQt4.cmake:1314 (message):
  Found unsuitable Qt version "" from NOTFOUND, this code requires Qt 4.x
Call Stack (most recent call first):
  cmake/SearchForStuff.cmake:546 (find_package)
  CMakeLists.txt:128 (include)

-- Configuring incomplete, errors occurred!
```

scpeters, 2020-09-23:
I tried but wasn't able to reproduce this on my mojave machine. I noticed that this job is explicitly running on high sierra:

* https://github.com/ignition-tooling/release-tools/blob/35e122906ccf7b6cff8623a82cd150bf5b364386/jenkins-scripts/dsl/gazebo.dsl#L566-L567

I'll try reproducing this on the high sierra machine for confirmation, but will likely update the job to use mojave instead, since high sierra will be unsupported by homebrew soon.

scpeters, 2020-09-23:
It seems that the `qt@4` bottle is installed but the `brew link` step is not performed for some reason.

scpeters, 2020-09-23:
I just noticed a small bit in the console log:

~~~
==> Reinstalling 1 broken dependent from source:
cartr/qt4/qt@4
~~~

I think there's a problem with the `qt@4` bottle

chapulina, 2020-11-02:
Because Gazebo 7 will EOL in 3 months (2021-01-25) and given the complexity of the fix, we don't think it's worth it fixing this now. I'll close this issue and disable CI for Gazebo 7 on macOS. Please speak up if you're a user affected by this.

closed 2020-11-02 by chapulina