osrf/homebrew-simulation #1120: ignition-math6 6.5.0 brew install error

tomcattigerkkk, 2020-09-11:
The problem is due to the Ruby version on macOS (Ruby 2.6).
How to enforce the package to use the brew installed version (Ruby 2.7)? Thanks

[ 45%] Building CXX object src/CMakeFiles/math.dir/CMakeFiles/math.dir/rubyRUBY_wrap.cxx.o
cd /tmp/ignition-math6-20200911-34806-1ontjo2/ignition-math-6.5.0/src && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DMAC_OS_X_VERSION=1080 -Dmath_EXPORTS -I/tmp/ignition-math6-20200911-34806-1ontjo2/ignition-math-6.5.0/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/include/ruby-2.6.0 -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/include/ruby-2.6.0/universal-darwin19 -stdlib=libc++ -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -fPIC -std=c++17 -o CMakeFiles/math.dir/CMakeFiles/math.dir/rubyRUBY_wrap.cxx.o -c /tmp/ignition-math6-20200911-34806-1ontjo2/ignition-math-6.5.0/src/CMakeFiles/math.dir/rubyRUBY_wrap.cxx
In file included from /tmp/ignition-math6-20200911-34806-1ontjo2/ignition-math-6.5.0/src/CMakeFiles/math.dir/rubyRUBY_wrap.cxx:889:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/include/ruby-2.6.0/ruby.h:33:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/include/ruby-2.6.0/ruby/ruby.h:24:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/include/ruby-2.6.0/universal-darwin19/ruby/config.h:413:31: error: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wreserved-user-defined-literal]
#define RUBY_ARCH "universal-"RUBY_PLATFORM_OS
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/include/ruby-2.6.0/universal-darwin19/ruby/config.h:414:35: error: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wreserved-user-defined-literal]
#define RUBY_PLATFORM "universal."RUBY_PLATFORM_CPU"-"RUBY_PLATFORM_OS
                                  ^

/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/include/ruby-2.6.0/universal-darwin19/ruby/config.h:414:55: error: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wreserved-user-defined-literal]
#define RUBY_PLATFORM "universal."RUBY_PLATFORM_CPU"-"RUBY_PLATFORM_OS
                                                      ^

In file included from /tmp/ignition-math6-20200911-34806-1ontjo2/ignition-math-6.5.0/src/CMakeFiles/math.dir/rubyRUBY_wrap.cxx:889:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/include/ruby-2.6.0/ruby.h:33:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/include/ruby-2.6.0/ruby/ruby.h:2111:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/include/ruby-2.6.0/ruby/intern.h:56:19: error: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
void rb_mem_clear(register VALUE*, register long);
                  ^~~~~~~~~
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/include/ruby-2.6.0/ruby/intern.h:56:36: error: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
void rb_mem_clear(register VALUE*, register long);

traversaro, 2020-09-11:
Example of the failure on a GitHub Action job: https://github.com/robotology/robotology-superbuild/pull/470/checks?check_run_id=1101300827 .

scpeters, 2020-09-11:
I'm able to reproduce this if `swig` is installed. Our CI currently doesn't test with `swig` on macOS.

scpeters, 2020-09-11:
it looks like the issue has been reported here:

* https://trac.macports.org/ticket/58255

I'm not sure if it's been reported upstream to ruby though

scpeters, 2020-09-11:
I think this is an upstream problem with ruby when provided as a universal binary, which is how the system versions of ruby on macOS are provided. I think a workaround would be to depend on the homebrew version of ruby. I would add a dependency on `swig` to the `ignition-math6` formula as well.

scpeters, 2020-09-11:
potential fix in https://github.com/osrf/homebrew-simulation/pull/1121; please review if you have a chance

tomcattigerkkk, 2020-09-12:
@scpeters excellent work! Thanks a lot. macOS 10.15.6 can install ignition-math6.6.5 through brew now. 
The problem by (https://trac.macports.org/ticket/58255) I have already manually change the lines in the header file. But your configuration in .rb solves the issue totally. Thanks again.

closed 2020-09-12 by tomcattigerkkk