osrf/homebrew-simulation #1123: Error compiling gazebo9 ModelTreeWidget.cc

egradman, 2020-09-13:
Hi, I'm installing gazebo9 (for use with px4-sim).

I ran `brew install osrf/homebrew-simulation/gazebo9 --verbose` to get this build log:

https://gist.github.com/egradman/3aabc6b4d6a3242ba81d4cdb9d87bc55

Here's the good (bad) bits:

```
/tmp/gazebo9-20200912-95689-1ea5hlg/gazebo-9.14.0/gazebo/gui/model/ModelTreeWidget.cc:165:56: error: use of undeclared identifier '_1'
      boost::bind(&ModelTreeWidget::OnSaveModel, this, _1)));
```

egradman, 2020-09-13:
FYI, same error on gazebo10

scpeters, 2020-09-14:
thanks, we've merged a fix upstream in https://github.com/osrf/gazebo/commit/c55942ed348580317ea77312f7efce5c7937a49c; I'll add it as a patch to the gazebo9 formula

scpeters, 2020-09-14:
if you want to test the patch, there is example syntax in https://github.com/osrf/homebrew-simulation/pull/992/commits/f4e822a13ab55826a3fa9fee5f6b8673cb75c64d

scpeters, 2020-09-15:
patch in #1128

closed 2020-09-15 by scpeters

egradman, 2020-09-15:
worked for me!  Thanks for fixing it so quickly.

PR #1128 (2020-09-15): gazebo9/10: patch for boost 1.73.0
Use patch from osrf/gazebo#2809 to fix #1123.