osrf/homebrew-simulation #1172: can't load library.

tomcattigerkkk, 2020-10-01:
Hello, 

I got an odd issue with libraries on my macOS,

ign gazebo -s
Library error for [/usr/local/Cellar/ignition-gazebo4/4.0.0/lib/libignition-gazebo4-ign.4.0.0.dylib]: can't load /usr/local/Cellar/ignition-gazebo4/4.0.0/lib/libignition-gazebo4-ign.4.0.0.dylib

Any suggestions? Thank a lot

ahcorde, 2020-10-02:
Hi, 

Probably we need to fix the bottle. Anyhow the support for macOS is not complete. For now, you can only run ignition gazebo in server mode.

```
ign gazebo -s
```

The Scene3D plugin it's failing for some reason when you run ign gazebo with the gui.

scpeters, 2020-12-14:
I have seen this `Library error for [*.dylib]: can't load` message before, and for me it was related to System Integrity Protection (SIP) restrictions on the `/usr/bin/ruby` binary that was executing the `ign` ruby script. I recently added tests to the `ignition-gazebo*` formulae in #1221 that explicitly invoke the `brew` version of `ruby` from `/usr/local/opt/ruby/bin`, and those tests have been passing for me, so I recommend trying `brew install ruby` and `/usr/local/opt/ruby/bin/ruby $(which ign) gazebo -s -v 4`

tomcattigerkkk, 2020-12-18:
@scpeters  Hi, thanks for your comment. However, I still have the same error, that these dylib cannot be found.

scpeters, 2020-12-18:
@tomcattigerkkk can you provide the output of `brew config` and `brew doctor`?

Ratatinator97, 2022-10-26:
Hello
I just installed on my M1 the `ignition-gazebo7` package and I have the following errors:
```
➜  gz plugin                                                                               
dyld[3862]: Library not loaded: '@rpath/libgz-plugin2-loader.2.dylib'
  Referenced from: '/opt/homebrew/Cellar/gz-plugin2/2.0.0/libexec/gz/plugin2/gz-plugin'
  Reason: tried: '/opt/homebrew/Cellar/gz-plugin2/2.0.0/libexec/gz/plugin2/../lib/libgz-plugin2-loader.2.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.0/libexec/gz/plugin2/../lib/libgz-plugin2-loader.2.dylib' (no such file), '/usr/local/lib/libgz-plugin2-loader.2.dylib' (no such file), '/usr/lib/libgz-plugin2-loader.2.dylib' (no such file)
Error: Version mismatch. Your configuration file version is
            [2.0.0] but /opt/homebrew/Cellar/gz-plugin2/2.0.0/libexec/gz/plugin2/gz-plugin version is [].
```
Same issue when I run a SDF file with `gz sim -s MySDF.sdf`:
```
Error while loading the library [/opt/homebrew/Cellar/gz-sim7/7.0.0_1/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib]: dlopen(/opt/homebrew/Cellar/gz-sim7/7.0.0_1/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib, 0x0085): Library not loaded: '@rpath/libgz-sim7.7.dylib'
  Referenced from: '/opt/homebrew/Cellar/gz-sim7/7.0.0_1/lib/gz-sim-7/plugins/libgz-sim7-physics-system.7.0.0.dylib'
  Reason: tried: '/opt/homebrew/Cellar/gz-sim7/7.0.0_1/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.0.0_1/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.0/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/local/lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file)
```
and for `gz-sim-physics-system`, `gz-sim-user-commands-system`, `gz-sim-scene-broadcaster-system`, `gz-sim-particle-emitter-system` ... 

```
➜  ls /opt/homebrew/Cellar/gz-sim7/7.0.0_1/lib/gz-sim-7/plugins/                              
gui                                                       libgz-sim7-joint-position-controller-system.7.dylib
libgz-sim-ackermann-steering-system.dylib                 libgz-sim7-joint-position-controller-system.dylib
libgz-sim-air-pressure-system.dylib                       libgz-sim7-joint-state-publisher-system.7.0.0.dylib
libgz-sim-altimeter-system.dylib                          libgz-sim7-joint-state-publisher-system.7.dylib
libgz-sim-apply-joint-force-system.dylib                  libgz-sim7-joint-state-publisher-system.dylib
libgz-sim-apply-link-wrench-system.dylib                  libgz-sim7-joint-trajectory-controller-system.7.0.0.dylib
libgz-sim-breadcrumbs-system.dylib                        libgz-sim7-joint-trajectory-controller-system.7.dylib
libgz-sim-buoyancy-engine-system.dylib                    libgz-sim7-joint-trajectory-controller-system.dylib
libgz-sim-buoyancy-system.dylib                           libgz-sim7-kinetic-energy-monitor-system.7.0.0.dylib
libgz-sim-camera-video-recorder-system.dylib              libgz-sim7-kinetic-energy-monitor-system.7.dylib
libgz-sim-collada-world-exporter-system.dylib             libgz-sim7-kinetic-energy-monitor-system.dylib
libgz-sim-comms-endpoint-system.dylib                     libgz-sim7-label-system.7.0.0.dylib
libgz-sim-contact-system.dylib                            libgz-sim7-label-system.7.dylib
libgz-sim-detachable-joint-system.dylib                   libgz-sim7-label-system.dylib
libgz-sim-diff-drive-system.dylib                         libgz-sim7-lift-drag-system.7.0.0.dylib
libgz-sim-elevator-system.dylib                           libgz-sim7-lift-drag-system.7.dylib
libgz-sim-follow-actor-system.dylib                       libgz-sim7-lift-drag-system.dylib
libgz-sim-forcetorque-system.dylib                        libgz-sim7-linearbatteryplugin-system.7.0.0.dylib
libgz-sim-hydrodynamics-system.dylib                      libgz-sim7-linearbatteryplugin-system.7.dylib
libgz-sim-imu-system.dylib                                libgz-sim7-linearbatteryplugin-system.dylib
libgz-sim-joint-controller-system.dylib                   libgz-sim7-log-system.7.0.0.dylib
libgz-sim-joint-position-controller-system.dylib          libgz-sim7-log-system.7.dylib
libgz-sim-joint-state-publisher-system.dylib              libgz-sim7-log-system.dylib
libgz-sim-joint-trajectory-controller-system.dylib        libgz-sim7-log-video-recorder-system.7.0.0.dylib
libgz-sim-kinetic-energy-monitor-system.dylib             libgz-sim7-log-video-recorder-system.7.dylib
libgz-sim-label-system.dylib                              libgz-sim7-log-video-recorder-system.dylib
libgz-sim-lift-drag-system.dylib                          libgz-sim7-logical-camera-system.7.0.0.dylib
libgz-sim-linearbatteryplugin-system.dylib                libgz-sim7-logical-camera-system.7.dylib
libgz-sim-log-system.dylib                                libgz-sim7-logical-camera-system.dylib
libgz-sim-log-video-recorder-system.dylib                 libgz-sim7-logicalaudiosensorplugin-system.7.0.0.dylib
libgz-sim-logical-camera-system.dylib                     libgz-sim7-logicalaudiosensorplugin-system.7.dylib
libgz-sim-logicalaudiosensorplugin-system.dylib           libgz-sim7-logicalaudiosensorplugin-system.dylib
libgz-sim-magnetometer-system.dylib                       libgz-sim7-magnetometer-system.7.0.0.dylib
libgz-sim-mecanum-drive-system.dylib                      libgz-sim7-magnetometer-system.7.dylib
[... cut ...]
Warning: Unbrewed dylibs were found in /usr/local/lib.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected dylibs:
  /usr/local/lib/librtlsdr.0.0.0.dylib
  /usr/local/lib/librtlsdr.0.dylib
  /usr/local/lib/librtlsdr.dylib
  /usr/local/lib/libusb-1.0.0.dylib
```
My Rbenv config:
```
➜  ~ which ruby
/Users/alex/.rbenv/shims/ruby
➜  ~ ruby --version
ruby 3.1.2p20 (2022-04-12 revision 4491bb740a) [arm64-darwin21]
```

scpeters, 2022-11-02:
`ign gazebo -s` should now work if you use `ruby` from `brew` with the following command:

~~~
/usr/local/opt/ruby/bin/ruby $(which ign) gazebo -s
~~~

closed 2022-11-02 by scpeters