osrf/homebrew-simulation #1190: SHA256 Errors

mydansun, 2020-10-27:
When I was trying...
```
brew install gazebo7
```

I got some SHA256 errors and I checked the following commit.

https://github.com/osrf/homebrew-simulation/commit/dffb4b450c6bc9d3c98aed9d44696a35bac77068

I am wondering why these SHA256 values are not updated along with the patch replacing?

Thanks.

jordanrburger, 2020-11-12:
I've also gotten the same error when trying to upgrade Gazebo11 via Homebrew. 

```brew upgrade osrf/simulation/gazebo11
Updating Homebrew...
==> Upgrading 1 outdated package:
osrf/simulation/gazebo11 HEAD-7441aaecc696_2 -> HEAD-7441aaecc696_1
==> Upgrading osrf/simulation/gazebo11 HEAD-7441aaecc696_2 -> HEAD-7441aaecc696_1 
```

scpeters, 2020-11-12:
that was a mistake on my part; pull requests are welcome if I don't fix it quickly enough

scpeters, 2020-11-12:
I fixed a few in 2c92509cdc25545377ff3416021e19af75a264dd

scpeters, 2020-11-12:
I believe I've fixed the rest in a55fb5d7367f4430f0e4defac5283c0ccc912cdf; please re-open if you find any other such errors

thanks for reporting

closed 2020-11-12 by scpeters

jordanrburger, 2020-11-12:
Hey, @scpeters 

Getting a new error when trying now. 
```
(robotics) Jordans-MacBook-Pro:~ jordanburger$ brew upgrade osrf/simulation/gazebo11
==> Upgrading 1 outdated package:
osrf/simulation/gazebo11 HEAD-7441aaecc696_2 -> HEAD-7441aaecc696_1
==> Upgrading osrf/simulation/gazebo11 HEAD-7441aaecc696_2 -> HEAD-7441aaecc696_1 
==> Downloading https://homebrew.bintray.com/bottles/openblas-0.3.12_1.mojave.bottle.tar.gz
Already downloaded: /Users/jordanburger/Library/Caches/Homebrew/downloads/08703f6cf31f1d02c59f14a5ae2afee0c425bf7144c9e83ab9dfe987f987d1d0--openblas-0.3.12_1.mojave.bottle.tar.gz
==> Downloading https://homebrew.bintray.com/bottles/dartsim-6.9.5_1.mojave.bottle.tar.gz
Already downloaded: /Users/jordanburger/Library/Caches/Homebrew/downloads/c3c617bccb4eef8169f549b77648ffc6ec3e37f951df0ed399f805f018d8dfd3--dartsim-6.9.5_1.mojave.bottle.tar.gz
==> Downloading https://github.com/osrf/gazebo/commit/02e77e7bbc86697c1c55e1a1bbaf1360e5f31a68.patch?full_index=1
Already downloaded: /Users/jordanburger/Library/Caches/Homebrew/downloads/9ac8e802004ab4f02f9d92db1376a66c2d7316f96627a3e2de60cc6de22cf928--02e77e7bbc86697c1c55e1a1bbaf1360e5f31a68.patch
==> Downloading https://gist.githubusercontent.com/scpeters/9199370/raw/afe595587e38737c537124a3652db99de026c272/brew_python_fix.patch
######################################################################## 100.0%
==> Downloading https://github.com/osrf/gazebo
######################################################################## 100.0%
Warning: Cannot verify integrity of b8ae08905b8df6b63a4b6cecb4b5bbdc6add66c7c789b77b834548f7c9171bf2--gazebo
A checksum was not provided for this resource.
For your reference the SHA-256 is: dcab7b4d1cce3f52657183581426267ade9561232388debd13e0f7e83e33e41c
```

Additionally, this message has popped up: 

```
==> Summary
🍺  /usr/local/Cellar/openblas/0.3.12_1: 23 files, 120.2MB
==> Installing osrf/simulation/gazebo11 dependency: dartsim
==> Pouring dartsim-6.9.5_1.mojave.bottle.tar.gz
Warning: dartsim dependency gcc was built with a different C++ standard
library (libstdc++ from clang). This may cause problems at runtime.
🍺  /usr/local/Cellar/dartsim/6.9.5_1: 1,081 files, 44.4MB
==> Installing osrf/simulation/gazebo11 --HEAD
==> Patching
==> Applying 02e77e7bbc86697c1c55e1a1bbaf1360e5f31a68.patch
can't find file to patch at input line 18
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From 02e77e7bbc86697c1c55e1a1bbaf1360e5f31a68 Mon Sep 17 00:00:00 2001
|From: Steve Peters <scpeters@openrobotics.org>
|Date: Fri, 23 Oct 2020 13:34:53 -0700
|Subject: [PATCH] SimpleTrackedVehiclePlugin: fix for boost 1.74
|
|Disable some code in SimpleTrackedVehiclePlugin for boost 1.74
|or greater due to changes in boost shared_ptr.hpp
|
|Signed-off-by: Steve Peters <scpeters@openrobotics.org>
|---
| plugins/SimpleTrackedVehiclePlugin.cc | 4 ++++
| 1 file changed, 4 insertions(+)
|
|diff --git a/plugins/SimpleTrackedVehiclePlugin.cc b/plugins/SimpleTrackedVehiclePlugin.cc
|index 3be4776f8f0d4f2ab80a42907de4caf4fdb087c6..c5a22b0278cb53ea424c4923a108ee4b75dacaa5 100644
|--- a/plugins/SimpleTrackedVehiclePlugin.cc
|+++ b/plugins/SimpleTrackedVehiclePlugin.cc
--------------------------
No file to patch.  Skipping patch.
3 out of 3 hunks ignored

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/osrf/homebrew-simulation/issues
```

scpeters, 2020-11-12:
ok, it looks like you've built with `--HEAD`?

scpeters, 2020-11-12:
can you use the bottled version instead? we don't have enough resources to support `--HEAD` builds