osrf/homebrew-simulation #1196: Provide components separately

chapulina, 2020-11-03:
We've been breaking Ignition libraries into components that can be used independent of each other at the CMake level, as well as the Debian level. However, when it comes to Homebrew, each library is still provided as a single bottle, which makes it impossible for downstream projects to depend only on the specific components they need.

Would it be possible to add separate formulae for each component of a library, and leave the formula with the library's name as a "meta-formula" that just brings the rest?

For example, we could have the following bottles, each building a single component:

* `ignition-common3-core`
* `ignition-common3-av`
* `ignition-common3-graphics`
* `ignition-common3-events`
* `ignition-common3-profiler`

And the `ignition-common3` formula either keeps doing what it does now, or it installs all the components above.

This would make it possible for more libraries to use `ignition-common3-core` without bringing the large dependencies of `av`, `graphics`, etc.

scpeters, 2020-11-04:
Homebrew doesn't make it easy to support components from individual formulae, though it is not impossible. For example, the [boost](https://github.com/Homebrew/homebrew-core/blob/master/Formula/boost.rb) formula includes most boost components but [explicitly excludes python and api components](https://github.com/Homebrew/homebrew-core/blob/1c3606a8426812bafde9d23f22476ae6975ecbe8/Formula/boost.rb#L48-L55) as there are separate formulae for these ([boost-python3](https://github.com/Homebrew/homebrew-core/blob/master/Formula/boost-python3.rb) and [boost-mpi](https://github.com/Homebrew/homebrew-core/blob/master/Formula/boost-mpi.rb)) as well as the [boost-bcp](https://github.com/Homebrew/homebrew-core/blob/master/Formula/boost-bcp.rb) tool. These formulae all point to the same tarball, which must be updated in a synchronous manner when there is a new version.

Here is an example for how to install component-specific files from [boost-mpi](https://github.com/Homebrew/homebrew-core/blob/c40d454/Formula/boost-mpi.rb#L48-L58). These lists of files should match the content of the `*.install` files in [ign-common3-release](https://github.com/ignition-release/ign-common3-release/tree/master/ubuntu/debian).

I did notice that `ignition-common3`'s header files are not completely segregated by component in the debian packages. For example, `libignition-common3-core-dev` contains some header files from the `av` component:

~~~
$ dpkg -L libignition-common3-core-dev | grep d.o
/usr/include/ignition/common3/ignition/common/AudioDecoder.hh
/usr/include/ignition/common3/ignition/common/Video.hh
/usr/include/ignition/common3/ignition/common/VideoEncoder.hh
~~~

It seems that `boost` has the same issue, which installs the `boost/mpi` header files from the main `boost` formula, not `boost-mpi`.

chapulina, 2020-11-04:
> ignition-common3's header files are not completely segregated by component in the debian packages.

I think that's something to be fixed: https://github.com/ignition-release/ign-common4-release/issues/1

---

Thank you for the analysis, @scpeters . I'm not hearing that it would be impossible, it would just take some work to make sure the formulae are cleanly split.

scpeters, 2020-11-04:
> I'm not hearing that it would be impossible, it would just take some work to make sure the formulae are cleanly split.

it's not impossible, but it will take work to implement and the added complexity will increase the on-going maintenance burden to keep component formulae in sync and keep an accurate list of files to be installed for each component

scpeters, 2024-11-22:
this is not easy to do with brew, so we aren't planning to do it at this point

closed 2024-11-22 by scpeters