osrf/homebrew-simulation #1200: Ignition-math6 build failing

varunagrawal, 2020-11-13:
When trying to install `sdformat8` via `brew install sdformat8`, I get the error `Error: osrf/simulation/ignition-math6 6.6.0 did not build`.

This does not happen when I build `ignition-math` from the Github source.

[Here](https://gist.github.com/varunagrawal/fb007a210e4ef2c309425b704240e360) is the link to the error log.

scpeters, 2020-11-23:
Thanks for the report. I've never tried using this tap with LinuxBrew since we support installation through debian packages on Linux. Out of curiosity, what is your reason for using LinuxBrew instead of our debian packages?

It looks like the problem is related to swig, which we added as a build dependency in https://github.com/osrf/homebrew-simulation/pull/1121 to fix https://github.com/osrf/homebrew-simulation/issues/1120. Can you try commenting out the `depends_on "ruby"` line? If that doesn't work, can you also comment out `depends_on "swig"`? We could potentially make that dependency OS-dependent

varunagrawal, 2020-12-27:
We use brew across Linux and MacOS to manage dependencies better, so it made sense to use brew to install this on Linux as well.

scpeters, 2020-12-27:
I just tested this, and I can't reproduce the failure. Can you confirm it is still failing? I believe it was caused by the `depends on "swig"` line that was removed in #1124

varunagrawal, 2020-12-27:
Works for me on Ubuntu 20.04, previously I was using 16.04 but I don't think that was the issue. Thanks for the update @scpeters and apologies for the late response. Happy holidays!

varunagrawal, 2021-08-03:
I am facing this issue again on Ubuntu 18.04 (as the environment on Github Actions). I'd really like to figure this out properly this time.

varunagrawal, 2021-08-03:
Another issue is that I can't even use `apt` since our application depends on SDFormat8, which isn't available for Ubuntu.

closed 2022-04-12 by varunagrawal