osrf/homebrew-simulation #1245: Change qt dependencies to qt5

scpeters, 2020-12-24:
A formula for qt 6.0.0 has been submitted to homebrew-core in https://github.com/Homebrew/homebrew-core/pull/67536, so we should change our dependencies on `qt` to use the `qt5` alias instead. We also need to update build scripts and documentation instructions for setting `CMAKE_PREFIX_PATH` to `/usr/local/opt/qt`.

scpeters, 2021-01-04:
I've proposed creating a `qt@5` alias in https://github.com/Homebrew/homebrew-core/pull/68241, so use that when it is available

scpeters, 2021-01-04:
> I've proposed creating a `qt@5` alias in [Homebrew/homebrew-core#68241](https://github.com/Homebrew/homebrew-core/pull/68241), so use that when it is available

that has been merged, so we can now do this

traversaro, 2021-03-27:
I guess this was fixed by https://github.com/osrf/homebrew-simulation/pull/1264/files ? @scpeters

scpeters, 2021-05-18:
this has been done

closed 2021-05-18 by scpeters