osrf/homebrew-simulation #1248: Unable to Download Gazebo 11

sdhruv616, 2020-12-25:
I am new to downloading apps with homebrew but everytime i've tried to install gazebo11 for a project, the installation crashes towards the end. I'm really confused about what to do. Towards the end of the installation, the terminal shows the following:

>
==> Installing osrf/simulation/gazebo11
==> Patching
==> Applying brew_python_fix.patch
patching file deps/opende/CMakeLists.txt
==> cmake .. -DQWT_WIN_INCLUDE_DIR=/usr/local/lib/qwt.framework/Heade
==> make install
Last 15 lines from /Users/Dhruv/Library/Logs/Homebrew/gazebo11/02.make:
cd /tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0/build/gazebo/msgs && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBUILDING_DLL -DBUILDING_DLL_GZ_MSGS -DHAVE_OPENGL -DIGN_PROFILER_ENABLE=0 -DLIBBULLET_VERSION=3.08 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -Dgazebo_msgs_EXPORTS -I/tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0/test/gtest/include -I/tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0 -I/tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0/build -I/tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0/build/gazebo/msgs -I/usr/local/Cellar/ffmpeg/4.3.1_4/include -I/usr/local/include/ignition/transport8 -isystem /usr/local/Cellar/tinyxml2/8.0.0/include -isystem /usr/local/include/ignition/msgs5 -isystem /usr/local/include/ignition/math6 -isystem /usr/local/include/ignition/common3 -isystem /usr/local/include/ignition/fuel_tools4 -iframework /Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk/System/Library/Frameworks -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/Paging -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/Overlay -isystem /usr/local/Cellar/sdformat9/9.3.0_1/include/sdformat-9.3 -isystem /usr/local/opt/tinyxml/include -isystem /usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -isystem /usr/local/include/sdformat-9.3/sdf/.. -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Wswitch-default -Wmissing-include-dirs -pedantic -std=c++11  -s -O3 -DNDEBUG -mssse3 -msse3 -msse -msse2  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk -fPIC   -fPIC -fvisibility=default -I/usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -std=c++17 -o CMakeFiles/gazebo_msgs.dir/world_stats.pb.cc.o -c /tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0/build/gazebo/msgs/world_stats.pb.cc
[ 39%] Building CXX object gazebo/msgs/CMakeFiles/gazebo_msgs.dir/wrench.pb.cc.o
cd /tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0/build/gazebo/msgs && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBUILDING_DLL -DBUILDING_DLL_GZ_MSGS -DHAVE_OPENGL -DIGN_PROFILER_ENABLE=0 -DLIBBULLET_VERSION=3.08 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -Dgazebo_msgs_EXPORTS -I/tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0/test/gtest/include -I/tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0 -I/tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0/build -I/tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0/build/gazebo/msgs -I/usr/local/Cellar/ffmpeg/4.3.1_4/include -I/usr/local/include/ignition/transport8 -isystem /usr/local/Cellar/tinyxml2/8.0.0/include -isystem /usr/local/include/ignition/msgs5 -isystem /usr/local/include/ignition/math6 -isystem /usr/local/include/ignition/common3 -isystem /usr/local/include/ignition/fuel_tools4 -iframework /Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk/System/Library/Frameworks -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/Paging -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/Overlay -isystem /usr/local/Cellar/sdformat9/9.3.0_1/include/sdformat-9.3 -isystem /usr/local/opt/tinyxml/include -isystem /usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -isystem /usr/local/include/sdformat-9.3/sdf/.. -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Wswitch-default -Wmissing-include-dirs -pedantic -std=c++11  -s -O3 -DNDEBUG -mssse3 -msse3 -msse -msse2  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk -fPIC   -fPIC -fvisibility=default -I/usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -std=c++17 -o CMakeFiles/gazebo_msgs.dir/wrench.pb.cc.o -c /tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0/build/gazebo/msgs/wrench.pb.cc
[ 39%] Building CXX object gazebo/msgs/CMakeFiles/gazebo_msgs.dir/wrench_stamped.pb.cc.o
cd /tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0/build/gazebo/msgs && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBUILDING_DLL -DBUILDING_DLL_GZ_MSGS -DHAVE_OPENGL -DIGN_PROFILER_ENABLE=0 -DLIBBULLET_VERSION=3.08 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -Dgazebo_msgs_EXPORTS -I/tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0/test/gtest/include -I/tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0 -I/tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0/build -I/tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0/build/gazebo/msgs -I/usr/local/Cellar/ffmpeg/4.3.1_4/include -I/usr/local/include/ignition/transport8 -isystem /usr/local/Cellar/tinyxml2/8.0.0/include -isystem /usr/local/include/ignition/msgs5 -isystem /usr/local/include/ignition/math6 -isystem /usr/local/include/ignition/common3 -isystem /usr/local/include/ignition/fuel_tools4 -iframework /Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk/System/Library/Frameworks -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/Paging -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/Overlay -isystem /usr/local/Cellar/sdformat9/9.3.0_1/include/sdformat-9.3 -isystem /usr/local/opt/tinyxml/include -isystem /usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -isystem /usr/local/include/sdformat-9.3/sdf/.. -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Wswitch-default -Wmissing-include-dirs -pedantic -std=c++11  -s -O3 -DNDEBUG -mssse3 -msse3 -msse -msse2  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk -fPIC   -fPIC -fvisibility=default -I/usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -std=c++17 -o CMakeFiles/gazebo_msgs.dir/wrench_stamped.pb.cc.o -c /tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0/build/gazebo/msgs/wrench_stamped.pb.cc
[ 39%] Building CXX object gazebo/msgs/CMakeFiles/gazebo_msgs.dir/msgs.cc.o
[... cut ...] sr/local/lib/libtar.dylib /usr/local/lib/libsdformat9.9.3.0.dylib -ltinyxml /usr/local/lib/libignition-msgs5.5.5.0.dylib "/usr/local/lib/libignition-math6.6.8.0~pre1.dylib" /usr/local/lib/libprotobuf.dylib 
cd /tmp/gazebo11-20201225-1938-oua9zf/gazebo-11.3.0/build/gazebo/msgs && /usr/local/Cellar/cmake/3.19.2/bin/cmake -E cmake_symlink_library libgazebo_msgs.11.3.0.dylib libgazebo_msgs.11.dylib libgazebo_msgs.dylib
[ 40%] Built target gazebo_msgs
make: *** [all] Error 2

Any help is appreciated!

scpeters, 2020-12-26:
I think the console is displaying a truncated log. Perhaps you can look at the full contents of `~/Library/Logs/Homebrew/gazebo11/02.make`

sdhruv616, 2020-12-26:
I looked into the contents of 02.make but found no mistakes or anything. I think the problem is emerging when the installation is being patched. Because all the dependencies were successfully downloaded

scpeters, 2020-12-27:
I think there is an error message in the log file I mentioned. Can you upload a copy to gist.github.com and post a link here? Look for a line with `***`

thebirdgr, 2020-12-29:
@scpeters I have the similar issue. I've added my log file, I'm new so I'm not sure what error to look for, ***. 
Here's the gist [link](https://gist.github.com/thebirdgr/b584297521ae254021ebd638b3ab5052).

lweingart, 2020-12-29:
Hello,
I'm facing the same issue as well.
[Here](https://gist.github.com/lweingart/ea98a58c6b612f465bca793f3171fa09) is the link to my 02.make

*** at lines 754, 755, 780, 781

Cheers

thefireholder, 2020-12-29:
Hi, I have the same problem downloading gazebo 11.

the error appears after
[ 40%] Built target gazebo_msgs
for gazebo11

I also tried downloading older version of gazebo (gazebo 9).
it seems to be suffering from the same problem:

the error appears after
[ 41%] Built target gazebo_msgs
for gazebo9

what @sdhruv616 mentioned about "xcode developer tools " is NOT the problem because when I run "brew info qt" I have 
==> Dependencies
Build: pkg-config ✔
==> Requirements
Build: Xcode ✔
Required: macOS >= 10.12 ✔

scpeters, 2020-12-30:
I believe this is an issue on Big Sur; please :+1: this comment if you have this issue on Big Sur or reply with the name of the macOS version you are using when you experience it.

I believe it is a problem with a brew dependency, since I saw this line:

~~~
make[2]: *** No rule to make target `/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/lib/libm.tbd', needed by `deps/opende/libgazebo_ode.11.3.0.dylib'.  Stop.
~~~

The `MacOSX11.0.sdk` path is outdated; it should now be 11.1, and I believe these paths are hard-coded in some of the homebrew-core packages.

brofori, 2021-01-04:
I had the same issue and solved it by creating a symlink from MacOSXsdk version 11.1 to 11.0.
`ln -s /Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk/ /Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk`

It's just a workaround but I hope it helps

onerachel, 2021-01-05:
I got this:

ln: /Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk: Permission denied

any idea how to solve it?thx

> I had the same issue and solved it by creating a symlink from MacOSXsdk version 11.1 to 11.0.
> `ln -s /Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk/ /Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk`
> 
> It's just a workaround but I hope it helps

onerachel, 2021-01-05:
I got the following message:
ln: /Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk: Permission denied

anyway to fix this? thx

> I had the same issue and solved it by creating a symlink from MacOSXsdk version 11.1 to 11.0.
> `ln -s /Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk/ /Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk`
> 
> It's just a workaround but I hope it helps

onerachel, 2021-01-06:
I had the same issue. In my case one of the equirements -Build: Xcode doesn't meet. 
therefore I installed the full Xcode ( https://wilsonmar.github.io/xcode/#UnInstallCU), then problem solved.

jasonhan-vassar, 2021-01-08:
Hi, I'm having trouble on Big Sur 11.1 as well. Here is the [02.make](https://github.com/osrf/homebrew-simulation/files/5787713/02.make.zip).

It seems like the line says `No rule to make target `/usr/lib/libm.dylib', needed by `deps/opende/libgazebo_ode.11.3.0.dylib'`, which is similar to your line @scpeters. I also see it calls `/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk`, which is a symlink to `MacOSX11.1.sdk` so I'm not sure if the hard-coded issue was fixed.

jasonhan-vassar, 2021-01-08:
After doing a bit of digging, it seems that Big Sur changed the location of these libraries:
[missing libraries from /usr/lib macOS 11 (formerly part of SDK) causing crashes](https://developer.apple.com/forums/thread/655588)
[Missing librairies in /usr/lib on Big Sur?](https://developer.apple.com/forums/thread/655588)

scpeters, 2021-01-08:
some of the brew bottles that are dependencies of gazebo11 had outdated SDK paths encoded within them. I think you can find this if you `grep -rlI MacOSX11.0.sdk $(brew --cellar)`, but since some of these have already been fixed, I think a `brew outdated` followed by a `brew upgrade` may fix some of it

please let me know if you see any dependencies that still have outdated paths

jasonhan-vassar, 2021-01-12:
@scpeters So Gazebo was successfully installed from homebrew, but I can't seem to compile the tutorial's [Hello WorldPlugin](http://gazebosim.org/tutorials/?tut=plugins_hello_world). I get the same ```No rule to make target `/usr/lib/libm.dylib'``` error when I `cmake ../ && make`. I did `make -n` and here is my output: [output.txt](https://github.com/osrf/homebrew-simulation/files/5799208/output.txt). Please let me know if you need any more info.

I did a `brew outdated` followed by a `brew upgrade`, but I'm not sure how relevant that is to plugins not compiling.

sandeepwadhwa-ai, 2021-07-25:
I am having trouble with the command
brew install gazebo11
Also tried building from source. My hardware is the Mac Book Air M1. The problem is occurring with installing ignition-tools which fails. Any idea how to circumvent this issue? I am only on ARM-64 arch.

-----Output from build from source or binary install is the same as below ------

Installing dependencies for osrf/simulation/gazebo11: ignition-tools, ignition-msgs5, ignition-fuel-tools4, ignition-transport8, libtar, ogre1.9, qwt, sdformat9, swig, tbb@2020_u3 and simbody
==> Installing osrf/simulation/gazebo11 dependency: ignition-tools
==> cmake ..
Last 15 lines from /Users/sandeepwadhwa/Library/Logs/Homebrew/ignition-tools/01.cmake:
  	ronn not found, manpages won't be generated

CMake Warning at CMakeLists.txt:213 (message):
  END BUILD WARNINGS

-- Custom CFlags:   -O3 -DNDEBUG  -std=c++11 -stdlib=libc++ -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic
-- Build Type: Release
-- Install path: /opt/homebrew/Cellar/ignition-tools/1.2.0
-- Configuration successful. Type make to compile
  ignition-tools
-- Configuring incomplete, errors occurred!
See also "/tmp/ignition-tools-20210725-20657-1w8i6g1/ignition-tools-1.2.0/build/CMakeFiles/CMakeOutput.log".

Do not report this issue to Homebrew/brew or Homebrew/core!

scpeters, 2021-12-23:
I believe this has been resolved. Please reopen if the issue persists

closed 2021-12-23 by scpeters