osrf/homebrew-simulation #1314: Nightlies for main branches

chapulina, 2021-01-30:
Our debian nightlies are very convenient to get fresh debians from the `main` branches of all Ignition libraries. It would be great to have the same for homebrew. 

At the moment, it's still necessary to open a pull request here bumping the commit version of main branches when they're needed by downstream libraries. This could be automated, with GitHub actions for example, to happen nightly.

scpeters, 2021-02-01:
We have separate repositories in packages.osrfoundation.org for stable, prerelease, and nightly packages, but for homebrew we only have one version of things, which typically points to the latest stable release or prerelease. Before a formula has had any releases, we just point to the tarball for a specific commit on github and update it manually as needed.

I assume this proposal would apply only to packages that have not yet had a prerelease or stable release?

chapulina, 2021-02-03:
> I assume this proposal would apply only to packages that have not yet had a prerelease or stable release?

Correct, just like the nightlies, this would only apply to the `main` branches. Although in some cases there will be overlap between pre-release and nightly, especially as we start shipping pre-releases close to the stable release date.

chapulina, 2021-05-12:
> just like the nightlies, this would only apply to the main branches.

The debian nightlies are now using Fortress branches instead of `main`, see https://github.com/ignition-tooling/release-tools/pull/432

> I assume this proposal would apply only to packages that have not yet had a prerelease or stable release?

The homebrew nightlies would either:

* Use `main` branches, which are considered unstable and haven't had stable releases.
    * This should leave no ambiguity about the status of those packages.
    * We'd need to make pre- / stable releases  whenever we need a change from stable branches.
* Use Fortress branches, which consist of a combination of stable and unstable branches.
    * This would allow iterating faster on Fortress, like we're doing with debian nightlies.
    * This may introduce regressions on stable branches. We could consider this a good thing (detecting regressions early) or a bad thing (using homebrew as a canary).

scpeters, 2021-05-13:
One very easy option occurred to me for unreleased packages: use the git repository as the url instead of a tarball of an explicit commit. We already do this for `cppzmq` for example:

https://github.com/osrf/homebrew-simulation/blob/master/Formula/cppzmq.rb#L4-L5

This would actually be closer to our windows CI than Ubuntu nightlies, since it would build the `main` branch of unreleased packages from source. We may actually be able to build binaries, but I think it wouldn't be easy to know which commit was used since the `version` tag will not include the commit `sha`

scpeters, 2021-05-22:
Using main branch for all new fortress packages in #1481

scpeters, 2021-05-25:
closing this since we are using the `main` branch for all unreleased fortress packages

closed 2021-05-25 by scpeters