osrf/homebrew-simulation #1343: 👩‍🌾 New warning: conflicting chdir during another chdir block

chapulina, 2021-02-23:
This warning started showing up at our Homebrew CI recently:

```
==> Pouring gettext-0.21.mojave.bottle.tar.gz
/usr/local/Homebrew/Library/Homebrew/download_strategy.rb:94: warning: conflicting chdir during another chdir block
```

It shows up as _Clang (LLVM based) Warnings_ and turns the build yellow, like [this one](https://build.osrfoundation.org/job/ignition_gazebo-ci-pr_any-homebrew-amd64/5080/clang/).

The warning is coming from [this line on homebrew](https://github.com/Homebrew/brew/blob/3efab3efa7114afe713b2485032b740e6b55d6e4/Library/Homebrew/download_strategy.rb#L94).

I'm not sure the exact situation that causes the nested chdir calls or whether it can be prevented from our code. My little understanding tells me that it isn't something we can fix, in which case I think we should suppress this warning.

---

https://github.com/osrf/buildfarmer/issues/161

scpeters, 2021-02-23:
I think this was introduced by https://github.com/Homebrew/brew/pull/10598; I'll make a comment there

scpeters, 2021-02-23:
I can't reproduce this on my laptop

scpeters, 2021-02-23:
> I can't reproduce this on my laptop

it seems to happen only on the first install after deleting everything. I'm trying a test run with some extra debug information

* https://github.com/ignition-tooling/release-tools/commit/fb2bdc9b8111d160d0a71189d4009b2dfc4f2891
* [image] https://build.osrfoundation.org/job/ignition_transport8-install_bottle-homebrew-amd64/262/console

scpeters, 2021-02-24:
I'm removing all `brew` packages from a test machine to debug further

scpeters, 2021-02-24:
I've commented there: https://github.com/Homebrew/brew/pull/10598/files#r581498395

chapulina, 2021-02-24:
Thanks for investigating, @scpeters . Do you know if there's a way to suppress this warning on our CI?

Rylan12, 2021-02-24:
You can set the `HOMEBREW_RUBY_WARNINGS` environment variable to `-W0` to suppress this warning.

scpeters, 2021-02-24:
> You can set the `HOMEBREW_RUBY_WARNINGS` environment variable to `-W0` to suppress this warning.

@Rylan12 thanks for the tip, but I'm having trouble getting that to work in this case. It didn't seem to help with our test build using https://github.com/ignition-tooling/release-tools/pull/401

* testing: [image] https://build.osrfoundation.org/view/os_homebrew/job/ignition_cmake-ci-ign-cmake2-homebrew-amd64/75/

I'll suppress it manually with a `grep -v` for now

closed 2021-02-24 by chapulina

Rylan12, 2021-02-24:
Hmm, not sure why that didn't work. When I simulate a warning and run the following it's ignored as expected. Sorry, it didn't work, but it sounds like you've handled it with the `grep`.

```sh
HOMEBREW_RUBY_WARNINGS="-W0" brew command
```

scpeters, 2021-03-03:
> Hmm, not sure why that didn't work. When I simulate a warning and run the following it's ignored as expected. Sorry, it didn't work, but it sounds like you've handled it with the `grep`.
> 
> ```shell
> HOMEBREW_RUBY_WARNINGS="-W0" brew command
> ```

I just had another warning pop up, and I looked into why setting `HOMEBREW_RUBY_WARNINGS` didn't have an effect and found the following bit of code:

* https://github.com/Homebrew/brew/blob/3.0.4/Library/Homebrew/brew.sh#L559-L569

we don't have the `HOMEBREW_DEVELOPER` environment variable set, so `brew.sh` is unsetting it and forcing `-W1`. I'll try setting that variable as well.

scpeters, 2021-03-03:
> we don't have the `HOMEBREW_DEVELOPER` environment variable set, so `brew.sh` is unsetting it and forcing `-W1`. I'll try setting that variable as well.

even `HOMEBREW_DEVELOPER` doesn't seem to help. I'll try to disable clang warning scans for our bottle building jobs since we don't need it there anyways

* https://build.osrfoundation.org/job/generic-release-homebrew_triggered_bottle_builder/292/label=osx_mojave/clang/

Rylan12, 2021-03-03:
> I just had another warning pop up

Any chance you can report this to Homebrew? It should probably either be fixed or ignored on our end (depending on what it is).

We do have a PR open to fix the other warning but it's blocked at the moment because no one's figured out how to properly deprecate the old (and incorrect) functionality because its part of the public API.

scpeters, 2021-03-03:
I think https://github.com/Homebrew/brew/pull/10689 has already been merged

the other warning I mentioned is a complaint about permissions in the vendor/bundle folder:

~~~
++ /usr/local/bin/brew audit cmake
/usr/local/Homebrew/Library/Homebrew/utils/gems.rb:69: warning: Insecure world writable dir /usr/local/Homebrew/Library/Homebrew/vendor/bundle in PATH, mode 040777
~~~

I manually fixed this, but it should be straightforward to reproduce. @Rylan12 to clarify, what should I report?

* The fact that this warning pops up in a `brew audit` call
* My failure to suppress it with `HOMEBREW_RUBY_WARNINGS`

Rylan12, 2021-03-03:
> I think [Homebrew/brew#10689](https://github.com/Homebrew/brew/pull/10689) has already been merged

Oops, you're right, guess I wasn't paying enough attention 😅

If you could report the `brew audit` warning that would be the main thing. It's probably a good idea to mention that `HOMEBREW_RUBY_WARNINGS` (and `HOMEBREW_DEVELOPER`) didn't seem to have any effect.