osrf/homebrew-simulation #1350: gazebo-config.cmake not found

traversaro, 2021-03-02:
I am not sure if this is happening due to something in this repo, but since yesterday all the C++ projects that uses Gazebo which have GitHub Action-based CI and installed gazebo with `brew install ` are failing with the following error:
~~~
2021-03-02T02:45:24.1357690Z CMake Error at CMakeLists.txt:49 (find_package):
2021-03-02T02:45:24.1459720Z   By not providing "FindGazebo.cmake" in CMAKE_MODULE_PATH this project has
2021-03-02T02:45:24.1559900Z   asked CMake to find a package configuration file provided by "Gazebo", but
2021-03-02T02:45:24.1661500Z   CMake did not find one.
2021-03-02T02:45:24.1761340Z 
2021-03-02T02:45:24.1861940Z   Could not find a package configuration file provided by "Gazebo" with any
2021-03-02T02:45:24.1963270Z   of the following names:
2021-03-02T02:45:24.2063420Z 
2021-03-02T02:45:24.2163950Z     GazeboConfig.cmake
2021-03-02T02:45:24.2265890Z     gazebo-config.cmake
2021-03-02T02:45:24.2365210Z 
2021-03-02T02:45:24.2466220Z   Add the installation prefix of "Gazebo" to CMAKE_PREFIX_PATH or set
2021-03-02T02:45:24.2567270Z   "Gazebo_DIR" to a directory containing one of the above files.  If "Gazebo"
2021-03-02T02:45:24.2668060Z   provides a separate development package or SDK, be sure it has been
2021-03-02T02:45:24.2768320Z   installed.
~~~
even if `brew install osrf/simulation/gazebo11` is installed. I am not sure if this is related to this message:
~~~
2021-03-02T02:23:49.7826230Z [34m==>[0m [1mReinstalling 4 broken dependents from source:[0m
2021-03-02T02:23:49.7828630Z osrf/simulation/ignition-msgs5, osrf/simulation/ignition-fuel-tools4, osrf/simulation/ignition-transport8, osrf/simulation/gazebo11
~~~

For an example of failing builds:
* https://github.com/robotology/robotology-superbuild/runs/2009469519?check_suite_focus=true
* https://github.com/robotology/gazebo-yarp-plugins/runs/2004219148?check_suite_focus=true

traversaro, 2021-03-02:
Indeed, in the successful builds there is no message "Reinstalling 4 broken dependents from source:", see https://github.com/robotology/gazebo-yarp-plugins/runs/1996417168?check_suite_focus=true .

scpeters, 2021-03-02:
I believe this is related to #1340 

I just disabled all the broken bottles, so things should build from source for now

traversaro, 2021-03-03:
Thanks @scpeters , the CI jobs are working correctly now (see for example https://github.com/robotology/robotology-superbuild/actions/runs/616114431).

closed 2021-03-03 by traversaro