osrf/homebrew-simulation #1356: Unable to install Gazebo9 on macOS Big Sur

jpowell7, 2021-03-05:
Ran the following commands:

brew update
brew upgrade
brew tap osrf/simulation
brew tap dartsim/dart
brew install dartsim
brew install gazebo9 --build-from-source -v

I got the following error:

cd /tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/build/gazebo/physics && /usr/local/Cellar/cmake/3.19.6/bin/cmake -E cmake_symlink_library libgazebo_physics.9.16.0.dylib libgazebo_physics.9.dylib libgazebo_physics.dylib
[ 61%] Built target gazebo_physics
make: *** [all] Error 2

==> Formula
Tap: osrf/simulation
Path: /usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation/Formula/gazebo9.rb
==> Configuration
HOMEBREW_VERSION: 3.0.4-42-g2d1d20a
ORIGIN: https://github.com/Homebrew/brew
HEAD: 2d1d20a236c8f5d868605e54e4c0365bd5d50ebb
Last commit: 3 hours ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: b6c3df9d487259e09a517cb6886a2220d51f2216
Core tap last commit: 27 minutes ago
Core tap branch: master
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.kbxLQE0NkC/org.xquartz:0
HOMEBREW_MAKE_JOBS: 8
Homebrew Ruby: 2.6.3 => /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby
CPU: octa-core 64-bit kabylake
Clang: 12.0 build 1200
Git: 2.30.1 => /usr/local/bin/git
Curl: 7.64.1 => /usr/bin/curl
macOS: 11.2.2-x86_64
CLT: 12.4.0.0.1.1610135815
Xcode: N/A
XQuartz: 2.7.11 => /opt/X11
==> ENV
HOMEBREW_CC: clang
HOMEBREW_CXX: clang++
MAKEFLAGS: -j8
CMAKE_PREFIX_PATH: /usr/local/opt/icu4c:/usr/local/opt/libffi:/usr/local/opt/openssl@1.1:/usr/local/opt/readline:/usr/local/opt/sqlite:/usr/local/opt/tcl-tk:/usr/local/opt/qt@5:/usr/local/opt/openblas:/usr/local/opt/jpeg-turbo:/usr/local
CMAKE_INCLUDE_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Headers
CMAKE_LIBRARY_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries
CMAKE_FRAMEWORK_PATH: /usr/local/opt/python@3.9/Frameworks:/usr/local/opt/qt/Frameworks:/usr/local/opt/qt@5/Frameworks
PKG_CONFIG_PATH: /usr/local/opt/icu4c/lib/pkgconfig:/usr/local/opt/libpng/lib/pkgconfig:/usr/local/opt/freetype/lib/pkgconfig:/usr/local/opt/fontconfig/lib/pkgconfig:/usr/local/opt/jpeg/lib/pkgconfig:/usr/local/opt/libtiff/lib/pkgconfig:/usr/local/opt/webp/lib/pkgconfig:/usr/local/opt/gd/lib/pkgconfig:/usr/local/opt/libffi/lib/pkgconfig:/usr/local/opt/pcre/lib/pkgconfig:/usr/local/opt/openssl@1.1/lib/pkgconfig:/usr/local/opt/readline/lib/pkgconfig:/usr/local/opt/sqlite/lib/pkgconfig:/usr/local/opt/tcl-tk/lib/pkgconfig:/usr/local/opt/xz/lib/pkgconfig:/usr/local/opt/python@3.9/lib/pkgconfig:/usr/local/opt/glib/lib/pkgconfig:/usr/local/opt/jasper/lib/pkgconfig:/usr/local/opt/netpbm/lib/pkgconfig:/usr/local/opt/gts/lib/pkgconfig:/usr/local/opt/libpthread-stubs/lib/pkgconfig:/usr/local/opt/libxau/lib/pkgconfig:/usr/local/opt/libxdmcp/lib/pkgconfig:/usr/local/opt/libxcb/lib/pkgconfig:/usr/local/opt/libx11/lib/pkgconfig:/usr/local/opt/libxext/lib/pkgconfig:/usr/local/opt/libxrender/lib/pkgconfig:/usr/local/opt/lzo/lib/pkgconfig:/usr/local/opt/pixman/lib/pkgconfig:/usr/local/opt/cairo/lib/pkgconfig:/usr/local/opt/gdk-pixbuf/lib/pkgconfig:/usr/local/opt/fribidi/lib/pkgconfig:/usr/local/opt/gobject-introspection/lib/pkgconfig:/usr/local/opt/graphite2/lib/pkgconfig:/usr/local/opt/harfbuzz/lib/pkgconfig:/usr/local/opt/pango/lib/pkgconfig:/usr/local/opt/librsvg/lib/pkgconfig:/usr/local/opt/graphviz/lib/pkgconfig:/usr/local/opt/aom/lib/pkgconfig:/usr/local/opt/dav1d/lib/pkgconfig:/usr/local/opt/frei0r/lib/pkgconfig:/usr/local/opt/gmp/lib/pkgconfig:/usr/local/opt/bdw-gc/lib/pkgconfig:/usr/local/opt/guile/lib/pkgconfig:/usr/local/opt/libidn2/lib/pkgconfig:/usr/local/opt/libtasn1/lib/pkgconfig:/usr/local/opt/nettle/lib/pkgconfig:/usr/local/opt/p11-kit/lib/pkgconfig:/usr/local/opt/libevent/lib/pkgconfig:/usr/local/opt/c-ares/lib/pkgconfig:/usr/local/opt/jemalloc/lib/pkgconfig:/usr/local/opt/nghttp2/lib/pkgconfig:/usr/local/opt/unbound/lib/pkgconfig:/usr/local/opt/gnutls/lib/pkgconfig:/usr/local/opt/libass/lib/pkgconfig:/usr/local/opt/libbluray/lib/pkgconfig:/usr/local/opt/libsoxr/lib/pkgconfig:/usr/local/opt/libvidstab/lib/pkgconfig:/usr/local/opt/libogg/lib/pkgconfig:/usr/local/opt/libvorbis/lib/pkgconfig:/usr/local/opt/libvpx/lib/pkgconfig:/usr/local/opt/opencore-amr/lib/pkgconfig:/usr/local/opt/little-cms2/lib/pkgconfig:/usr/local/opt/openjpeg/lib/pkgconfig:/usr/local/opt/opus/lib/pkgconfig:/usr/local/opt/rav1e/lib/pkgconfig:/usr/local/opt/rtmpdump/lib/pkgconfig:/usr/local/opt/flac/lib/pkgconfig:/usr/local/opt/libsndfile/lib/pkgconfig:/usr/local/opt/libsamplerate/lib/pkgconfig:/usr/local/opt/rubberband/lib/pkgconfig:/usr/local/opt/sdl2/lib/pkgconfig:/usr/local/opt/speex/lib/pkgconfig:/usr/local/opt/srt/lib/pkgconfig:/usr/local/opt/leptonica/lib/pkgconfig:/usr/local/opt/tesseract/lib/pkgconfig:/usr/local/opt/theora/lib/pkgconfig:/usr/local/opt/x264/lib/pkgconfig:/usr/local/opt/x265/lib/pkgconfig:/usr/local/opt/libsodium/lib/pkgconfig:/usr/local/opt/zeromq/lib/pkgconfig:/usr/local/opt/zimg/lib/pkgconfig:/usr/local/opt/ffmpeg/lib/pkgconfig:/usr/local/opt/ignition-math4/lib/pkgconfig:/usr/local/opt/ossp-uuid/lib/pkgconfig:/usr/local/opt/tinyxml2/lib/pkgconfig:/usr/local/opt/ignition-common1/lib/pkgconfig:/usr/local/opt/jsoncpp/lib/pkgconfig:/usr/local/opt/libyaml/lib/pkgconfig:/usr/local/opt/libzip/lib/pkgconfig:/usr/local/opt/ignition-fuel-tools1/lib/pkgconfig:/usr/local/opt/protobuf/lib/pkgconfig:/usr/local/opt/ignition-tools/lib/pkgconfig:/usr/local/opt/ignition-msgs1/lib/pkgconfig:/usr/local/opt/ignition-transport4/lib/pkgconfig:/usr/local/opt/libzzip/lib/pkgconfig:/usr/local/opt/ogre1.9/lib/pkgconfig:/usr/local/opt/protobuf-c/lib/pkgconfig:/usr/local/opt/assimp/lib/pkgconfig:/usr/local/opt/dbus/lib/pkgconfig:/usr/local/opt/libb2/lib/pkgconfig:/usr/local/opt/libproxy/lib/pkgconfig:/usr/local/opt/pcre2/lib/pkgconfig:/usr/local/opt/zstd/lib/pkgconfig:/usr/local/opt/qt@5/lib/pkgconfig:/usr/local/opt/tinyxml/lib/pkgconfig:/usr/local/opt/sdformat6/lib/pkgconfig:/usr/local/opt/bullet/lib/pkgconfig:/usr/local/opt/libccd/lib/pkgconfig:/usr/local/opt/octomap/lib/pkgconfig:/usr/local/opt/fcl/lib/pkgconfig:/usr/local/opt/isl/lib/pkgconfig:/usr/local/opt/mpfr/lib/pkgconfig:/usr/local/opt/flann/lib/pkgconfig:/usr/local/opt/openblas/lib/pkgconfig:/usr/local/opt/ipopt/lib/pkgconfig:/usr/local/opt/nlopt/lib/pkgconfig:/usr/local/opt/ode/lib/pkgconfig:/usr/local/opt/atk/lib/pkgconfig:/usr/local/opt/gtk+/lib/pkgconfig:/usr/local/opt/gtkglext/lib/pkgconfig:/usr/local/opt/jpeg-turbo/lib/pkgconfig:/usr/local/opt/sdl/lib/pkgconfig:/usr/local/opt/open-scene-graph/lib/pkgconfig:/usr/local/opt/console_bridge/lib/pkgconfig:/usr/local/opt/urdfdom_headers/lib/pkgconfig:/usr/local/opt/urdfdom/lib/pkgconfig:/usr/local/opt/dartsim/lib/pkgconfig:/usr/local/opt/simbody/lib/pkgconfig:/usr/local/opt/xorgproto/share/pkgconfig:/usr/local/opt/eigen/share/pkgconfig
PKG_CONFIG_LIBDIR: /usr/lib/pkgconfig:/usr/local/Homebrew/Library/Homebrew/os/mac/pkgconfig/11
HOMEBREW_GIT: git
HOMEBREW_SDKROOT: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk
ACLOCAL_PATH: /usr/local/share/aclocal
PATH: /usr/local/Homebrew/Library/Homebrew/shims/mac/super:/usr/local/opt/cmake/bin:/usr/local/opt/pkg-config/bin:/usr/local/opt/icu4c/bin:/usr/local/opt/doxygen/bin:/usr/local/opt/libpng/bin:/usr/local/opt/freetype/bin:/usr/local/opt/fontconfig/bin:/usr/local/opt/jpeg/bin:/usr/local/opt/libtiff/bin:/usr/local/opt/webp/bin:/usr/local/opt/gd/bin:/usr/local/opt/gettext/bin:/usr/local/opt/pcre/bin:/usr/local/opt/gdbm/bin:/usr/local/opt/openssl@1.1/bin:/usr/local/opt/sqlite/bin:/usr/local/opt/tcl-tk/bin:/usr/local/opt/xz/bin:/usr/local/opt/python@3.9/bin:/usr/local/opt/glib/bin:/usr/local/opt/jasper/bin:/usr/local/opt/netpbm/bin:/usr/local/opt/gts/bin:/usr/local/opt/cairo/bin:/usr/local/opt/gdk-pixbuf/bin:/usr/local/opt/fribidi/bin:/usr/local/opt/gobject-introspection/bin:/usr/local/opt/graphite2/bin:/usr/local/opt/harfbuzz/bin:/usr/local/opt/pango/bin:/usr/local/opt/librsvg/bin:/usr/local/opt/libtool/bin:/usr/local/opt/graphviz/bin:/usr/local/opt/aom/bin:/usr/local/opt/dav1d/bin:/usr/local/opt/guile/bin:/usr/local/opt/libidn2/bin:/usr/local/opt/libtasn1/bin:/usr/local/opt/nettle/bin:/usr/local/opt/p11-kit/bin:/usr/local/opt/libevent/bin:/usr/local/opt/c-ares/bin:/usr/local/opt/jemalloc/bin:/usr/local/opt/nghttp2/bin:/usr/local/opt/gnutls/bin:/usr/local/opt/lame/bin:/usr/loc [... cut ...]
/usr/bin/curl --disable --globoff --show-error --user-agent Homebrew/3.0.4-42-g2d1d20a\ \(Macintosh\;\ Intel\ Mac\ OS\ X\ 11.2.2\)\ curl/7.64.1 --header Accept-Language:\ en --retry 3 --location https://api.github.com/search/issues\?q=gazebo9\+repo\%3Aosrf\%2Fhomebrew-simulation\+state\%3Aopen\+in\%3Atitle\&per_page=100 --header Accept:\ application/vnd.github.v3\+json --write-out '
'\%\{http_code\} --header Accept:\ application/vnd.github.antiope-preview\+json --dump-header /private/tmp/github_api_headers20210305-9357-1ep28uh

scpeters, 2021-03-05:
there isn't quite enough output here to see what actually went wrong; I think there may be more info in one of the following files:

~~~
/Users/jpowell/Library/Logs/Homebrew/gazebo9/00.options.out
/Users/jpowell/Library/Logs/Homebrew/gazebo9/01.cmake.cc
/Users/jpowell/Library/Logs/Homebrew/gazebo9/01.cmake
/Users/jpowell/Library/Logs/Homebrew/gazebo9/02.make.cc
/Users/jpowell/Library/Logs/Homebrew/gazebo9/02.make
~~~

I'm guessing `/Users/jpowell/Library/Logs/Homebrew/gazebo9/02.make`

that said, I just fixed an error in the `qt` dependency in c981d1df09cd4ca1294b2685057d2e2ee5512d24 and am rebuilding the bottle in https://github.com/osrf/homebrew-simulation/pull/1357 for mojave and catalina. I haven't tried building it on big sur yet, so please share more detailed logs with your failure once you are able to do so

jpowell7, 2021-03-05:
02.make has more info. Here is the output when it started erroring.

[ 58%] Generating qrc_resources.cpp
cd /tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/build/gazebo/gui && /usr/local/opt/qt@5/bin/rcc --name resources --output /tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/build/gazebo/gui/qrc_resources.cpp /tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/gazebo/gui/resources.qrc
[ 60%] Building CXX object gazebo/physics/CMakeFiles/gazebo_physics.dir/Model.cc.o
cd /tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/build/gazebo/physics && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBUILDING_DLL -DBUILDING_DLL_GZ_PHYSICS -DGZ_PROFILER_ENABLE=0 -DHAVE_OPENGL -DLIBBULLET_VERSION=3.08 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -Dgazebo_physics_EXPORTS -I/tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/test/gtest/include -I/tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0 -I/tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/build -I/usr/local/Cellar/bullet/3.09/include/bullet -I/usr/local/Cellar/bullet/3.09/include -isystem /tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/deps/opende/include -isystem /usr/local/Cellar/tinyxml2/8.0.0/include -isystem /usr/local/Cellar/ffmpeg/4.3.2/include -isystem /usr/local/include/ignition/msgs1 -isystem /usr/local/include/ignition/transport4 -isystem /usr/local/include/ignition/fuel_tools1 -isystem /usr/local/include/ignition/common1 -isystem /usr/local/include/ignition/math4 -iframework /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/Paging -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/Overlay -isystem /usr/local/Cellar/sdformat6/6.2.0_1/include/sdformat-6.2 -isystem /usr/local/opt/tinyxml/include -isystem /usr/local/include/simbody -isystem /usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -isystem /usr/local/include/eigen3 -isystem /usr/local/Cellar/urdfdom/1.0.4_2/include -isystem /usr/local/lib/urdfdom_headers/cmake/../../../include -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Wswitch-default -Wmissing-include-dirs -pedantic    -O3 -DNDEBUG -mssse3 -msse3 -msse -msse2  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk -fPIC   -fPIC -I/usr/local/Cellar/bullet/3.09/include/bullet -I/usr/local/Cellar/bullet/3.09/include -I"/usr/local/include/simbody" -I/usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -std=c++14 -o CMakeFiles/gazebo_physics.dir/Model.cc.o -c /tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/gazebo/physics/Model.cc
cd /tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/build && /usr/local/Cellar/cmake/3.19.6/bin/cmake -E cmake_depends "Unix Makefiles" /tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0 /tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/gazebo/gui /tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/build /tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/build/gazebo/gui /tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/build/gazebo/gui/CMakeFiles/gazebo_gui.dir/DependInfo.cmake --color=
[ 60%] Building CXX object gazebo/physics/CMakeFiles/gazebo_physics.dir/ModelState.cc.o
cd /tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/build/gazebo/physics && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBUILDING_DLL -DBUILDING_DLL_GZ_PHYSICS -DGZ_PROFILER_ENABLE=0 -DHAVE_OPENGL -DLIBBULLET_VERSION=3.08 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -Dgazebo_physics_EXPORTS -I/tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/test/gtest/include -I/tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0 -I/tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/build -I/usr/local/Cellar/bullet/3.09/include/bullet -I/usr/local/Cellar/bullet/3.09/include -isystem /tmp/gazebo9-20210305-9583-1qf7up9/gazebo-9.16.0/deps/opende/include -isystem /usr/local/Cellar/tinyxml2/8.0.0/include -isystem /usr/local/Cellar/ffmpeg/4.3.2/include -isystem /usr/local/include/ignition/msgs1 -isystem /usr/local/include/ignition/transport4 -isystem /usr/local/include/ignition/fuel_tools1 -isystem /usr/local/include/ignition/common1 -isystem /usr/local/include/ignition/math4 -iframework /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ff [... cut ...] ssimp/bin:/usr/local/opt/dbus/bin:/usr/local/opt/libproxy/bin:/usr/local/opt/pcre2/bin:/usr/local/opt/zstd/bin:/usr/local/opt/qt/bin:/usr/local/opt/qt@5/bin:/usr/local/opt/octomap/bin:/usr/local/opt/gcc/bin:/usr/local/opt/hdf5/bin:/usr/local/opt/flann/bin:/usr/local/opt/ampl-mp/bin:/usr/local/opt/ipopt/bin:/usr/local/opt/ode/bin:/usr/local/opt/gtk+/bin:/usr/local/opt/jpeg-turbo/bin:/usr/local/opt/sdl/bin:/usr/local/opt/open-scene-graph/bin:/usr/local/opt/urdfdom/bin:/usr/bin:/bin:/usr/sbin:/sbin

scpeters, 2021-03-05:
yeah that looks related to the `qt@5` change. I've just merged the new bottle for catalina in #1357; can you try installing again?

jpowell7, 2021-03-05:
I am using Big Sur. Do you have a new bottle for Big Sur yet?

I am getting similar errors with #1357.

[ 60%] Building CXX object gazebo/gui/CMakeFiles/gazebo_gui.dir/gazebo_gui_autogen/mocs_compilation.cpp.o
cd /tmp/gazebo9-20210305-33903-1vth71t/gazebo-9.16.0/build/gazebo/gui && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBUILDING_DLL -DBUILDING_DLL_GZ_GUI -DGZ_PROFILER_ENABLE=0 -DHAVE_OPENGL -DHAVE_STRING_H -DLIBBULLET_VERSION=3.08 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -DQGVCORE_LIB -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQWT_DLL -DWITH_CGRAPH -Dgazebo_gui_EXPORTS -I/tmp/gazebo9-20210305-33903-1vth71t/gazebo-9.16.0/build/gazebo/gui/gazebo_gui_autogen/include -I/tmp/gazebo9-20210305-33903-1vth71t/gazebo-9.16.0/test/gtest/include -I/tmp/gazebo9-20210305-33903-1vth71t/gazebo-9.16.0 -I/tmp/gazebo9-20210305-33903-1vth71t/gazebo-9.16.0/build -I/tmp/gazebo9-20210305-33903-1vth71t/gazebo-9.16.0/deps -I/tmp/gazebo9-20210305-33903-1vth71t/gazebo-9.16.0/gazebo/gui/qgv -I/tmp/gazebo9-20210305-33903-1vth71t/gazebo-9.16.0/gazebo/gui/qgv/private -I/usr/local/lib/qwt.framework/Headers -I/usr/local/Cellar/graphviz/2.46.1/include/graphviz -isystem /usr/local/Cellar/tinyxml2/8.0.0/include -isystem /usr/local/Cellar/ffmpeg/4.3.2/include -isystem /usr/local/include/ignition/msgs1 -isystem /usr/local/include/ignition/transport4 -isystem /usr/local/include/ignition/fuel_tools1 -isystem /usr/local/include/ignition/common1 -isystem /usr/local/include/ignition/math4 -iframework /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/Paging -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_9/include/OGRE/Overlay -isystem /usr/local/Cellar/sdformat6/6.2.0_1/include/sdformat-6.2 -isystem /usr/local/opt/tinyxml/include -iframework /usr/local/opt/qt@5/lib -isystem /usr/local/opt/qt@5/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt@5/./mkspecs/macx-clang -isystem /usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -isystem /usr/local/opt/qt@5/lib/QtWidgets.framework/Headers -isystem /usr/local/opt/qt@5/lib/QtGui.framework/Headers -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Wswitch-default -Wmissing-include-dirs -pedantic    -O3 -DNDEBUG -mssse3 -msse3 -msse -msse2  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk -fPIC   -fPIC -F/usr/local/lib  -I/usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -fPIC -std=c++14 -o CMakeFiles/gazebo_gui.dir/gazebo_gui_autogen/mocs_compilation.cpp.o -c /tmp/gazebo9-20210305-33903-1vth71t/gazebo-9.16.0/build/gazebo/gui/gazebo_gui_autogen/mocs_compilation.cpp
In file included from /tmp/gazebo9-20210305-33903-1vth71t/gazebo-9.16.0/build/gazebo/gui/gazebo_gui_autogen/mocs_compilation.cpp:2:
In file included from /tmp/gazebo9-20210305-33903-1vth71t/gazebo-9.16.0/build/gazebo/gui/gazebo_gui_autogen/EWIEGA46WW/moc_Actions.cpp:10:
In file included from /tmp/gazebo9-20210305-33903-1vth71t/gazebo-9.16.0/build/gazebo/gui/gazebo_gui_autogen/EWIEGA46WW/../../../../../gazebo/gui/Actions.hh:22:
In file included from /tmp/gazebo9-20210305-33903-1vth71t/gazebo-9.16.0/gazebo/gui/qt.h:25:
In file included from /usr/local/lib/QtCore.framework/Headers/QtCore:4:
/usr/local/lib/QtCore.framework/Headers/qglobal.h:667:43: error: no template named 'is_arithmetic_v' in namespace 'std'; did you mean 'is_arithmetic'?
         typename = std::enable_if_t<std::is_arithmetic_v<T> && std::is_arithmetic_v<U> &&
                                     ~~~~~^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/type_traits:994:50: note: 'is_arithmetic' declared here
template <class _Tp> struct _LIBCPP_TEMPLATE_VIS is_arithmetic
                                                 ^
In file included from /tmp/gazebo9-20210305-33903-1vth71t/gazebo-9.16.0/build/gazebo/gui/gazebo_gui_autogen/mocs_compilation.cpp:2:
In file included from /tmp/gazebo9-20210305-33903-1vth71t/gazebo-9.16.0/build/gazebo/gui/gazebo_gui_autogen/EWIEGA46WW/moc_Actions.cpp:10:
In file included from /tmp/gazebo9-20210305-33903-1vth71t/gazebo-9.16.0/build/gazebo/gui/gazebo_gui_autogen/EWIEGA46WW/../../../../../gazebo/gui/Actions.hh:22:
In file included from /tmp/gazebo9-20210305-33903-1vth71t/gazebo-9.16.0/gazebo/gui/qt.h:25:
In file included from /usr/local/lib/QtCore.framework/Headers/QtCore:4:
/usr/local/lib/QtCore.framework/Headers/qglobal.h:667:70: error: no template named 'is_arithmetic_v' in namespace 'std'; did you mean 'is_arithmetic'?
         typename = std::enable_if_t<std::is_arithmetic_v<T> && std::is_arithmetic_v<U> &&
                                                                ~~~~~^
[... cut ...] r/local/opt/libtar/bin:/usr/local/opt/libzzip/bin:/usr/local/opt/ogre1.9/bin:/usr/local/opt/protobuf-c/bin:/usr/local/opt/qt@5/bin:/usr/local/opt/assimp/bin:/usr/local/opt/octomap/bin:/usr/local/opt/gcc/bin:/usr/local/opt/hdf5/bin:/usr/local/opt/flann/bin:/usr/local/opt/ampl-mp/bin:/usr/local/opt/ipopt/bin:/usr/local/opt/ode/bin:/usr/local/opt/gtk+/bin:/usr/local/opt/jpeg-turbo/bin:/usr/local/opt/sdl/bin:/usr/local/opt/open-scene-graph/bin:/usr/local/opt/urdfdom/bin:/usr/bin:/bin:/usr/sbin:/sbin

scpeters, 2021-03-05:
it looks like it's picking up header files from Qt 6 (homebrew formula `qt`). if you aren't using that formula for something else, can you try `brew remove qt` and then try building again?

jpowell7, 2021-03-08:
It worked! Thanks for the help.

closed 2021-03-08 by jpowell7