osrf/homebrew-simulation #1454: Gazebo build failure on M1 Macbook Air

JinraeKim, 2021-05-13:
➜  PX4-Autopilot git:(master)    brew install gazebo11 --build-from-source --verbose
==> Installing gazebo11 from osrf/simulation
==> Downloading https://osrf-distributions.s3.amazonaws.com/ign-tools/releases/ignition-tools-1.2.0.tar.bz2
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/1b06eaff964a3f9d6a4932baa28022e1e3e1532e5696f9719c3c9069ba63c28a--ignition-tools-1.2.0.tar.bz2
==> Verifying checksum for '1b06eaff964a3f9d6a4932baa28022e1e3e1532e5696f9719c3c9069ba63c28a--ignition-tools-1.2.0.tar.bz2'
==> Downloading https://osrf-distributions.s3.amazonaws.com/ign-msgs/releases/ignition-msgs5-5.7.0.tar.bz2
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/93b1b63dc9eb18d3adc8da06b82941d70c378bc0967343357d7feba626f7a1ea--ignition-msgs5-5.7.0.tar.bz2
==> Verifying checksum for '93b1b63dc9eb18d3adc8da06b82941d70c378bc0967343357d7feba626f7a1ea--ignition-msgs5-5.7.0.tar.bz2'
==> Downloading https://ghcr.io/v2/homebrew/core/libzip/manifests/1.7.3
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/1eae81f519e07c4554ea8c250548bdfef3ba8a04b8c44b6d8e75d21262329cdd--libzip-1.7.3.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/libzip/blobs/sha256:e58c6e33b3a74dc3cae81e1b3ab482e0df007fe65c561a62e4750e206c068688
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/4ee11b1771f6eb6458837d0955ff5202b4592b43cbe020b5dcfa1de2c969ab92--libzip--1.7.3.arm64_big_sur.bottle.tar.gz
==> Verifying checksum for '4ee11b1771f6eb6458837d0955ff5202b4592b43cbe020b5dcfa1de2c969ab92--libzip--1.7.3.arm64_big_sur.bottle.tar.gz'
==> Downloading https://osrf-distributions.s3.amazonaws.com/ign-fuel-tools/releases/ignition-fuel-tools4-4.3.0.tar.bz2
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/87d57ec7f631ee3cd50afe6c2af55c2d4eb5c75ec9d509f705cb05f030a2afeb--ignition-fuel-tools4-4.3.0.tar.bz2
==> Verifying checksum for '87d57ec7f631ee3cd50afe6c2af55c2d4eb5c75ec9d509f705cb05f030a2afeb--ignition-fuel-tools4-4.3.0.tar.bz2'
==> Cloning https://github.com/zeromq/cppzmq.git
Updating /Users/jinrae/Library/Caches/Homebrew/cppzmq--git
git config remote.origin.url https://github.com/zeromq/cppzmq.git
git config remote.origin.fetch \+refs/heads/master:refs/remotes/origin/master
git config remote.origin.tagOpt --no-tags
git fetch origin
==> Checking out branch master
git checkout -f master --
Already on 'master'
Your branch is up to date with 'origin/master'.
git reset --hard origin/master --
HEAD is now at b65dde8 Add support for socket_ref to multipart_t ctor/send/recv  (#487)
==> Downloading https://osrf-distributions.s3.amazonaws.com/ign-transport/releases/ignition-transport8-8.2.0.tar.bz2
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/303db63b922955ab6314e8fd467ebb3dc9fc7c7069716ed59aa4b5fc80fab4fe--ignition-transport8-8.2.0.tar.bz2
==> Verifying checksum for '303db63b922955ab6314e8fd467ebb3dc9fc7c7069716ed59aa4b5fc80fab4fe--ignition-transport8-8.2.0.tar.bz2'
==> Downloading https://ghcr.io/v2/homebrew/core/libtar/manifests/1.2.20-2
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/23b46358622748699c1fe273ecb72904ce38c0eff55d7d60d4d0a23726dd7640--libtar-1.2.20-2.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/libtar/blobs/sha256:7481da1834936b6237f152fe8b7e22196ad5c76833af39c9e9f74eae6347c9a5
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/10a1480ca13bc589d4eaf0555bb0a3e6961878c2d5d6bf5299e54b7fdd2cf89f--libtar--1.2.20.arm64_big_sur.bottle.2.tar.gz
==> Verifying checksum for '10a1480ca13bc589d4eaf0555bb0a3e6961878c2d5d6bf5299e54b7fdd2cf89f--libtar--1.2.20.arm64_big_sur.bottle.2.tar.gz'
==> Downloading https://ghcr.io/v2/homebrew/core/libzzip/manifests/0.13.72_1-1
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/a46abdc4edd838a203756040b54bd264f5bee02eb4e770e41500beb6c7000120--libzzip-0.13.72_1-1.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/libzzip/blobs/sha256:43fbba2b7e506170bf0f03a8c281c142b04cd1b95365392d36dcc014e5f24743
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/20b0de216f40fb264721b6f475dc765d6cbe29f5515d875272c5d4304da00d0f--libzzip--0.13.72_1.arm64_big_sur.bottle.1.tar.gz
==> Verifying checksum for '20b0de216f40fb264721b6f475dc765d6cbe29f5515d875272c5d4304da00d0f--libzzip--0.13.72_1.arm64_big_sur.bottle.1.tar.gz'
==> Downloading https://gist.github.com/NikolausDemmel/2b11d1b49b35cd27a102/raw/bf4a4d16020821218f73db0d56aa111ab2fde679/fix-1.9-HEAD.diff
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/e7c203b88a808ca752ebac27f8b00b295445f8651766e688eabac24ca91f2af8--fix-1.9-HEAD.diff
==> Verifying checksum for 'e7c203b88a808ca752ebac27f8b00b295445f8651766e688eabac24ca91f2af8--fix-1.9-HEAD.diff'
==> Downloading https://gist.githubusercontent.com/NikolausDemmel/927bd7bb3f14c1788599/raw/c9a5ba88b758e80d3f46511629c4e8026b92c462/ogre1.9.patch
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/ed6176508d9571d70a8a8b9c812c24e9fa7e56e4bfb6284e836cbafbfaa9c98c--ogre1.9.patch
==> Verifying checksum for 'ed6176508d9571d70a8a8b9c812c24e9fa7e56e4bfb6284e836cbafbfaa9c98c--ogre1.9.patch'
==> Downloading https://gist.githubusercontent.com/iche033/68349eebfc436e484b70e6e3508ae27b/raw/d96227cd79ed1d63f051689a18c95a1c174a4efa/ogre-1.9-ois.patch
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/4cf7ce49769de5262fd9a6a9858ff48c9104aaa3873fdd80419964a928372486--ogre-1.9-ois.patch
==> Verifying checksum for '4cf7ce49769de5262fd9a6a9858ff48c9104aaa3873fdd80419964a928372486--ogre-1.9-ois.patch'
==> Downloading https://gist.githubusercontent.com/iche033/b73766fac9ab3d628a79b5ed986677cd/raw/878d0902704c7fb51511163052c95294361f1dbe/ogre-1.9-agl.patch
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/decf335241bb718c38a96a2a2a70cfa096762c4beb4dc2f3e4a16f889ed36002--ogre-1.9-agl.patch
==> Verifying checksum for 'decf335241bb718c38a96a2a2a70cfa096762c4beb4dc2f3e4a16f889ed36002--ogre-1.9-agl.patch'
==> Downloading https://gist.githubusercontent.com/iche033/e0080a592c890cc9a4fce31f6863a5ed/raw/875ae8ad1d9f0eaa271fd44eab8e0979bac74119/ogre-1.9-cocoa_window_scale.patch
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/c15e631644dee75637c0cc48b026154a91280d8c057a25bfb1b3eda50e7499c8--ogre-1.9-cocoa_window_scale.patch
==> Verifying checksum for 'c15e631644dee75637c0cc48b026154a91280d8c057a25bfb1b3eda50e7499c8--ogre-1.9-cocoa_window_scale.patch'
==> Downloading https://gist.githubusercontent.com/iche033/e2b152d9df080b21f71ba3b65aa39922/raw/5397fa149c7570c17f2d78421d44b12dfa175387/ogre-1.9-cxx_flags.patch
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/d170e4463cc8b76d13766d512b11bdab092bbf4407be160a479c718b9a0a099b--ogre-1.9-cxx_flags.patch
==> Verifying checksum for 'd170e4463cc8b76d13766d512b11bdab092bbf4407be160a479c718b9a0a099b--ogre-1.9-cxx_flags.patch'
==> Downloading https://gist.githubusercontent.com/iche033/2b5e2ba31436881f1db29f9b60c7a5b2/raw/b6ab953ebd82127ad1177744f367a36e059312a9/ogre-1.9-findogre.patch
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/685334483037480ddd52ca2f1dbad6eb35a573818150b82cc2de714cb99becaf--ogre-1.9-findogre.patch
==> Verifying checksum for '685334483037480ddd52ca2f1dbad6eb35a573818150b82cc2de714cb99becaf--ogre-1.9-findogre.patch'
==> Downloading https://github.com/OGRECave/ogre/commit/cade48b4c2215dd77ea74b7aa219a22c2a5d6654.patch?full_index=1
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/d1d1b1546f3d7d50f8a0a37c28e7221c498c83ce5db8a239d770b7dc74bd1633--cade48b4c2215dd77ea74b7aa219a22c2a5d6654.patch
==> Verifying checksum for 'd1d1b1546f3d7d50f8a0a37c28e7221c498c83ce5db8a239d770b7dc74bd1633--cade48b4c2215dd77ea74b7aa219a22c2a5d6654.patch'
==> Downloading https://github.com/OGRECave/ogre/commit/2371c8d001725190a9cda62dc5df282cde78f951.patch?full_index=1
Already downloaded: /Users/jinrae/Library/Caches/Homebrew/downloads/b9b188483f25c506521c31da22540bd8c596fc9c38039cf0c96375a271f391b3--2371c8d001725190a9cda62dc5df282cde78f951.patch
==> Verifying checksum for 'b9b188483f25c506521c31da22540bd8c596fc9c38039cf0c96375a271f391b3--2371c8d001725190a9cda62dc5df282cde78f951.patch'
==> Downloading https://osrf-distributions.s3.amazonaws.com/ogre/releases/sinbad-ogre-108ab0bcc696.tar.bz2
[... cut ...]
PATH: /opt/homebrew/Library/Homebrew/shims/mac/super:/opt/homebrew/opt/cmake/bin:/usr/bin:/bin:/usr/sbin:/sbin

Error: osrf/simulation/ignition-tools 1.2.0 did not build
Logs:
     /Users/jinrae/Library/Logs/Homebrew/ignition-tools/00.options.out
     /Users/jinrae/Library/Logs/Homebrew/ignition-tools/01.cmake.cc
     /Users/jinrae/Library/Logs/Homebrew/ignition-tools/01.cmake
Do not report this issue to Homebrew/brew or Homebrew/core!

➜  PX4-Autopilot git:(master) brew gist-logs gazebo11
Error: No logs.

scpeters, 2021-12-23:
I believe this is mostly fixed with one remaining issue in #1698

scpeters, 2021-12-29:
should be fixed; please reopen if the issue persists

closed 2021-12-29 by scpeters