osrf/homebrew-simulation #1463: Rebuild bottles for protobuf 3.17.0

scpeters, 2021-05-18:
The protobuf 3.17.0 bottle is pending in https://github.com/Homebrew/homebrew-core/pull/77373 and will break all our bottles once it is merged. We should do the following:

1. Remove bottle blocks for everything downstream of protobuf, staring from `ignition-msgs*` and `gazebo*` so that they will build from source until bottle are rebuilt
2. Rebuild bottles incrementally.

j-rivero, 2021-05-18:
I think I can take this one, the homebrew-core was merged some minutes ago.

> Rebuild bottles incrementally.

Does it makes sense to use the same pull request to do it? Changing several revisions at once will work?

scpeters, 2021-05-18:
> I think I can take this one, the homebrew-core was merged some minutes ago.
> 
> > Rebuild bottles incrementally.
> 
> Does it makes sense to use the same pull request to do it? Changing several revisions at once will work?

it can work, but an sporadic infrastructure problem is more likely to affect them all and make the builds longer. it's a bit safer to cluster them together: one for each ignition stack (citadel, dome, edifice, fortress), and one (or 2) for gazebo9 / 11. it is more effort, but in my experience with build failures, it's nicer to segregate the packages

scpeters, 2021-05-18:
since all the bottles are gone at the moment, another alternative is to increment all the ign-msgs* versions in a single PR, then ign-transport* and go gradually that way. we could try to rebuild them all at once and see if it works, but it may take 12+ hours

scpeters, 2021-05-18:
thanks for the help @j-rivero; I'll let you take it from here

scpeters, 2021-05-19:
the build farm is clear so I got it started with #1466

closed 2021-05-23 by scpeters