osrf/homebrew-simulation #1477: macOS Big Sur gazebo11 runtime failure => dyld: Symbol not found: __ZNSt12length_errorD1Ev

mcjsk, 2021-05-21:
I installed `gazebo11` with the default [one-liner][0] on my MBP when running Catalina, and Gazebo worked.

I just upgraded to Big Sur, then ran:

`brew update && brew upgrade`

Now `gazebo` fails to run with:

```sh
~ » gazebo
dyld: Symbol not found: __ZNSt12length_errorD1Ev
  Referenced from: /usr/local/lib/libgazebo_common.11.dylib
  Expected in: /usr/local/opt/tbb/lib/libtbb.dylib
 in /usr/local/lib/libgazebo_common.11.dylib
zsh: abort      gazebo
```

`brew doctor` output:

```sh
~ » brew doctor
Please note that these warnings are just used to help the Homebrew maintainers
with debugging if you file an issue. If everything you use Homebrew for is
working fine: please don't worry or file an issue; just ignore this. Thanks!

Warning: You have the following deprecated, cask taps tapped:
  caskroom/cask
Untap them with `brew untap`.

Warning: "config" scripts exist outside your system or Homebrew directories.
`./configure` scripts often look for *-config scripts to determine if
software packages are installed, and which additional flags to use when
compiling and linking.

Having additional scripts in your path can confuse software installed via
Homebrew if the config script overrides a system or Homebrew-provided
script of the same name. We found the following "config" scripts:
  /Library/Frameworks/Python.framework/Versions/3.7/bin/python3.7-config
  /Library/Frameworks/Python.framework/Versions/3.7/bin/python3.7m-config
  /Library/Frameworks/Python.framework/Versions/3.7/bin/python3-config

Warning: You have unlinked kegs in your Cellar.
Leaving kegs unlinked can lead to build-trouble and cause formulae that depend on
those kegs to fail to run properly once built. Run `brew link` on these:
  python@3.9

Warning: Homebrew's "sbin" was not found in your PATH but you have installed
formulae that put executables in /usr/local/sbin.
Consider setting your PATH for example like so:
  echo 'export PATH="/usr/local/sbin:$PATH"' >> ~/.zshrc
```

`brew config` output:

```sh
~ » brew config
HOMEBREW_VERSION: 3.1.8
ORIGIN: https://github.com/Homebrew/brew
HEAD: e0bfcdaf258dadaaf8c102d4cb51f23b6fdb2c4c
Last commit: 3 days ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 4a2cac5078c318dcfc0068909cdb2631172ee844
Core tap last commit: 3 hours ago
Core tap branch: master
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.gXz3AcoN9O/org.xquartz:0
HOMEBREW_MAKE_JOBS: 4
Homebrew Ruby: 2.6.3 => /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3_2/bin/ruby
CPU: quad-core 64-bit kabylake
Clang: 12.0.5 build 1205
Git: 2.30.1 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 7.64.1 => /usr/bin/curl
macOS: 11.3.1-x86_64
CLT: 12.5.0.0.1.1617976050
Xcode: 12.5
```

[0]: http://gazebosim.org/tutorials?tut=install_on_mac&cat=install

scpeters, 2021-05-21:
thanks I had just noticed this and had started filling out a new issue; I'll add my comments here

scpeters, 2021-05-21:
There has been a new tbb version merged in https://github.com/Homebrew/homebrew-core/pull/77662, which has caused our gazebo bottles to break:

* [image] https://build.osrfoundation.org/job/gazebo-install-one_liner-homebrew-amd64/2055/

~~~
dyld: Symbol not found: __ZNSt12length_errorD1Ev
  Referenced from: /usr/local/lib/libgazebo_common.11.dylib
  Expected in: /usr/local/opt/tbb/lib/libtbb.dylib
 in /usr/local/lib/libgazebo_common.11.dylib
Error: osrf/simulation/gazebo11: failed
~~~

Furthermore, it appears that the API has changed and gazebo doesn't build from source against this new version in the `tbb` formula:

* [image] https://build.osrfoundation.org/job/gazebo-ci-pr_any-homebrew-amd64/2407/

~~~
gazebo/transport/Connection.hh:57:65: error: expected class name
    class GZ_TRANSPORT_VISIBLE ConnectionReadTask : public tbb::task
                                                                ^
~~~

Fortunately, they added a versioned `tbb@2020` formula in https://github.com/Homebrew/homebrew-core/pull/77662, so I think we should be able to use that with gazebo11

scpeters, 2021-05-21:
I've removed the broken bottles in #1478 and will update the gazebo formulae to use `tbb@2020` in a subsequent PR

scpeters, 2021-05-21:
upstream issue: https://github.com/osrf/gazebo/issues/2867

scpeters, 2021-05-21:
gazebo11 bottle updated in #1479

mcjsk, 2021-05-22:
Thanks, @scpeters. This is now fixed.

closed 2021-05-22 by mcjsk