osrf/homebrew-simulation #1486: Building with gazebo11 fails due to tbb issues

nkalupahana, 2021-05-28:
I'm working on building navigation2 on macOS for Foxy, and I'm using the gazebo11 formula. When it goes to build the `gazebo_ros` package using files from Gazebo, however, some different errors involving tbb show up:

```
In file included from /Users/runner/ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_properties.cpp:23:
In file included from /usr/local/include/gazebo-11/gazebo/transport/Node.hh:30:
In file included from /usr/local/include/gazebo-11/gazebo/transport/TopicManager.hh:35:
In file included from /usr/local/include/gazebo-11/gazebo/transport/SubscriptionTransport.hh:24:
/usr/local/include/gazebo-11/gazebo/transport/Connection.hh:57:65: error: expected class name
    class GZ_TRANSPORT_VISIBLE ConnectionReadTask : public tbb::task
                                                                ^
/usr/local/include/gazebo-11/gazebo/transport/Connection.hh:73:20: error: no type named 'task' in namespace 'tbb'
      public: tbb::task *execute()
              ~~~~~^
/usr/local/include/gazebo-11/gazebo/transport/Connection.hh:313:61: error: no member named 'allocate_root' in namespace 'tbb::v1::task'
                  ConnectionReadTask *task = new(tbb::task::allocate_root())
                                                 ~~~~~~~~~~~^
/usr/local/include/gazebo-11/gazebo/transport/Connection.hh:315:19: error: no type named 'enqueue' in namespace 'tbb::v1::task'; did you mean 'ifqueue'?
                  tbb::task::enqueue(*task);
                  ^~~~~~~~~~~~~~~~~~
                  ifqueue
/Applications/Xcode_12.4.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk/usr/include/net/if_var.h:230:9: note: 'ifqueue' declared here
struct  ifqueue {
        ^
In file included from /Users/runner/ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_properties.cpp:23:
/usr/local/include/gazebo-11/gazebo/transport/Node.hh:39:58: error: expected class name
    class GZ_TRANSPORT_VISIBLE PublishTask : public tbb::task
                                                         ^
/usr/local/include/gazebo-11/gazebo/transport/Node.hh:54:20: error: no type named 'task' in namespace 'tbb'
      public: tbb::task *execute()
              ~~~~~^
/usr/local/include/gazebo-11/gazebo/transport/Node.hh:162:52: error: no member named 'allocate_root' in namespace 'tbb::v1::task'
                PublishTask *task = new(tbb::task::allocate_root())
                                        ~~~~~~~~~~~^
/usr/local/include/gazebo-11/gazebo/transport/Node.hh:165:17: error: no type named 'enqueue' in namespace 'tbb::v1::task'; did you mean 'ifqueue'?
                tbb::task::enqueue(*task);
                ^~~~~~~~~~~~~~~~~~
                ifqueue
/Applications/Xcode_12.4.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk/usr/include/net/if_var.h:230:9: note: 'ifqueue' declared here
struct  ifqueue {
        ^
8 errors generated.
```

Since I'm doing this on GitHub Actions, the full run logs can be viewed at https://github.com/nkalupahana/ros2-foxy-macos/runs/2691418249?check_suite_focus=true.

Any idea what's going on? I have a feeling this might have something to do with the new tbb@20202_U3 formula integration, but I'm not sure.

scpeters, 2021-05-28:
yes, we have to set some additional environment variables in our CI to compile gazebo against the proper version of `tbb` (https://github.com/ignition-tooling/release-tools/pull/466):

~~~
export CMAKE_PREFIX_PATH=${CMAKE_PREFIX_PATH}:/usr/local/opt/tbb@2020_u3
export CPATH=${CPATH}:/usr/local/opt/tbb@2020_u3/include
export LIBRARY_PATH=${LIBRARY_PATH}:/usr/local/opt/tbb@2020_u3/lib
~~~

nkalupahana, 2021-05-28:
It worked! Thanks so much.

closed 2021-05-28 by nkalupahana