osrf/homebrew-simulation #1489: gazebo-config.cmake not found after Homebrew prints Reinstalling 4 broken dependents from source

traversaro, 2021-06-09:
Since ~2 days, the GitHub Action builds on our CI on macOS/Homebrew started failing, with errors similar to:
* https://github.com/osrf/homebrew-simulation/issues/1350
* https://github.com/osrf/homebrew-simulation/issues/1439

Upstream issue: https://github.com/robotology/robotology-superbuild/issues/789
Example of failure build: https://github.com/robotology/robotology-superbuild/runs/2779758876

@scpeters if you are already aware of this issues, feel free to let me know, I don't want to increase the noise level. : )

j-rivero, 2021-06-09:
We have seen the same problems in our buildfarm.

traversaro, 2021-06-10:
On our CI the problem seems solved: https://github.com/robotology/robotology-superbuild/actions/runs/923699821 .

j-rivero, 2021-06-10:
> On our CI the problem seems solved: https://github.com/robotology/robotology-superbuild/actions/runs/923699821 . Thanks @j-rivero !

you are welcome Silvio. Still rebuilding the bottles but broken ones should be out of the repository. https://github.com/osrf/homebrew-simulation/pull/1490 and linked PR can serve as a tracker.

traversaro, 2021-07-30:
This was fixed back in June: https://github.com/robotology/robotology-superbuild/issues/789#issuecomment-858368275 .

closed 2021-07-30 by traversaro