osrf/homebrew-simulation #1504: Gazebo failed to launch macOS 11.4

shrijitsingh99, 2021-06-10:
Gazebo installation command : ```brew install gazebo11 --build-from-source```

then include the link in the issue. Thanks!

aronim, 2021-06-12:
@shrijitsingh99 I had the same problem. I built ign-transport8 from source and replaced the dylibs installed by brew and it is now working.

```
git clone git@github.com:ignitionrobotics/ign-transport.git
cd ign-transport
git fetch
git checkout ignition-transport8_8.2.0
mkdir build
cd build
cmake .. -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/ignition-transport8/8.2.0_<whatever_version_is_installed>
cp lib/libignition-transport8-log.8.2.0.dylib lib/libignition-transport8.8.2.0.dylib /usr/local/Cellar/ignition-transport8/8.2.0_<whatever_version_is_installed>/lib/.
```

j-rivero, 2021-06-15:
probably fixed after #1511 and #1502

aronim, 2021-06-15:
@j-rivero thank you

j-rivero, 2021-06-15:
you are welcome!

closed 2021-06-15 by j-rivero