osrf/homebrew-simulation #1510: gazebo11 installation error on macOS 10.15

nkalupahana, 2021-06-14:
When running `brew install osrf/simulation/gazebo11` on macos-latest on GitHub Actions, I'm running into the following error:

```
==> Installing osrf/simulation/gazebo11
==> Patching
==> Applying brew_python_fix.patch
patching file deps/opende/CMakeLists.txt
==> cmake .. -DQWT_WIN_INCLUDE_DIR=/usr/local/lib/qwt.framework/Headers -DQWT_WIN_LIBRARY_DIR=/usr/local/lib/qwt.framework
==> make install
Last 15 lines from /Users/runner/Library/Logs/Homebrew/gazebo11/02.make:
  static const ClassData _class_data_;
               ^
/usr/local/include/ignition/msgs5/ignition/msgs/material.pb.h:422:43: error: no type named 'ClassData' in 'google::protobuf::Message'
  const ::PROTOBUF_NAMESPACE_ID::Message::ClassData*GetClassData() const final;
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/usr/local/include/ignition/msgs5/ignition/msgs/material.pb.h:769:16: error: unknown type name 'ClassData'
  static const ClassData _class_data_;
               ^
/usr/local/include/ignition/msgs5/ignition/msgs/material.pb.h:770:43: error: no type named 'ClassData' in 'google::protobuf::Message'
  const ::PROTOBUF_NAMESPACE_ID::Message::ClassData*GetClassData() const final;
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
14 errors generated.
make[2]: *** [gazebo/msgs/CMakeFiles/gazebo_msgs.dir/msgs.cc.o] Error 1
make[1]: *** [gazebo/msgs/CMakeFiles/gazebo_msgs.dir/all] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/osrf/homebrew-simulation/issues

These open issues may also help:
gazebo11 11.6.0 https://github.com/osrf/homebrew-simulation/pull/1502
Gazebo11 fails to launch macOS 11.4  https://github.com/osrf/homebrew-simulation/issues/1504
Unable to install gazebo11 on Big Sur https://github.com/osrf/homebrew-simulation/issues/1248
```

Any ideas what this might be about?
Here's the full CI run: https://github.com/nkalupahana/ros2-foxy-macos/runs/2815266622?check_suite_focus=true

nkalupahana, 2021-06-14:
Here's another CI run, where I just run `brew install osrf/simulation/gazebo11`, and get the same error: https://github.com/nkalupahana/ros2-foxy-macos/runs/2815831362?check_suite_focus=true

j-rivero, 2021-06-15:
probably fixed after #1511 and #1502

nkalupahana, 2021-06-15:
Alright, I'll watch them and once they both merge in I'll re-run my CI.