osrf/homebrew-simulation #1518: brew test-bot errors: conflict declarations

chapulina, 2021-06-21:
The latest PRs are failing with the following output. I'm not sure if something changed on brew that's requiring these conflicts to be declared now, or if we did something that caused a regression.

```
==> brew audit --tap=osrf/simulation --except=version
==> FAILED
osrf/simulation/gazebo10:
  * Formula gazebo2 should also have a conflict declared with gazebo10
  * Formula gazebo3 should also have a conflict declared with gazebo10
  * Formula gazebo4 should also have a conflict declared with gazebo10
  * Formula gazebo5 should also have a conflict declared with gazebo10
  * Formula gazebo6 should also have a conflict declared with gazebo10
  * Formula gazebo8 should also have a conflict declared with gazebo10
osrf/simulation/gazebo11:
  * Formula gazebo2 should also have a conflict declared with gazebo11
  * Formula gazebo3 should also have a conflict declared with gazebo11
  * Formula gazebo4 should also have a conflict declared with gazebo11
  * Formula gazebo5 should also have a conflict declared with gazebo11
  * Formula gazebo6 should also have a conflict declared with gazebo11
  * Formula gazebo8 should also have a conflict declared with gazebo11
osrf/simulation/gazebo9:
  * Formula gazebo2 should also have a conflict declared with gazebo9
  * Formula gazebo3 should also have a conflict declared with gazebo9
  * Formula gazebo4 should also have a conflict declared with gazebo9
  * Formula gazebo5 should also have a conflict declared with gazebo9
  * Formula gazebo6 should also have a conflict declared with gazebo9
  * Formula gazebo8 should also have a conflict declared with gazebo9
osrf/simulation/ignition-math4:
  * Formula ignition-math2 should also have a conflict declared with ignition-math4
osrf/simulation/ignition-math5:
  * Formula ignition-math2 should also have a conflict declared with ignition-math5
osrf/simulation/ignition-math6:
  * Formula ignition-math2 should also have a conflict declared with ignition-math6
osrf/simulation/sdformat6:
  * Formula sdformat should also have a conflict declared with sdformat6
  * Formula sdformat3 should also have a conflict declared with sdformat6
osrf/simulation/sdformat7:
  * Formula sdformat should also have a conflict declared with sdformat7
  * Formula sdformat3 should also have a conflict declared with sdformat7
Error: 25 problems in 8 formulae detected
Error: Formula gazebo2 should also have a conflict declared with gazebo10
Error: Formula gazebo3 should also have a conflict declared with gazebo10
Error: Formula gazebo4 should also have a conflict declared with gazebo10
Error: Formula gazebo5 should also have a conflict declared with gazebo10
Error: Formula gazebo6 should also have a conflict declared with gazebo10
Error: Formula gazebo8 should also have a conflict declared with gazebo10
Error: Formula gazebo2 should also have a conflict declared with gazebo11
Error: Formula gazebo3 should also have a conflict declared with gazebo11
Error: Formula gazebo4 should also have a conflict declared with gazebo11
Error: Formula gazebo5 should also have a conflict declared with gazebo11
Error: Formula gazebo6 should also have a conflict declared with gazebo11
Error: Formula gazebo8 should also have a conflict declared with gazebo11
Error: Formula gazebo2 should also have a conflict declared with gazebo9
Error: Formula gazebo3 should also have a conflict declared with gazebo9
Error: Formula gazebo4 should also have a conflict declared with gazebo9
Error: Formula gazebo5 should also have a conflict declared with gazebo9
Error: Formula gazebo6 should also have a conflict declared with gazebo9
Error: Formula gazebo8 should also have a conflict declared with gazebo9
Error: Formula ignition-math2 should also have a conflict declared with ignition-math4
Error: Formula ignition-math2 should also have a conflict declared with ignition-math5
Error: Formula ignition-math2 should also have a conflict declared with ignition-math6
Error: Formula sdformat should also have a conflict declared with sdformat6
Error: Formula sdformat3 should also have a conflict declared with sdformat6
Error: Formula sdformat should also have a conflict declared with sdformat7
Error: Formula sdformat3 should also have a conflict declared with sdformat7

Error: Process completed with exit code 1.
```

scpeters, 2021-06-21:
it was an upstream brew change: https://github.com/Homebrew/brew/pull/11541

I think we can just delete the conflicts between gazebo9/11 and the outdated versions. I'll make a PR to that effect

scpeters, 2021-06-21:
fix in https://github.com/osrf/homebrew-simulation/pull/1519

PR #1519 (2021-06-21): Remove asymmetric conflicts_with statements
A recent change to brew audit disallows asymmetric
conflicts_with statements. Since these all involve
old formulae, this removes the conflicts_with
statements rather than trying to make them symmetric.

Fixes #1518.
commit 5840fa55 (2021-06-21): Remove asymmetric conflicts_with statements (#1519)

A recent change to brew audit disallows asymmetric
conflicts_with statements. Since these all involve
old formulae, this removes the conflicts_with
statements rather than trying to make them symmetric.

Fixes #1518.
closed 2021-06-21 by scpeters