osrf/homebrew-simulation #1543: Installing and testing gazebo11 leads to error

nkalupahana, 2021-07-29:
Currently, I'm running the following simplified workflow on GitHub Actions:
```
jobs:
  build:
    runs-on: macos-latest
    steps:
      - name: Install
        run: |
          brew tap osrf/simulation
          brew update
          brew install osrf/simulation/gazebo11
          brew test osrf/simulation/gazebo11
```

However, I keep running into the following error when running the `test` command: https://github.com/nkalupahana/ros2-foxy-macos/runs/3196791174?check_suite_focus=true

This just started last night, and it worked the night before, so any insight into why this might be happening would be appreciated.

traversaro, 2021-07-30:
The critical message there is "Reinstalling 1 broken dependent from source", it tipically happens when gazebo's dependency break for some reason, see similar issues from the path:
* https://github.com/osrf/homebrew-simulation/issues/1350
* https://github.com/osrf/homebrew-simulation/issues/1439
* https://github.com/osrf/homebrew-simulation/issues/1489

traversaro, 2021-08-05:
Thanks a lot @scpeters, after https://github.com/osrf/homebrew-simulation/pull/1545 the issue is solved for me.

nkalupahana, 2021-08-05:
Yep, my builds are now passing as well.

closed 2021-08-05 by nkalupahana