osrf/homebrew-simulation #1665: gazebo9,11 bottles broken by dartsim 6.12.1

scpeters, 2021-11-11:
The minor version of dartsim in homebrew-core was recently upgraded from 6.11.1 to 6.12.1 in https://github.com/Homebrew/homebrew-core/pull/89141, which has broken our gazebo bottles:

* [image] https://build.osrfoundation.org/job/gazebo-install-one_liner-homebrew-amd64/2234/

We should open separate pull requests to bump the revision for gazebo9 and gazebo11 and rebuild the bottles. To do this, on a new branch run the following command for gazebo11:

~~~
brew bump-revision gazebo11 --message="for dartsim"
~~~

push the branch, open a pull request, and comment `build bottle`.

Then do the same for gazebo9.

nkalupahana, 2021-11-11:
^ can confirm, this has broken our builds that use gazebo11: https://github.com/nkalupahana/ros2-foxy-macos/runs/4176692550?check_suite_focus=true

nkalupahana, 2021-11-16:
All fixed for me!

closed 2021-12-08 by scpeters