osrf/homebrew-simulation #1682: Use --only-json-tab when building bottles to support reproducible builds

scpeters, 2021-11-16:
Per discussion in https://github.com/Homebrew/homebrew-test-bot/issues/705#issuecomment-970627557, use `--only-json-tab` when building bottles to support reproducible builds.

scpeters, 2021-11-16:
currently included in https://github.com/ignition-tooling/release-tools/pull/556

MikeMcQuaid, 2021-11-17:
Note: you'll need a way to have the tab installed from somewhere else. We get the tab from the GitHub Packages JSON metadata so you'd need some other way to get this tab outside of the tarball. This would be a fair addition to Homebrew/brew.

scpeters, 2021-11-17:
> Note: you'll need a way to have the tab installed from somewhere else. We get the tab from the GitHub Packages JSON metadata so you'd need some other way to get this tab outside of the tarball. This would be a fair addition to Homebrew/brew.

ok, that's a fair point. I'll revert the one line change from https://github.com/ignition-tooling/release-tools/pull/556 until we figure out the rest of this. Thanks for the warning!