osrf/homebrew-simulation #1746: Install works, but I get error running gazebo -  Library not loaded: @rpath/libSimTKmath.3.7.dylib

eecue, 2021-12-23:
Building in brew on M1 MacBook Pro.  Build works great (`brew install gazebo 11`, then I get the following error when trying to actually run `gazebo`:

```
~ gazebo
dyld[22367]: Library not loaded: @rpath/libSimTKmath.3.7.dylib
  Referenced from: /opt/homebrew/Cellar/simbody/3.7/lib/libSimTKsimbody.3.7.dylib
  Reason: tried: '/opt/homebrew/Cellar/gazebo11/11.9.1_2/lib/../lib/libSimTKmath.3.7.dylib' (no such file), '/opt/homebrew/Cellar/gazebo11/11.9.1_2/lib/../lib/libSimTKmath.3.7.dylib' (no such file), '/opt/homebrew/Cellar/gazebo11/11.9.1_2/lib/../lib/libSimTKmath.3.7.dylib' (no such file), '/opt/homebrew/Cellar/gazebo11/11.9.1_2/bin/../lib/libSimTKmath.3.7.dylib' (no such file), '/opt/homebrew/Cellar/gazebo11/11.9.1_2/bin/../lib/libSimTKmath.3.7.dylib' (no such file), '/usr/local/lib/libSimTKmath.3.7.dylib' (no such file), '/usr/lib/libSimTKmath.3.7.dylib' (no such file)Library not loaded: @rpath/libSimTKcommon.3.7.dylib
  Referenced from: /opt/homebrew/Cellar/simbody/3.7/lib/libSimTKmath.3.7.dylib
  Reason: tried: '/opt/homebrew/Cellar/gazebo11/11.9.1_2/lib/../lib/libSimTKcommon.3.7.dylib' (no such file), '/opt/homebrew/Cellar/gazebo11/11.9.1_2/lib/../lib/libSimTKcommon.3.7.dylib' (no such file), '/opt/homebrew/Cellar/gazebo11/11.9.1_2/lib/../lib/libSimTKcommon.3.7.dylib' (no such file), '/opt/homebrew/Cellar/gazebo11/11.9.1_2/bin/../lib/libSimTKcommon.3.7.dylib' (no such file), '/opt/homebrew/Cellar/gazebo11/11.9.1_2/bin/../lib/libSimTKcommon.3.7.dylib' (no such file), '/usr/local/lib/libSimTKcommon.3.7.dylib' (no such file), '/usr/lib/libSimTKcommon.3.7.dylib' (no such file)
[1]    22367 abort      gazebo
```

eecue, 2021-12-23:
Figured I'd include a `printenv` too:

```
➜  ~ printenv
__CFBundleIdentifier=com.apple.Terminal
XPC_FLAGS=0x0
TERM=xterm-256color
XPC_SERVICE_NAME=0
TERM_PROGRAM=Apple_Terminal
TERM_PROGRAM_VERSION=443
SHELL=/bin/zsh
PATH=/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/Library/Apple/usr/bin
SHLVL=1
HOMEBREW_PREFIX=/opt/homebrew
HOMEBREW_CELLAR=/opt/homebrew/Cellar
HOMEBREW_REPOSITORY=/opt/homebrew
MANPATH=/opt/homebrew/share/man::
INFOPATH=/opt/homebrew/share/info:
ZSH=/Users/eecue/.oh-my-zsh
PAGER=less
LESS=-R
LSCOLORS=Gxfxcxdxbxegedabagacad
EDITOR=mvim
LANG=en_US.UTF-8
_=/usr/bin/printenv
```

eecue, 2021-12-23:
Note that I did this from a totally clean install of homebrew with only `brew install homebrew/cask/xquartz` first.  `brew doctor` said all was good.

scpeters, 2021-12-23:
the simbody change in https://github.com/osrf/homebrew-simulation/pull/1698/ will fix it. I'll merge that shortly

scpeters, 2021-12-23:
this should be fixed by #1748

closed 2021-12-23 by scpeters